esphome/issues #2417: LightCall `set_transition_length` with value > 0 does not respect `set_publish(false)`

matthewmazzanti, 2021-09-13:
### The problem

I am attempting to build a "candle flicker" effect via the light effect lambda. This involves reading the brightness set via Home Assistant (the `remote_values` of the light class), and setting a transition via `set_transition_length` to smooth things out. As a part of the effect, I don't want the published light value constantly updating, instead I would like the effect to take into account the value set in Home Assistant.

It seems the call to `set_transition_length(100)` (or any value > 0) causes the light to update its `remote_values` field without updating Home Assistant. It would be preferred if `set_transition_length` respected the `set_publish(false)` value.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

light

### Example YAML snippet

```yaml
output:
  - platform: esp8266_pwm
    pin: D2
    frequency: 120HZ
    id: J3_pwm

light:
  - platform: monochromatic
    output: J3_pwm
    name: J3
    id: J3
    effects:
      - lambda:
          name: Broken
          update_interval: 100ms
          lambda: |-
            auto call = id(J4)->turn_on();
            float remote = id(J4)->remote_values.get_brightness();
            esph_log_d(__FILE__, "brightness: %f", remote);
            call.set_brightness(remote * 0.5);
            call.set_transition_length(100);
            call.set_publish(false);
            call.set_save(false);
            call.set_state(true);
            call.perform();

The output of the above is:
```
[23:57:02][D][candle.yaml:076]: brightness: 1.000000
[23:57:02][D][candle.yaml:076]: brightness: 0.500000
[23:57:03][D][candle.yaml:076]: brightness: 0.250000
[23:57:03][D][candle.yaml:076]: brightness: 0.125000
[23:57:03][D][candle.yaml:076]: brightness: 0.062500
[23:57:03][D][candle.yaml:076]: brightness: 0.031250
[23:57:03][D][candle.yaml:076]: brightness: 0.015625
[23:57:03][D][candle.yaml:076]: brightness: 0.007812
...
```

Changing the call to `set_transition_length(0)` gives the correct output: 
```
[00:09:44][D][candle.yaml:076]: brightness: 1.000000
[00:09:44][D][candle.yaml:076]: brightness: 1.000000
[00:09:44][D][candle.yaml:076]: brightness: 1.000000
[00:09:44][D][candle.yaml:076]: brightness: 1.000000
...
```
```

### Anything in the logs that might be useful for us?

```txt
Looking into the history of this it seem that (these changes)[https://github.com/esphome/esphome/pull/2124/files#diff-e96397dff2fe7d303bdd0cd720a8b22085dfb8860d7dacde26d01a54f7dc3e8fR220-R235] are the root cause. In particular, the reads and updates of `state->current_values` have been changed to `state->remote_values`. This is done without an actual API call, which I think is also an issue - but not one I'm focused on right now.
```

oxan, 2021-09-15:
> Looking into the history of this it seem that these changes are the root cause. In particular, the reads and updates of state->current_values have been changed to state->remote_values.

I'm not convinced this diagnosis is correct. The only change from `current_values` to `remote_values` there is in `start_flash_`, which you aren't using here.

matthewmazzanti, 2021-09-15:
You are entirely right - this diagnosis is wrong. That was the first time I'd looked at the API and didn't quite understand how `start_flash_` and `start_transition_` played into the `LightCall` and how it all tied together. I dug a bit deeper to understand things and made a PR [here](https://github.com/esphome/esphome/pull/2313) that addresses (what I think is) the real issue: calling `set_publish(false)` should prevent `remote_values` from being set at all, which is done in the `start_transition_` and `start_flash_` functions.

closed 2021-09-15 by jesserockz