esphome/issues #2419: Wifi disconnect results in exception crash

kristjanbjarni, 2021-09-13:
### The problem

If I reboot the wireless access point then all my sonoff switches crash when reconnecting to the access point.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

sonoff basic

### Component causing the issue

wifi

### Example YAML snippet

```yaml
substitutions:
  devicename: sbasic_2
  
esphome:
  name: ${devicename}
  comment: Test
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: -100
    then:
      - switch.turn_on: automation
wifi:
  reboot_timeout: 0s
  networks:
  - ssid: "Skynet"
    password: !secret wifi
  ap:
    ssid: "${devicename}_fallback"
    password: !secret wifi

captive_portal:

logger:

api:
  reboot_timeout: 0s

ota:

binary_sensor:
  - platform: gpio
    id: push_button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    internal: true
    on_press:
[... 173 lines cut ...]
[20:09:35][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[20:09:35][C][switch.gpio:048]: GPIO Switch 'Sonoff sbasic_2 Relay'
[20:09:35][C][switch.gpio:049]:   Pin: GPIO12 (Mode: OUTPUT)
[20:09:35][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[20:09:35][C][switch.gpio:048]: GPIO Switch 'blue_led'
[20:09:35][C][switch.gpio:049]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[20:09:35][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[20:09:35][C][template.switch:058]: Template Switch 'Sonoff sbasic_2 Manual'
[20:09:35][C][template.switch:059]:   Restore State: NO
[20:09:35][C][template.switch:060]:   Optimistic: NO
[20:09:35][C][template.switch:058]: Template Switch 'Sonoff sbasic_2 Automation'
[20:09:35][C][template.switch:059]:   Restore State: NO
[20:09:35][C][template.switch:060]:   Optimistic: NO
[20:09:35][C][logger:189]: Logger:
[20:09:35][C][logger:190]:   Level: DEBUG
[20:09:35][C][logger:191]:   Log Baud Rate: 115200
[20:09:35][C][logger:192]:   Hardware UART: UART0
[20:09:35][C][captive_portal:148]: Captive Portal:
[20:09:35][C][ota:029]: Over-The-Air Updates:
[20:09:35][C][ota:030]:   Address: sbasic_2.local:8266
[20:09:35][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
[20:09:35][C][api:095]: API Server:
[20:09:35][C][api:096]:   Address: sbasic_2.local:6053
[20:09:37][D][api.connection:730]: Client 'Home Assistant 2021.9.6 (192.168.3.60)' connected successfully!
```

bkaufx, 2021-09-13:
I have been playing with this same problem today.  Finally got it narrowed down to the access point / captive portal.

I have had yaml as simple as the following and it crashes every time wifi disconnects and then reconnects.  It also crashes when you first enter wifi credentials in the captive portal.  If I comment out the ap: and captive_portal: lines then there is no crash.

I've also tried various yaml and hardware configurations and it seems to clearly come down to the ap/captive portal.  I have yet to be able to get captive portal to work without crashing under any circumstance.

```
esphome:
  name: tester
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: faketester
  password: asdfasdfasdfasdf

  output_power: 17

  ap:
    ssid: "tester"

captive_portal:

logger:
api:
debug:
ota:
    password: !secret esphome_pass
    id: my_ota

```

bkaufx, 2021-09-14:
I get an exception (9) instead of (28), but the crash happens at the exact same time and the logs look the same otherwise.

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x402160fd epc2=0x00000000 epc3=0x00000000 excvaddr=0x6944203e depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd50 end: 3fffffc0 offset: 0190
3ffffee0:  3fff3a7c 3fffff14 3ffe8344 4022ada0
3ffffef0:  3fff3a7c 3fff10c4 3fff2404 4020a664
3fffff00:  00000000 00000102 3fff000c 402217b8
3fffff10:  3fff3a80 00000102 3fff000c 4020558d
3fffff20:  3fff000c 000d4016 401008ae 402055a0
3fffff30:  00000000 00000002 3fff000c 40227464
3fffff40:  3fff000c 00000102 3fff000c 40227509
3fffff50:  3fff3a80 00000102 3fff000c 4020f70f
3fffff60:  007a1200 bc8ffb77 feefef00 feefeffe
3fffff70:  3fff3a94 3fff19a4 00000020 3fff21a4
3fffff80:  00000000 00000000 00000001 3fff041c
3fffff90:  3fffdad0 00000000 3fff03dc 40212678
3fffffa0:  3fffdad0 00000000 3fff03dc 402245c4
3fffffb0:  feefeffe feefeffe 3ffe858c 40100725
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

oxan, 2021-09-14:
Can you obtain the crash logs through ESPHome? That should decode the addresses and give more information about what's going on.

bkaufx, 2021-09-14:
How do I get crash logs?

Apparently this has been a problem for a while:  https://github.com/esphome/issues/issues/990.

bkaufx, 2021-09-14:
Okay, I figured out to run esphome on the same computer where I compiled and output logs over USB, and got this stack trace with some decoding.  I think that is what you meant.

```
[15:44:34]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:44:34]
[15:44:34]Exception (28):
←[33mWARNING Exception type: Access to invalid address: LOAD (wild pointer?)←[0m
[15:44:34]epc1=0x4020d01d epc2=0x00000000 epc3=0x00000000 excvaddr=0x06400044 depc=0x00000000
←[33mWARNING Decoded 0x4020d01d: UdpContext::next()←[0m
[15:44:34]
[15:44:34]>>>stack>>>
←[33mWARNING Found stack trace! Trying to decode it←[0m
[15:44:34]
[15:44:34]ctx: cont
[15:44:34]sp: 3ffffd50 end: 3fffffc0 offset: 0190
[15:44:34]3ffffee0:  00000000 4bc6a7f0 c45a1cac 00000000
[15:44:34]3ffffef0:  00000000 3fff1d6c 4bc6a7f0 00000000
[15:44:34]3fffff00:  3fff118c 00000002 3ffef4f8 402178c0
←[33mWARNING Decoded 0x402178c0: DNSServer::processNextRequest()←[0m
[15:44:34]3fffff10:  3fff0210 00000002 3ffef4f8 40203df9
←[33mWARNING Decoded 0x40203df9: esphome::captive_portal::CaptivePortal::loop()←[0m
[15:44:34]3fffff20:  234822db 001e8393 3fff0a14 40203e0c
←[33mWARNING Decoded 0x40203e0c: non-virtual thunk to esphome::captive_portal::CaptivePortal::loop()←[0m
[15:44:34]3fffff30:  001e8393 00000000 3ffef4f8 4021ca44
←[33mWARNING Decoded 0x4021ca44: esphome::Component::call_loop()←[0m
[15:44:34]3fffff40:  3ffef4f8 00000002 3ffef4f8 4021cac9
←[33mWARNING Decoded 0x4021cac9: esphome::Component::call()←[0m
[15:44:34]3fffff50:  3fff0210 00000002 3ffef4f8 40208d27
←[33mWARNING Decoded 0x40208d27: esphome::Application::loop()←[0m
[15:44:34]3fffff60:  007a1200 596a96d7 00000000 4021e110
←[33mWARNING Decoded 0x4021e110: precache←[0m
[15:44:34]3fffff70:  3fff0220 3fff191c 00000020 00000000
[15:44:34]3fffff80:  00000000 00000000 00000001 3ffef834
[15:44:34]3fffff90:  3fffdad0 00000000 3ffef7f4 4020b298
←[33mWARNING Decoded 0x4020b298: loop←[0m
[15:44:34]3fffffa0:  3fffdad0 00000000 3ffef7f4 4021a6a4
←[33mWARNING Decoded 0x4021a6a4: loop_wrapper() at core_esp8266_main.cpp←[0m
[15:44:34]3fffffb0:  feefeffe feefeffe 3ffe84f4 401005c1
←[33mWARNING Decoded 0x401005c1: cont_wrapper←[0m
[15:44:34]<<<stack<<<
[15:44:34]
[15:44:34]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

bkaufx, 2021-09-14:
Here is another one.  So far I am seeing one of these two stack traces every time I turn wifi back on while the ap / captive portal is running.  This one seems to be more popular, and then I will randomly get the previous one I posted now and then.

```
[16:16:41]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[16:16:41]
[16:16:41]Exception (28):
←[33mWARNING Exception type: Access to invalid address: LOAD (wild pointer?)←[0m
[16:16:41]epc1=0x4021cab4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0401a8e8 depc=0x00000000
←[33mWARNING Decoded 0x4021cab4: esphome::Component::call()←[0m
[16:16:41]
[16:16:41]>>>stack>>>
←[33mWARNING Found stack trace! Trying to decode it←[0m
[16:16:41]
[16:16:41]ctx: cont
[16:16:41]sp: 3ffffdb0 end: 3fffffc0 offset: 0190
[16:16:41]3fffff40:  3ffef4f8 00000002 3ffef4f8 401001f6
←[33mWARNING Decoded 0x401001f6: esphome::Application::feed_wdt()←[0m
[16:16:41]3fffff50:  3fff1628 00000002 3ffef4f8 40208d27
←[33mWARNING Decoded 0x40208d27: esphome::Application::loop()←[0m
[16:16:41]3fffff60:  007a1200 52e4e33f 00000000 4021e110
←[33mWARNING Decoded 0x4021e110: precache←[0m
[16:16:41]3fffff70:  3fff1638 3fff15fc 00000020 00000000
[16:16:41]3fffff80:  00000000 00000000 00000001 3ffef834
[16:16:41]3fffff90:  3fffdad0 00000000 3ffef7f4 4020b29c
←[33mWARNING Decoded 0x4020b29c: loop←[0m
[16:16:41]3fffffa0:  3fffdad0 00000000 3ffef7f4 4021a6a8
←[33mWARNING Decoded 0x4021a6a8: loop_wrapper() at core_esp8266_main.cpp←[0m
[16:16:41]3fffffb0:  feefeffe feefeffe 3ffe84f4 401005c1
←[33mWARNING Decoded 0x401005c1: cont_wrapper←[0m
[16:16:41]<<<stack<<<
[16:16:41]
[16:16:41]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

bkaufx, 2021-09-15:
Here is a run with very verbose logging if that helps.  In another run of the exact same firmware, the exception occurs right after `mode : sta(3c:61:05:ec:71:66)`.  The next two line were not printed.

```
[19:01:18][D][wifi:522]: Disabling AP...
[19:01:18][V][wifi_esp8266:064]: Disabling AP.
[19:01:18]bcn 0
[19:01:18]del if1
[19:01:18]pm open,type:0 0
[19:01:18]mode : sta(3c:61:05:ec:71:66)
[19:01:18][V][wifi_esp8266:502]: Event: Changed Mode old=AP+STA new=STA
[19:01:18]wifi evt: 8
[19:01:20]Fatal exception 9(LoadStoreAlignmentCause):
[19:01:20]epc1=0x4021e904, epc2=0x00000000, epc3=0x00000000, excvaddr=0xa5a5a5cd, depc=0x00000000
←[32mINFO Need to fetch platformio IDE-data, please stand by←[0m
←[32mINFO Running:  platformio run -d tester -t idedata←[0m
←[33mWARNING Decoded 0x4021e904: esphome::Component::call()←[0m
[19:01:22]
[19:01:22]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[19:01:22]
[19:01:22]Exception (9):
←[33mWARNING Exception type: LoadStoreAlignmentCause: Load or store to an unaligned address←[0m
[19:01:22]epc1=0x4021e904 epc2=0x00000000 epc3=0x00000000 excvaddr=0xa5a5a5cd depc=0x00000000
←[33mWARNING Decoded 0x4021e904: esphome::Component::call()←[0m
[19:01:22]
[19:01:22]>>>stack>>>
←[33mWARNING Found stack trace! Trying to decode it←[0m
[19:01:22]
[19:01:22]ctx: cont
[19:01:22]sp: 3ffffdb0 end: 3fffffc0 offset: 0190
[19:01:22]3fffff40:  3fff0e38 00000002 3fff0e38 401001f6
←[33mWARNING Decoded 0x401001f6: esphome::Application::feed_wdt()←[0m
[19:01:22]3fffff50:  3fff3358 00000002 3fff0e38 4020a6d9
←[33mWARNING Decoded 0x4020a6d9: esphome::Application::loop()←[0m
[19:01:22]3fffff60:  007a1200 ec725a61 00000000 40220000
←[33mWARNING Decoded 0x40220000: precache←[0m
[19:01:22]3fffff70:  3fff2458 3fff3368 00000020 00000000
[19:01:22]3fffff80:  00000000 00000000 00000001 3fff1174
[19:01:22]3fffff90:  3fffdad0 00000000 3fff1134 4020cda8
←[33mWARNING Decoded 0x4020cda8: loop←[0m
[19:01:22]3fffffa0:  3fffdad0 00000000 3fff1134 4021c6f4
←[33mWARNING Decoded 0x4021c6f4: loop_wrapper() at core_esp8266_main.cpp←[0m
[19:01:22]3fffffb0:  feefeffe feefeffe 3ffe84f4 401005fd
←[33mWARNING Decoded 0x401005fd: cont_wrapper←[0m
[19:01:22]<<<stack<<<
[19:01:22]
[19:01:22]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

oxan, 2021-09-15:
Hmm, at a first look those call traces don't make much sense to me.

bkaufx, 2021-09-15:
The fact that it can have an exception at different points in the very verbose output makes me think maybe it's not directly caused by whatever code is running but by something indirect like running out of memory.  Also, the exceptions are LoadStoreAlignmentCause and invalid address, which it seems like would be hard to do for compiled code normally.

What other types of debugging can I do?  I tried to run this heap size PR (https://github.com/esphome/esphome/pull/1578) to see if free heap size craters right before it crashes, but I'm having trouble with compile errors.  I'm not sure what else I could be looking at.  Have you ever tried using the captive portal?  It seems to me that it always crashes no matter what when WiFi connects.  I've never seen it not crash.

bkaufx, 2021-09-16:
Just upgraded to 2021.9.0.  Making it a little further without crashing now...

```
[19:11:13][D][wifi:524]: Disabling AP...
[19:11:13][V][wifi_esp8266:062]: Disabling AP.
[19:11:13]bcn 0
[19:11:13]del if1
[19:11:13]pm open,type:0 0
[19:11:13]mode : sta(3c:61:05:ec:71:66)
[19:11:13][V][component:207]: Component wifi took a long time for an operation (0.06 s).
[19:11:13][V][component:208]: Components should block for at most 20-30ms.
[19:11:13][V][wifi_esp8266:515]: Event: Changed Mode old=AP+STA new=STA
[19:11:13]wifi evt: 8
[19:11:13][D][api:095]: Accepted 192.168.1.2
[19:11:13][D][api:095]: Accepted 192.168.1.2
[19:11:13][D][api:095]: Accepted 192.168.1.2
[19:11:13][VV][api.service:291]: on_hello_request: HelloRequest {
[19:11:13]  client_info: 'Home Assistant 2021.9.0'
[19:11:13]}
[19:11:13][V][api.connection:719]: Hello from client: 'Home Assistant 2021.9.0 (192.168.1.2)'
[19:11:13][VV][api.service:013]: send_hello_response: HelloResponse {
[19:11:13]  api_version_major: 1
[19:11:13]  api_version_minor: 6
[19:11:13]  server_info: 'tester (esphome v2021.9.0)'
[19:11:13]}
[19:11:13][VV][api.service:291]: on_hello_request: HelloRequest {
[19:11:13]  client_info: 'Home Assistant 2021.9.0'
[19:11:13]}
[19:11:13][V][api.connection:719]: Hello from client: 'Home Assistant 2021.9.0 (192.168.1.2)'
[19:11:13][VV][api.service:013]: send_hello_response: HelloResponse {
[19:11:13]  api_version_major: 1
[19:11:13]  api_version_minor: 6
[19:11:13]  server_info: 'tester (esphome v2021.9.0)'
[19:11:13]}
[19:11:13][VV][api.service:291]: on_hello_request: HelloRequest {
[19:11:13]  client_info: 'Home Assistant 2021.9.0'
[19:11:13]}
[19:11:13][V][api.connection:719]: Hello from client: 'Home Assistant 2021.9.0 (192.168.1.2)'
[19:11:13][VV][api.service:013]: send_hello_response: HelloResponse {
[19:11:13]  api_version_major: 1
[19:11:13]  api_version_minor: 6
[19:11:13]  server_info: 'tester (esphome v2021.9.0)'
[19:11:13]}
[19:11:13][V][component:207]: Component api took a long time for an operation (0.11 s).
[19:11:13][V][component:208]: Components should block for at most 20-30ms.
[19:11:13]Fatal exception 28(LoadProhibitedCause):
[19:11:13]epc1=0x4021fdf0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000028, depc=0x00000000
←[32mINFO Need to fetch platformio IDE-data, please stand by←[0m
←[32mINFO Running:  platformio run -d tester -t idedata←[0m
←[33mWARNING Decoded 0x4021fdf0: esphome::Component::call()←[0m
[19:11:14]
[19:11:14]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[19:11:14]
[19:11:14]Exception (28):
←[33mWARNING Exception type: Access to invalid address: LOAD (wild pointer?)←[0m
[19:11:14]epc1=0x4021fdf0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000028 depc=0x00000000
←[33mWARNING Decoded 0x4021fdf0: esphome::Component::call()←[0m
[19:11:14]
[19:11:14]>>>stack>>>
←[33mWARNING Found stack trace! Trying to decode it←[0m
[19:11:14]
[19:11:14]ctx: cont
[19:11:14]sp: 3ffffdb0 end: 3fffffc0 offset: 0190
[19:11:14]3fffff40:  3fff30c0 00000002 3fffff60 4020cef9
←[33mWARNING Decoded 0x4020cef9: esphome::WarnIfComponentBlockingGuard::WarnIfComponentBlockingGuard(esphome::Component*)←[0m
[19:11:14]3fffff50:  3fff30c0 00000002 3fff0a48 4020b93c
←[33mWARNING Decoded 0x4020b93c: esphome::Application::loop()←[0m
[19:11:14]3fffff60:  0003e607 3fff2070 00000000 40221590
←[33mWARNING Decoded 0x40221590: AsyncServer::setNoDelay(bool)←[0m
[19:11:14]3fffff70:  3fff30d0 3fff42bc 00000020 00000000
[19:11:14]3fffff80:  00000000 00000000 00000001 3fff0d84
[19:11:14]3fffff90:  3fffdad0 00000000 3fff0d44 4020e0c4
←[33mWARNING Decoded 0x4020e0c4: loop←[0m
[19:11:14]3fffffa0:  3fffdad0 00000000 3fff0d44 4021ced0
←[33mWARNING Decoded 0x4021ced0: loop_wrapper() at core_esp8266_main.cpp←[0m
[19:11:14]3fffffb0:  feefeffe feefeffe 3ffe84f4 4010061d
←[33mWARNING Decoded 0x4010061d: cont_wrapper←[0m
[19:11:14]<<<stack<<<
[19:11:14]
[19:11:14]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

bkaufx, 2021-09-16:
I commented out the line in wifi_component.cpp that disables the AP, and the device never crashes.  Not really a valid solution but just a data point for anyone who might be able to help figure this out.  I haven't figured out a good way to debug within the wifi_mode_ call because it seems like it's calling a precompiled library from platformio or something.  I couldn't find editable source code.

The commented out line of code is `this->wifi_mode_({}, false);` around line 526 of wifi_component.cpp.

oxan, 2021-09-19:
>  I haven't figured out a good way to debug within the wifi_mode_ call because it seems like it's calling a precompiled library from platformio or something.

The `wifi_mode_` function is implemented in `wifi_component_esp8266.cpp`.

bkaufx, 2021-09-19:
Yea.  Within that it looks like it just calls `wifi_set_opmode_current` from platformio.  I think that is what I need to debug.  Like, try to figure out at which step it freezes.  I haven't been able to find that function.

OttoWinter, 2021-10-08:
> So this is must be a bug in the ESP8266 NONOS SDK. How can this bug be reported there to get it fixed in the original source?

NonOS SDK is afaik not maintained much anymore, they want you to switch to RTOS SDK, but nonos is what's bundled by [esp8266/arduino](https://github.com/esp8266/Arduino/) so we can't switch.

> This is a pretty serious bug resulting in a kernel crash whenever there is any wifi outage.

I mean it could also be just that we access the SDK in a way that is not supported or something - doesn't necessarily have to be a bug in the SDK (and it's a bit less of a kernel and more just an SDK, especially considering the SDK runs in the same address space and there are no such things as processes)