esphome/issues #2420: ESP32-c3-devkitm-1 boot issues

markusg1234, 2021-09-14:
### The problem

I have found that a ESP32-C3-devkitm-1 no longer boots when I compile new firmware where it was working previously.

Error seen below and it will just keeps outputting these errors.

Regards,

Markus

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP32

### Board

ESP32-c3-devkitm-1

### Component causing the issue

firmware.bin

### Example YAML snippet

```yaml
esphome:
  name: test
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master
  
output:
  - id: relay_1
    platform: gpio
    pin: GPIO5
    inverted: true
```

### Anything in the logs that might be useful for us?

```txt
[13:05:36]invalid headeESP-ROM:esp32c3-api1-20210207
[13:05:36]Build:Feb  7 2021
[13:05:36]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[13:05:36]Saved PC:0x4004d1f8
[13:05:36]invalid header: 0xffffffff
[13:05:36]invalid header: 0xffffffff
```

jesserockz, 2021-09-14:
I dont have that board, I have the `ESP-C3-01M-KIT` devboard, and this works (in addition to what you have):

```yaml
esphome:
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB
```

markusg1234, 2021-09-14:
Thanks, 

I tried and same issue with same test yaml.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html

INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.92ce408 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-riscv32-esp 8.4.0+2021r1
Dependency Graph
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
Compiling /data/test/.pioenvs/test/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/output/automation.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/output/float_output.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/application.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/application_esp32.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/application_esp8266.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/color.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/component.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/controller.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/esphal.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/helpers.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/log.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/preferences.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/scheduler.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/util.cpp.o
Compiling /data/test/.pioenvs/test/src/main.cpp.o
Generating partitions /data/test/.pioenvs/test/partitions.bin
Compiling /data/test/.pioenvs/test/liba6f/WiFi/ETH.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFi.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiAP.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiClient.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiGeneric.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiMulti.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiSTA.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiScan.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiServer.cpp.o
Compiling /data/test/.pioenvs/test/liba6f/WiFi/WiFiUdp.cpp.o
Compiling /data/test/.pioenvs/test/libda9/ESPmDNS/ESPmDNS.cpp.o
Archiving /data/test/.pioenvs/test/libFrameworkArduinoVariant.a
Compiling /data/test/.pioenvs/test/FrameworkArduino/Esp.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/FirmwareMSC.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/HWCDC.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/HardwareSerial.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/IPAddress.cpp.o
Archiving /data/test/.pioenvs/test/liba6f/libWiFi.a
Compiling /data/test/.pioenvs/test/FrameworkArduino/IPv6Address.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/MD5Builder.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/Print.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/Stream.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/StreamString.cpp.o
Archiving /data/test/.pioenvs/test/libda9/libESPmDNS.a
Compiling /data/test/.pioenvs/test/FrameworkArduino/USB.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/USBCDC.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/USBMSC.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/WMath.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/WString.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/base64.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/cbuf.cpp.o
Compiling /data/test/.pioenvs/test/FrameworkArduino/esp32-hal-adc.c.o
[... 71 lines cut ...]
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 512.0 kbit/s)...
Hash of data verified.
Compressed 232720 bytes to 113465...
Writing at 0x00010000... (14 %)
Writing at 0x0001a4f7... (28 %)
Writing at 0x00021016... (42 %)
Writing at 0x0002893f... (57 %)
Writing at 0x0002f2e0... (71 %)
Writing at 0x00036a60... (85 %)
Writing at 0x00041eba... (100 %)
Wrote 232720 bytes (113465 compressed) at 0x00010000 in 3.9 seconds (effective 474.0 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[15:27:49] header: 0xffffffff
[15:27:49]invalid header: 0xffffffff
[15:27:49]invalid header: 0xffffffff
[15:27:49]invalid header: 0xffffffff

markusg1234, 2021-09-14:
Hi,
I have now got it working by doing the download directly from the HA server.

In the past I have used PlatformIO IDE with the C3 as I couldn't get it to download in the past from a direct connection.

Thanks for the info on dio.

Regards,

Markus

closed 2021-09-14 by markusg1234