esphome/issues #2421: BME680 stops working

ikomhoog, 2021-09-14:
### The problem

my BME680 sometimes stops working, the BH1750 that is on the exact same wires keeps working however.
Their pins on these sensors are exact reverse so the wires and sda-scl resistor can sit between the sensors on the breadboard.
The issue is resolved when I unplug the USB and plug it back in to restart the ESP.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.5, but also had this while esphome was running in docker

### What platform are you using?

ESP32

### Board

nodemcu-32s (it states ESP32-WROOM-32 on the chip)

### Component causing the issue

I2C

### Example YAML snippet

```yaml
esphome:
  name: bme680
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bme680 Fallback Hotspot"
    password: ""

captive_portal:

i2c:
  scl: 33
  sda: 13
  scan: true
  frequency: 100000
  
bme680_bsec:

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

  - platform: bh1750
    name: "BH1750 Illuminance"
    address: 0x23
    measurement_duration: 69
    update_interval: 1s

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"
```

### Additional information

Here is my home assistant history graph of the IAQ where you can see the missing pieces:
[image]

Warter21, 2021-09-17:
Does it work on 0x76 or 0x77? 

bme680 default address is 0x76 but for me it works only on 0x77. 

Can you check this option?

bme680_bsec:
    address: 0x77

ikomhoog, 2021-09-17:
This is not a problem when it boots, it gets recognized fine at boot with address 0x76 but it just stops after some time.

```
[21:46:11][I][i2c:033]: Scanning i2c bus for active devices...
[21:46:11][I][i2c:040]: Found i2c device at address 0x23
[21:46:12][I][i2c:040]: Found i2c device at address 0x76
```
The device at 0x23 is the BH1750

The BME680 has address 0x76 when the address pin is connected to GND and when it is connected to VCC it will be 0x77.
I also had the issue that it would only work at 0x77 but that was because I didn't connect the address pin which caused it to float.

Since I connected every pin correctly it works every boot on address 0x76.

rhaamo, 2021-12-07:
Having issues with a BME680 since some time too, only getting:
```
[20:08:49][W][bme680_bsec.sensor:297]: BSEC failed to process signals (BSEC Error Code -2)
[20:08:52][W][bme680_bsec.sensor:297]: BSEC failed to process signals (BSEC Error Code -2)
```

BME is detected:
```
[20:08:47][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[20:08:47][I][i2c.arduino:053]: Found i2c device at address 0x76
```

and using:
```
i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a

bme680_bsec:
  address: 0x76
  iaq_mode: static
  sample_rate: ulp
  state_save_interval: 6h
```

versions from the logs:
```
[20:10:50][I][app:099]: ESPHome version 2021.11.4 compiled on Dec  7 2021, 20:08:09
[20:10:50][C][bme680_bsec.sensor:124]: BME680 via BSEC:
[20:10:50][C][bme680_bsec.sensor:129]:   BSEC Version: 1.4.8.0
```

tried switching it to 0x77 instead of 0x76 and same result

ikomhoog, 2021-12-08:
@rhaamo I forgot this was still an open issue, it's actually solved for me.
Please try my setup to see if that fixes your issue:

This is my current config which works:
```
i2c:
  scl: 33
  sda: 13
  scan: true
  frequency: 100000
  
bme680_bsec:
 temperature_offset: 4

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

  - platform: bh1750
    name: "BH1750 Illuminance"
    address: 0x23
    measurement_duration: 69
    update_interval: 1s

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"
```
These are my connections:
```
BME VCC -> ESP 3V3
BME GND -> ESP GND
BME SCL -> ESP G33
BME SDA -> ESP G13
BME SDO -> BME GND
BME CS -> BME VCC
And there is a 10K resistor between the SDA and SCL pins.
```

I have weird SDA/SCL pins because I was testing and am not changing working pins.

rhaamo, 2021-12-08:
just tried to add the frequency part to i2c and nothing better.
I can try changing the pins or adding a 10k res later.

it's weird that it stopped working for no reasons, the thing is just sitting on a shelf and I never touched it since I made it

rhaamo, 2021-12-15:
tried 10k, changing pins, then 10k, no success, sensor is detected on i2c bus but that's all.

ikomhoog, 2021-12-16:
have you soldered the pins to it correctly?
I had another sensor that I just tested and gave similar issues when the pins weren't soldered correctly.
After resoldering with quality solder it connected successfully and hasn't failed since.

lolorc, 2022-03-03:
I also have these issues with 2 bme680. every 2 weeks or so I get the "Failed to set sensor settings (BME680 Error Code -2)" messages for no apparent reason.

after a soft restart of the esp32 => it doesn't detect the bme680.
I have to electrically restart both the esp and the sensor.

I thought I could power the bme680 from an always on gpio so I could electrically restart the bme680 then reboot the esp32, but it doesn't work. it looks like i2c gets started first, the bme680 is not yet powered...

is there a way to programmatically restart i2c ?

I'm looking for a solution as my heater is controlled by a climate component with a bme680 as main sensor :-) (=> heater stops heating or always on when the issue occurs...)

iankaufmann, 2022-03-03:
Chiming in here to say that I am having the same issue.

The BME680 works for about 24 hours, and then it just stops.  If you reboot ESPHome, it says "Communication with BME680 failed!" when it comes back up.  The only thing that fixes it is to completely unplug the ESP for 10 seconds or so and plug it back in.  Then it works for another day.

I have replaced every component involved too... including a 2nd BME680 sensor.  I have replaced the breadboard, the ESP8266, all jumper cables, and the sensor itself, with the same result.

I've checked the soldering on both BME sensors, and tried the other suggestions from this thread such as adding the 10k resistor, changing the address, and the other config changes.

I'm in the same boat as @lolorc where I'm using it to regulate a heater and I've had to add a zigbee based sensor as a fallback for when this one stops working.  The trouble with that is how slowly it updates vs the BME when it is working.

I am using this integration:
https://esphome.io/components/sensor/bme680.html

Since I have ruled out all components involved, it makes me wonder if this is a software issue.  There was a time when it wasn't like this, but it has been doing this for the past several months.

lolorc, 2022-03-04:
I'm using this one : https://esphome.io/components/sensor/bme680_bsec.html and it's not crashing that often.
I'm going to add a ds18b20 as a fallback. (been using a ble sensor before, but I have issues with ble at the moment https://github.com/esphome/issues/issues/3039)

lolorc, 2022-03-15:
in the end my "climate" esp32 has sps30, scd30 and one bme680 on its own i2c bus, no need for a 1wire sensor since I already have two temperature sensors(bme680+scd30) but bme680 is still crashing so I'm using sensor templates to workaround the issue:
- every time the bme680 temperature sensor gets a value I store uptime in a "last seen" template sensor
```
  - platform: bme680_bsec
    temperature:
      id: "esp32_01_BME680_Temperature"
      name: "esp32_01_BME680_Temperature"
      accuracy_decimals: 5
      on_value:
        then:
          - sensor.template.publish:
              id: esp32_01_bme680_temperature_last
              state: !lambda 'return id(esp32_01_uptime).state;'
```
- I also have a template sensor averaging between bme680 temp & scd30 temp when the bme680 sensor is still alive, otherwise it only returns the scd30 temp
```
  - platform: template
    id: esp32_01_tmplt_01_temperature
    name: esp32_01_tmplt_01_temperature
    update_interval: 5s
    accuracy_decimals: 5
    unit_of_measurement: °C
    lambda: |-
      if ( id(esp32_01_uptime).state - id(esp32_01_bme680_temperature_last).state > 120 ) {
        ESP_LOGI("main", "BME680 SENSOR GONE, temp: %F", id(esp32_01_Workshop_SCD30_Temperature).state);
        return id(esp32_01_Workshop_SCD30_Temperature).state;
      } else {
        return (id(esp32_01_BME680_Temperature).state + id(esp32_01_Workshop_SCD30_Temperature).state)/2;
      }
```
 - climate uses this template sensor

martijnbuts, 2022-05-28:
Is there any solution known for this problem? I experience the same problem that the ESP32 loses the BME680.

davewongillies, 2022-11-26:
Not sure if this will help anyone, but what fixed it for me was grounding the CS pin of the BME680 to solve this problem even though its configured to communicate with I2C.

iankaufmann, 2022-12-01:
> Not sure if this will help anyone, but what fixed it for me was grounding the CS pin of the BME680 to solve this problem even though its configured to communicate with I2C.

Thanks for posting this!  I had given up on this sensor, but grounding the CS pin caused it to hold steady for 5 days now.

thecheatah, 2023-04-17:
I wanted to confirm that setting the CS pin to ground fixed the issue. This issue was very helpful in finding the fix.

This issue should be closed.

orenmazor, 2023-06-21:
another confirmation here.

the datasheet for this module also specifies this for SDO, which I did as well (because CS alone did not do it for me):

[image]

robinostlund, 2023-07-06:
I also have the same issue, but for me it seems to happen when the humidity goes to high like when it is raining alot. Are your sensors in a climate with high humidity?

orenmazor, 2023-07-06:
> I also have the same issue, but for me it seems to happen when the humidity goes to high like when it is raining alot. Are your sensors in a climate with high humidity?

50-70% here

LordNex, 2023-11-14:
Im having ths same issue. I;ve tried connecting it via SPI and I2C and all come up that no ic2 devices are found and not results are displayed. Im not soldering yet, just running the pin headers into the breadboard and then us individual wires to take care of the connections to the senor. I thought maybe it was a bad sensor so I grabbed a new BME680 and still have the same issue. At thyis point im thinking of trying a different ESP32. No other sensors have been added yet, im just starting this build so everything is basic. Any suggestions?

robinostlund, 2023-11-24:
I also have issues, for me though the issue is that bme680 is changing the address from 0x76 to 0x77 and vise versa after a couple of days, sometimes it is working for months.

sokarovski, 2023-12-02:
Hey there, thanks to this issue and all the comments in it **grounding the CS pin helped me** solve my problem.

Long story: 
For me, this problem only started happening after I added a 6m cable to the I2C bus. I knew that this was a bad idea/practice as I2C is not designed for such a thing. My goal was to have my heating system working on a single ESP32 so it wouldn't be dependent on HA and reading values from other sensors via HA. In case of HA failure i would have I2C display and physical button to control everything. Note that using a short 10-20cm line, I don't have this problem.

I tried changing the speed to 10kHz and 5kHz, but that did NOT fix the problem.

Inspired by @lolorc, I also added an AHT21 as a backup which does work a lot more stable. Since then i decided to use it as main sensor and use the BME as backup. 

It would be nice though if the BME680 component at least reported NaN if the reading is not successful so we can setup HA automations/notifications.

mastersf, 2024-04-23:
grounding the pin hasn't worked for me. Anyone else seeing major inconsistency with this ?

hosh0r, 2024-07-25:
> Not sure if this will help anyone, but what fixed it for me was grounding the CS pin of the BME680 to solve this problem even though its configured to communicate with I2C.

I was facing the same issues, but since the CS pin is grounded as described, my bme680 works super stable with an wemos d1 mini. Thanks for the hint!