esphome/issues #2423: nextion tft_url ota update - error

adamcodemaker, 2021-09-15:
### The problem

I have this display https://nextion.ca/datasheets/nx3224k024_011/
I connected TX form display to RX on board (GPIO3) and RX from display to TX on board (GPIO1)
When I try to update using tft_url it fails.
It starts showing 99%
[image]
and this is strange because tft file is ~455KB
Then after 2min it goes to 100% and the display restarts with error
[image]
[image]
[image]
Upload works ok from Nextion Editor 1.63.3 using USB-UART PL2303 TA
It also works using SD card.
Communication with display is ok. I can read humidity from mqtt topic and display it.
I tried different board (esp32doit-devkit-v1) but with the same result.

### Which version of ESPHome has the issue?

esphome:2021.8.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

0.117.0

### What platform are you using?

ESP8266

### Board

nodemcu ESP8266 ESP-12E aka Waveshare 16608 or Amica

### Component causing the issue

display/nextion

### Example YAML snippet

```yaml
substitutions:
  ip_address: 192.168.0.62
esphome:
  name: bedroom-drawer
  platform: ESP8266
  board: nodemcuv2
  arduino_version: platformio/espressif8266@2.6.2
  build_path: bedroom-drawer
  platformio_options: {}
  esp8266_restore_from_flash: false
  board_flash_mode: dout
  includes: []
  libraries: []
  name_add_mac_suffix: false
logger:
  baud_rate: 0
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
sensor:
- platform: wifi_signal
  name: WiFi Signal Sensor
  update_interval: 60s
  disabled_by_default: false
  force_update: false
  unit_of_measurement: dBm
  accuracy_decimals: 0
  device_class: signal_strength
  state_class: measurement
switch:
- platform: restart
  name: amica restart
  disabled_by_default: false
  icon: mdi:restart
uart:
- rx_pin: 3
  tx_pin: 1
  baud_rate: 9600
  rx_buffer_size: 256
  stop_bits: 1
  data_bits: 8
  parity: NONE
text_sensor:
- platform: mqtt_subscribe
  id: humO
  topic: tmp/outside/sensor/humidity/state
  disabled_by_default: false
  qos: 0
  name: humO
  internal: true
display:
- platform: nextion
  id: nextion1
  brightness: 0.5
  tft_url: http://192.168.0.90:7003/firstproject.tft
  update_interval: 60s
  lambda: !lambda |-
    it.set_component_text_printf("humO", "%s", id(humO).state.c_str());
  auto_wake_on_touch: true
mqtt:
  broker: !secret 'mqtt_broker'
  username: !secret 'mqtt_username'
  password: !secret 'mqtt_password'
  discovery: true
  on_message:
  - topic: tmp/nextion/update
    then:
    - lambda: !lambda |-
        id(nextion1)->upload_tft();
    qos: 0
  port: 1883
  discovery_retain: true
  discovery_prefix: homeassistant
  topic_prefix: bedroom-drawer
  keepalive: 15s
  reboot_timeout: 15min
  birth_message:
    topic: bedroom-drawer/status
    payload: online
    qos: 0
    retain: true
  will_message:
    topic: bedroom-drawer/status
    payload: offline
    qos: 0
    retain: true
  shutdown_message:
    topic: bedroom-drawer/status
    payload: offline
    qos: 0
    retain: true
  log_topic:
    topic: bedroom-drawer/debug
    qos: 0
    retain: true
wifi:
  networks:
  - ssid: !secret 'wifi_ssid_0'
    password: !secret 'wifi_password'
    priority: 0.0
  - ssid: !secret 'wifi_ssid_1'
    password: !secret 'wifi_password'
    priority: 0.0
  manual_ip:
    static_ip: 192.168.0.62
    gateway: !secret 'wifi_gateway'
    subnet: !secret 'wifi_subnet'
    dns1: 0.0.0.0
    dns2: 0.0.0.0
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  output_power: 20.0
  use_address: 192.168.0.62
ota:
  safe_mode: true
  port: 8266
  password: ''
  reboot_timeout: 5min
  num_attempts: 10
binary_sensor: []
```

### Anything in the logs that might be useful for us?

```txt
logs during update

[01:58:03][W][nextion:072]: Nextion is not connected! 
[01:58:04][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[01:58:04][W][nextion:072]: Nextion is not connected! 
[01:58:04][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
```

vijaygill, 2021-09-15:
Similar issue happens to my setup also.

```
[nextion:118]:   Device Model:     NX4832K035_011R
[nextion:119]:   Firmware Version: 163
[nextion:120]:   Serial Number:    D363B874531F3D3
[nextion:121]:   Flash Size:       33554432
[nextion:122]:   Wake On Touch:    True
```
MCU used is ESP32.
Nextion is connected to UART on pins GPIO16 & GPIO17.

paul-at-pebbletech, 2022-01-12:
I have managed to get this working. My setup is d1 mini clone with the lcd connected to D6 / D7. latest ESP version running in a stand alone container on a raspi. The ESP code is a hacked version of the weather display project https://github.com/bruxy70/Home-Assistant-ESPHome-Weather-Station. I placed the tft file in my home assistant www file. Had to add "http_request:" to the esphome config file.
Service to update the Nextion file is called from Home assistant Developer Tools | Services then select the service in my case ESPHome: display_weather_update_nextion

Hope this helps some one to avoid the 5 hours I spent figuring it out!

`# This is ESPHome configuration file. This goes to the esphome folder
# You need to provide your WiFi ssis/password and passwords for API and OTA (or put them in your esphome/secrets.yaml)
esphome:
  name: display_weather
  platform: ESP8266
  board: d1_mini
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

api:
  password: !secret api_password
  services:
    - service: update_nextion
      then:
        - lambda: 'id(teplomer)->upload_tft();'

ota:
  password: !secret ota_password

logger:
  baud_rate: 0     # Disable UART logging (pins GPIO1/3 are used for Nextion communication)

uart:
  rx_pin: D6
  tx_pin: D7
  baud_rate: 115200

time:
  - platform: sntp
    id: sntp_time
http_request:
sensor:
 
  - platform: homeassistant   # Inside temperature
    id: temperature_inside
    entity_id: sensor.bedroom_temperature
  - platform: homeassistant   # Swimming pool
    id: temperature_pool
    entity_id: sensor.office_temperature_temperature
 

globals:
 - id: first_page             # First page of the display?
   type: bool
   restore_value: no
 - id: current_brightness             # Is display on?
   type: float
   restore_value: no
   initial_value: '-1.0'
   
display:
  - platform: nextion
    id: teplomer
    update_interval: 6s
    tft_url: 'http://rhome:8123/local/display.tft'
    lambda: |-      
      // Turn display on when somebody is home. Only update when display on.
      float br;
      
        
        // Swich first/second page on each update (every 5 seconds)
      if (id(first_page)) {
[... 52 lines cut ...]
[22:20:41][C][ota:083]:   Address: display_weather.local:8266
[22:20:41][C][ota:086]:   Using Password.
[22:20:41][C][api:134]: API Server:
[22:20:41][C][api:135]:   Address: display_weather.local:6053
[22:20:41][C][api:139]:   Using noise encryption: NO
[22:20:41][C][sntp:050]: SNTP Time:
[22:20:41][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[22:20:41][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[22:20:41][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[22:20:41][C][sntp:054]:   Timezone: 'UTC0'
[22:20:41][C][http_request:014]: HTTP Request:
[22:20:41][C][http_request:015]:   Timeout: 5000ms
[22:20:41][C][http_request:016]:   User-Agent: ESPHome
[22:20:41][C][mdns:084]: mDNS:
[22:20:41][C][mdns:085]:   Hostname: display_weather
[22:20:41][C][homeassistant.sensor:030]: Homeassistant Sensor 'temperature_inside'
[22:20:41][C][homeassistant.sensor:030]:   State Class: ''
[22:20:41][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[22:20:41][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[22:20:41][C][homeassistant.sensor:031]:   Entity ID: 'sensor.bedroom_temperature'
[22:20:41][C][homeassistant.sensor:030]: Homeassistant Sensor 'temperature_pool'
[22:20:41][C][homeassistant.sensor:030]:   State Class: ''
[22:20:41][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[22:20:41][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[22:20:41][C][homeassistant.sensor:031]:   Entity ID: 'sensor.office_temperature_temperature'`

yuyuhn, 2022-04-13:
在homeasstant上，tft文件应该放到哪里？完整的路径是什么？我掉用服务的时候一直报错误
[image]

yaploxoigolyb, 2022-04-26:
[image]
Board: esp32doit-devkit-v1 (ESP32-WROOM)
Display: NX3224T024_011

lnicolas83, 2022-05-18:
The uart timeout of 100ms is too short. It works with 250ms.

```
diff --git a/esphome/components/uart/uart_component.cpp b/esphome/components/uart/uart_component.cpp
index 09b8c975..94beed4b 100644
--- a/esphome/components/uart/uart_component.cpp
+++ b/esphome/components/uart/uart_component.cpp
@@ -11,7 +11,7 @@ bool UARTComponent::check_read_timeout_(size_t len) {

uint32_t start_time = millis();
while (this->available() < int(len)) {
-    if (millis() - start_time > 100) {
+    if (millis() - start_time > 250) {
ESP_LOGE(TAG, "Reading from UART timed out at byte %u!", this->available());
return false;
}
```

lordjaxom, 2022-08-24:
How do modify the sources locally? Where can I find uart_component.cpp?

pbranly, 2023-01-11:
Hi
I have exactly the same issue. Each time I use this procedure, the screen is bricked with such screen.
Solution is to use an external USB/ TTL adapter.
I don’t have logs that could explain the issue, as logs are stopped during the upload procedure but my feeling is that Esphome reboot too soon while nextion is still working the end of its tft load
Please help
Phil

pbranly, 2023-01-11:
@lnicolas83 could you explain your solution ? Where to put your modification ?
Thanks
Phil

OptimusGREEN, 2023-02-21:
`uart:
  debug:
    after:
      timeout: 250ms`

edwardtfn, 2023-11-04:
I was fighting this `Nextion is not connected!` on the past days and maybe this could help other with similar issues: https://github.com/esphome/issues/issues/5029#issuecomment-1793391302

Also, this TFT Upload is quite fragile. I've created https://github.com/esphome/esphome/pull/5667 to add support to ESP-IDF framework and could see some points to improve in reliability at the arduino code. A couple of those have been included on that PR (common code), but as soon that is merged I will try to find some time to invest in that code again and try to make it a bit more reliable than it is currently.

closed 2023-11-28 by kbx81