esphome/issues #2427: Unable to remotely connect to logger in 2021.9.0

xconverge, 2021-09-16:
### The problem

Just updated to 2021.9.0 on multiple devices (ESP32 and ESP8266), same issue with each. OTA upgraded fine, functions fine, but unable to connect to the logger remotely. This worked fine previously. Note I am crossing vlans/subnets from my client I am trying to connect with

in this log, 10.10.5.20 is the esphome device and 10.10.1.15 is the host I am running this from

```
$ esphome logs sample.yaml --device 10.10.5.20
←[32mINFO Reading configuration notification.yaml...←[0m
←[32mINFO Starting log output from 10.10.5.20 using esphome API←[0m
←[33mWARNING Can't connect to ESPHome API for 10.10.5.20: Unexpected error while connecting to 10.10.5.20: A Stream Socket was expected, got <socket.socket fd=896, family=AddressFamily.AF_INET, type=0, proto=6, laddr=('10.10.1.15', 51218), raddr=('10.10.5.20', 6053)>←[0m
←[32mINFO Trying to reconnect to 10.10.5.20 in the background←[0m
```

I am not too worried because it all seems to be running fine, but I am not going to be flashing the devices I do not have easy access to yet...

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

logger

moedtm, 2021-09-16:
Just wanted to add that I'm experiencing the same issue since updating to 2019.0, while I'm not crossing VLAN's/subnets to connect to the ESP32.

reaper7, 2021-09-16:
same for me...

**on windows10 + esp01:** 
```
$ esphome run powermeter111.yaml
INFO Reading configuration powermeter111.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d powermeter111
Processing powermeter111 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <EspSoftwareSerial> 6.8.5
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size .pioenvs\powermeter111\firmware.elf
Checking size .pioenvs\powermeter111\firmware.elf
RAM:   [====      ]  40.5% (used 33156 bytes from 81920 bytes)
Flash: [====      ]  43.7% (used 447364 bytes from 1023984 bytes)
========================= [SUCCESS] Took 3.75 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.111
INFO Uploading powermeter111\.pioenvs\powermeter111\firmware.bin (451520 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.111 using esphome API
WARNING Can't connect to ESPHome API for 192.168.0.111: Error connecting to ('192.168.0.111', 6053): [WinError 121] Przekroczono limit czasu semafora
INFO Trying to reconnect to 192.168.0.111 in the background
```

**on esp01:**
```
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[I][ota:046]: Boot seems successful, resetting boot loop counter.
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[... 67 lines cut ...]
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[I][ota:046]: Boot seems successful, resetting boot loop counter.
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:100]: 192.168.0.2 didn't respond to ping request in time. Disconnecting...
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
[W][api.connection:078]: 192.168.0.2: Connection reset
```

martgras, 2021-09-17:
I suspect it is related to https://github.com/esphome/aioesphomeapi/issues/87
(introduced with https://github.com/esphome/esphome/pull/2306 via
https://github.com/esphome/aioesphomeapi/pull/52/commits/f1dffff27a19cb0880df227f5b1997bb6d17db09)

For now I fixed it for me by adding `type=socket.SOCK_STREAM,` in C:\Python39\Lib\site-packages\aioesphomeapi\host_resolver.py line 189 
```
    try:
        # Limit to TCP IP protocol
        #res = await eventloop.getaddrinfo(host, port, proto=socket.IPPROTO_TCP)
        # fixes https://github.com/esphome/issues/issues/2427
        res = await eventloop.getaddrinfo(host, port, type=socket.SOCK_STREAM, proto=socket.IPPROTO_TCP)

    except OSError as err:
        raise APIConnectionError("Error resolving IP address: {}".format(err))
```

xconverge, 2021-09-17:
Yea this looks promising to be fixed by your PR

xconverge, 2021-09-28:
This seems fixed as mentioned above now that 2021.9.2 is out and has updated aioesphomeapi

closed 2021-09-28 by xconverge