esphome/issues #2428: Platform fastled_clockless does not compile for ESP32-C3

markusg1234, 2021-09-16:
### The problem

Was trying to use the onboard RGB Led and wouldn't compile.

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP32

### Board

ESP32-c3-devkitm-1

### Component causing the issue

fastled

### Example YAML snippet

```yaml
light:
  - platform: fastled_clockless
    chipset: WS2812
    name: "ESP32 LED"
    pin: GPIO8
    num_leds: 1
    rgb_order: RGB
```

### Anything in the logs that might be useful for us?

```txt
/root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/hal/include/hal/touch_sensor_types.h:165:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(touch_pad_intr_mask_t)
 ^~~~~~~~~
In file included from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/controller.h:9,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/FastLED.h:47,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/bitswap.cpp:2:
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/pixeltypes.h:831:13: note: candidate: 'CRGB operator&(const CRGB&, const CRGB&)'
 inline CRGB operator&( const CRGB& p1, const CRGB& p2)
             ^~~~~~~~
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/pixeltypes.h:831:13: note:   no known conversion for argument 1 from 'volatile<unnamed struct>::<unnamed union>' to 'const CRGB&'
In file included from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastled_esp32.h:3,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms.h:36,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/FastLED.h:52,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/bitswap.cpp:2:
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:47:24: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1'; did you mean 'out'?
       else return GPIO.out1.val & ~MASK;
                        ^~~~
                        out
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static volatile uint32_t* _ESPPIN<PIN, MASK>::port()':
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:51:34: error: cannot convert 'volatile<unnamed struct>::<unnamed union>*' to '_ESPPIN<PIN, MASK>::port_ptr_t' {aka 'volatile long unsigned int*'} in return
       if (PIN < 32) return &GPIO.out;
                                  ^~~
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:52:25: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1'; did you mean 'out'?
       else return &GPIO.out1.val;
                         ^~~~
                         out
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static volatile uint32_t* _ESPPIN<PIN, MASK>::sport()':
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:56:34: error: cannot convert 'volatile<unnamed struct>::<unnamed union>*' to '_ESPPIN<PIN, MASK>::port_ptr_t' {aka 'volatile long unsigned int*'} in return
       if (PIN < 32) return &GPIO.out_w1ts;
                                  ^~~~~~~~
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:57:25: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
       else return &GPIO.out1_w1ts.val;
                         ^~~~~~~~~
                         out_w1ts
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static volatile uint32_t* _ESPPIN<PIN, MASK>::cport()':
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:61:34: error: cannot convert 'volatile<unnamed struct>::<unnamed union>*' to '_ESPPIN<PIN, MASK>::port_ptr_t' {aka 'volatile long unsigned int*'} in return
       if (PIN < 32) return &GPIO.out_w1tc;
                                  ^~~~~~~~
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:62:25: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
       else return &GPIO.out1_w1tc.val;
                         ^~~~~~~~~
                         out_w1tc
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static bool _ESPPIN<PIN, MASK>::isset()':
/data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/platforms/esp/32/fastpin_esp32.h:68:37: error: no match for 'operator&' (operand types are 'volatile<unnamed struct>::<unnamed union>' and 'long unsigned int')
       if (PIN < 32) return GPIO.out & MASK;
                            ~~~~~~~~~^~~~~~
In file included from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:44,
                 from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:52,
                 from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:64,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/led_sysdefs.h:48,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/FastLED.h:41,
                 from /data/dining-room-curtain/.piolibdeps/dining-room-curtain/FastLED/bitswap.cpp:2:
```

oxan, 2021-09-19:
This needs upstream support. https://github.com/FastLED/FastLED/issues/1255

maddogjt, 2021-11-17:
I put up a pull request with preliminary support for ESP32-C3.   FastLED/FastLED#1320

maruel, 2021-12-06:
Note that you may be blocked by the workaround done for issue #1375 at line https://github.com/esphome/esphome/blob/dev/esphome/components/fastled_base/__init__.py#L47 that pins FastLED to an old version.

oxan, 2021-12-06:
You can specify a custom version of FastLED using the [`libraries` option](https://esphome.io/components/esphome.html#libraries).

maruel, 2021-12-06:
The version I specify gets overridden by the line above.

oxan, 2021-12-20:
> The version I specify gets overridden by the line above.

Are you sure about that? I just tested it and for me the version specified in the YAML gets preference.

cptskippy, 2024-12-04:
> > The version I specify gets overridden by the line above.
> 
> Are you sure about that? I just tested it and for me the version specified in the YAML gets preference.

For anyone wondering, this only compiles if you use the using the <name>=<source> syntax and version 3.6 or higher.  It doesn't actually work though from what I can tell.

### YAML:

```yaml
esphome:
  libraries:
    - FastLED=https://github.com/FastLED/FastLED#3.7.1

```

ssieb, 2024-12-04:
You should be using the rmt component for ESP32 devices.

closed 2024-12-04 by ssieb