esphome/issues #2430: ESP integration won't start anymore on HA

jjvandenberg, 2021-09-16:
### The problem

The ESPhome integration doesn't start  on HomeAssistant 

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

n/a

### Component causing the issue

n/a

### Anything in the logs that might be useful for us?

```txt
21-09-16 09:56:25 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmp599w_1u2'
21-09-16 09:56:25 ERROR (MainThread) [supervisor.addons.addon] Add-on a0d7b954_esphome can't write options
```

ajCary, 2021-09-16:
Same for me

Xesphome, 2021-09-16:
I had the same problem.  I uninstalled it and re-installed again and that fixed the issue.

jjvandenberg, 2021-09-16:
thanks @ajCary and @Xesphome !
I had to go on with my ESP testing so I reinstalled it - now it's running again. Must have gone wrong during the last Core update.

ajCary, 2021-09-18:
Many thanks @jjvandenberg & @Xesphome. Reinstall fixed esphome. I was having a different issue with esphome beta and reinstalling beta fixed it too.

agners, 2021-09-20:
Did any of you used Backup and/or Restore feature for the ESPHome Add-On before this issue appeared?

If somebody experience it again/has still logs, the log before `[Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmp599w_1u2'` might have some information how it came to that.

ajCary, 2021-09-22:
Sorry, I didn't backup the addon before the update that broke. I'm taking frequent full snapshots of HA but have make quite a few changes and so it would be quite disruptive to try restoring to find the relevant log.

dougle03, 2021-09-22:
> Sorry, I didn't backup the addon before the update that broke. I'm taking frequent full snapshots of HA but have make quite a few changes and so it would be quite disruptive to try restoring to find the relevant log.

Could be linked to this perhaps? [https://docs.linuxserver.io/faq](url)

jjvandenberg, 2021-09-22:
Actually, my error might have occurred after a restore af the full system: the GroWatt integration was also failing, this occurred right after a firmware upgrade, so I tried reverting back to the previous firmware version, which didn’t solve it. (turns out GroWatt changed something server side)

aal7, 2021-10-06:
@Xesphome, @jjvandenberg, @ajCary Would any of you be kind enough to say how you reinstalled ESPHome to get it working again? Is anything lost in the process that needs backing up first? Many thanks.

jjvandenberg, 2021-10-06:
> @Xesphome, @jjvandenberg, @ajCary Would any of you be kind enough to say how you reinstalled ESPHome to get it working again? Is anything lost in the process that needs backing up first? Many thanks.

Hi, i just did a "normal" remove and install, my existing files were maintained.

aal7, 2021-10-06:
Are you talking about pressing UNINSTALL here?
[image]
Thanks

jjvandenberg, 2021-10-06:
@aal7 : yes, that's the way i went ;-)

aal7, 2021-10-06:
OK, thank you.

aal7, 2021-10-06:
@jjvandenberg I've uninstalled ESPHome and restarted my Pi. Then I went to Supervisor -> Add on Store and selected ESPHome. I get this screen
[image]
Can you help to diagnose?

aal7, 2021-10-06:
I've got it going now. Tjhanks.

jjvandenberg, 2022-02-01:
old issue - solved by time ;-)