esphome/issues #2431: pulse_meter total reports 0 decimals of accuracy

ptr727, 2021-09-16:
### The problem

The pulse_meter default accuracy is 2 decimals, but the totals is reported as 0 decimals, I have to manually override the totals accuracy to 2 decimals.

The totals section could inherit the accuracy, icon and device_class from the parent counter, as is it has to be specified twice.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.9

### What platform are you using?

ESP32

### Board

devkitc

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
# https://esphome.io/components/sensor/index.html
sensor:
  # https://esphome.io/components/sensor/pulse_meter.html
  # https://esphome.io/components/sensor/pulse_counter.html
  - platform: pulse_meter
    pin:
      number: GPIO32
      mode: INPUT_PULLUP
    unit_of_measurement: 'gal/min'
    name: ${device_name}_water_meter
    icon: 'mdi:water'
    # There is no water class
    # device_class: 'water'
    filters:
      - multiply: 10
    total:
      name: ${device_name}_water_meter_total
      # Default device class and icon is not inherited
      icon: 'mdi:water'
      # There is no water class
      # device_class: 'water'
      id: water_total
      unit_of_measurement: 'gal'
      # Manually set the accuracy to match the main sensor
      accuracy_decimals: 2
      filters:
      - multiply: 10
  - platform: pulse_meter
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
    unit_of_measurement: 'ft³/min'
    name: ${device_name}_gas_meter
    icon: 'mdi:fire'
    device_class: 'gas'
    filters:
      - multiply: 10
    total:
      name: ${device_name}_gas_meter_total
      # Default device class and icon is not inherited
      icon: 'mdi:fire'
      device_class: 'gas'
      id: gas_total
      unit_of_measurement: 'ft³'
      # Manually set the accuracy to match the main sensor
      accuracy_decimals: 2
      filters:
      - multiply: 10
  # https://esphome.io/components/sensor/template.html
  - platform: template
    name: ${device_name}_water_meter_total_l
    # There is no water class
    # device_class: 'water'
    state_class: 'total_increasing'
    accuracy_decimals: 2
    unit_of_measurement: 'l'
    lambda: |-
      return id(water_total).state / 3.785;
  - platform: template
    name: ${device_name}_gas_meter_total_m3
    device_class: 'gas'
    state_class: 'total_increasing'
    accuracy_decimals: 2
    unit_of_measurement: 'm³'
    lambda: |-
      return id(gas_total).state / 35.315;
```

### Anything in the logs that might be useful for us?

```txt
- platform: pulse_meter
  pin:
    number: 33
    mode: INPUT_PULLUP
    inverted: false
  unit_of_measurement: ft³/min
  name: utility-pulse-counter_gas_meter
  icon: mdi:fire
  device_class: gas
  filters:
  - multiply: 10.0
  total:
    name: utility-pulse-counter_gas_meter_total
    icon: mdi:fire
    device_class: gas
    id: gas_total
    unit_of_measurement: ft³
    filters:
    - multiply: 10.0
    disabled_by_default: false
    force_update: false
    accuracy_decimals: 0
    state_class: total_increasing
  disabled_by_default: false
  force_update: false
  accuracy_decimals: 2
  state_class: measurement
  internal_filter: 13us
  timeout: 5min
```

### Additional information

I added the template sensors to convert the imperial units to metric, hoping HA would like my gas meter, but still no :(

jesserockz, 2022-01-18:
If you think about it, its the total number of pulses. Its an integer value and therefore needs no decimals by default.

closed 2022-01-18 by jesserockz

ptr727, 2022-01-18:
The number is a function of multiply, and multiply does not have to be an integer, so the result could be fractions.