esphome/issues #2432: rf_bridge.beep action does not work

Xesphome, 2021-09-16:
### The problem

If I call the rf_bridge.learn action, the device beeps, so the buzzer is working.  However, calling the rf_bridge.beep action does not produce any sounds no matter the amount of time specified.  The logs show that the code executed correctly. 

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

SonOff RF Bridge

### Component causing the issue

rf_bridge

### Example YAML snippet

```yaml
- rf_bridge.beep:
          id: my_rf_bridge
          duration: 2000
      - logger.log: 'alarm armed'

      - rf_bridge.beep:
            id: my_rf_bridge
            duration: 500
      
        - logger.log: 'alarm disarmed'
```

### Anything in the logs that might be useful for us?

```txt
[14:15:30][D][rf_bridge:220]: Beeping for 2000 ms
[14:15:30][D][main:125]: alarm armed
[14:15:32][D][binary_sensor:036]: 'Alarm Status': Sending state OFF
[14:15:32][D][binary_sensor:036]: 'Global Variable': Sending state OFF
[14:15:32][D][switch:025]: 'rf_bridge_1 Light' Turning OFF.
[14:15:32][D][switch:045]: 'rf_bridge_1 Light': Sending state OFF
[14:15:32][D][rf_bridge:220]: Beeping for 500 ms
[14:15:32][D][main:149]: alarm disarmed
```

### Additional information

If I call the rf_bridge.learn action, the device beeps, so the buzzer is working.

jesserockz, 2021-09-19:
What firmware are you running on the EFM chip? (Not ESP)

The `beep` command only works if you have flashed the custom [Portisch firmware](https://github.com/Portisch/RF-Bridge-EFM8BB1)

closed 2021-09-20 by jesserockz

Xesphome, 2021-09-20:
Thanks for clarifying that.  I do not have the Portisch firmware installed.  Thanks for all that you do.