esphome/issues #2433: ESP32 Last dev 2021-10-0 is no more connecting

SeByDocKy, 2021-09-16:
### The problem

Compule fine but no more connection .... Works fine with the stable release

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.4

### What platform are you using?

ESP32

### Board

WROOM32

### Component causing the issue

?

### Example YAML snippet

```yaml
esphome:
  name: puits
  platform: ESP32
  board: esp32dev
 
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging

logger:
  #level: debug

# Enable Home Assistant API
api:

ota:

time:
  #- platform: sntp
  #  id: my_time
    
  - platform: homeassistant
    id: homeassistant_time  
    
    on_time:
      - seconds: 59
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: template_puits_veille
              state: !lambda return id(daily_well).state;
              
              
i2c:
- sda: GPIO21
  scl: GPIO22
  id: i2c_bus1              
    
switch:
  - platform: restart
    name: "restart_puits"

dallas:
  - pin: GPIO15
    update_interval: 10s

sensor:
  - platform: bme280
    i2c_id: i2c_bus1
    temperature:
      name: "temperature_vs"
      oversampling: 16x
      id: temperature_vs
    pressure:
      name: "pression_vs"
      id: pression_vs
      unit_of_measurement: "bar"
      accuracy_decimals: 3
      filters:
      - lambda: return x / 1000.0;
    humidity:
      name: "humidite_vs"
      id: humidite_vs
    address: 0x76
    update_interval: 10s  

  - platform: homeassistant
    name: "reference_pression"
    entity_id: sensor.darksy_pressure
    id: reference_pression

  - platform: adc
    pin: GPIO36
    name: "pression_puits"
    update_interval: 10s
    attenuation: 11db
    unit_of_measurement: "bar"
    accuracy_decimals: 3
    icon: mdi:gauge
    id: pression_puits
    filters:
    - sliding_window_moving_average:
        window_size: 9
        send_every: 1
    - lambda: return (((x/5.0) + 0.04 )/0.4);
    
    
  - platform: adc
    pin: GPIO39
    name: "pression_surpresseur"
    update_interval: 10s
    attenuation: 11db
    unit_of_measurement: "bar"
    icon: mdi:gauge
    accuracy_decimals: 3
    filters:
    - sliding_window_moving_average:
        window_size: 7
        send_every: 1
    - lambda: return ((x/5.0 - 0.1)*1.333333*10);
    #- lambda: return ((x/5.0 - 0.1)*1.49993*10);

  - platform: dallas
    address: 0x3B8318507361FF28
    name: "temperature_puits"
    unit_of_measurement: "°C"
    resolution: 12
    accuracy_decimals: 2
    
    
  - platform: total_daily_energy
    name: "puits_du_jour"
    power_id: water_flow_meter_puits
    unit_of_measurement: "l"
    accuracy_decimals: 2
    id: daily_well
    
  - platform: template
    name: "puits_veille"
    id: template_puits_veille
    unit_of_measurement: "l"
    accuracy_decimals: 2
    icon: mdi:water
    update_interval: 10s
    
  - platform: template
    name: "volume_puits"
    id: template_volume_puits
    unit_of_measurement: "l"
    accuracy_decimals: 2
    update_interval: 10s
    icon: mdi:water
    filters:
    - sliding_window_moving_average:
        window_size: 9
        send_every: 1
    #lambda: return ( (id(pression_puits).state - 1.10)*7230.7233); 
    #lambda: return ( ( id(pression_puits).state - ( (id(reference_pression).state)*pow( (1.0 - (0.0065*104.0/288.15)) , 5.255)/1000.0 + 0.10))*77238.2294 );
    #lambda: return ( (id(pression_puits).state - ( (id(reference_pression).state)/1000.0 + 0.10 ) )*7230.7233 );
    #lambda: return ( (id(pression_puits).state - ( (id(reference_pression).state)*pow( (1.0 - (0.0065*94.5/288.15)) , 5.255) + 0.10 ) )*7238.2294 );     #108m - 14m = 94 
    
    lambda: return ( (id(pression_puits).state - ( (id(pression_vs).state)/(pow( (1.0 - (0.0065*14/288.15)) , 5.255)) + 0.10 ) )*7238.2294 );     #108m - 14m = 94 
    
    
    # 94.5 comes from my current altitude in order to correct pressure given for sea level
    # 0.1 <=> 1m between the sensor and the water pump
    #https://fr.wikipedia.org/wiki/Variation_de_la_pression_atmosph%C3%A9rique_avec_l%27altitude
    
    # My well is cylindric with a diameter of D=0.96m. My sensor is located 1m below the pump. So if I want to compute the remaining water for the pump I need substracte 0.1 on measured pressure
    # then h= (measured pressured - (1 + 0.1))*10 where we remove the 0 meter air pressure + 1m of difference between the sensor and the pump. Then we multiply *10 to have a meter value
    # Volume of water (L) = h*pi*R²*1000 = delta_p*10*pi*R²*1000= delta_p*pi*(D/2)²1000*10=h*3.14116*(0.96/2)² *1000*10=delta_p*7230.7233 for my particular case
    
    
                                                                  
  - platform: pulse_counter
    id: water_flow_meter_puits
    pin: GPIO2
    name: "water_flow_meter_puits"    
    update_interval: 1s
    icon: mdi:water
    filters:
    - lambda: return (x / 5.48);
    unit_of_measurement: "l"     
    
    
  - platform: wifi_signal
    name: "WiFi puissance_puits"
    update_interval: 10s
    
  - platform: uptime
    #name: "up_puits"
    id: uptime_sec
    #update_interval: 15s 
    #filters:
    #  - lambda: return x / 3600.0;
    #unit_of_measurement: "hours"
    #accuracy_decimals: 2
    #update_interval: 10s
    
binary_sensor:
  - platform: status
    name: "puits_status"
    
    
text_sensor:
  - platform: template
    name: puits_uptime
    lambda: |-
      int seconds = (id(uptime_sec).state);
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600); 
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      return { (String(days) +"d " + String(hours) +"h " + String(minutes) +"m "+ String(seconds) +"s").c_str() };
    icon: mdi:clock-start
    update_interval: 113s
```

### Anything in the logs that might be useful for us?

```txt
No connection ....
```

SeByDocKy, 2021-09-17:
Ok I commented everything except the connection part ... and it's connecting.... Btw, as soon as I had the I2C definition part on GPIO21/GPIO22.... it's connecting partially then is blocked

```INFO Reading configuration /config/esphome/puits.yaml...
INFO Starting log output from puits.local using esphome API
INFO Successfully connected to puits.local
[09:15:02][I][app:102]: ESPHome version 2021.10.0-dev compiled on Sep 17 2021, 09:09:33
[09:15:02][C][wifi:498]: WiFi:
[09:15:02][C][wifi:361]:   SSID: [redacted]
[09:15:02][C][wifi:362]:   IP Address: 192.168.1.72
[09:15:02][C][wifi:364]:   BSSID: [redacted]
[09:15:02][C][wifi:365]:   Hostname: 'puits'
[09:15:02][C][wifi:367]:   Signal strength: -76 dB ▂▄▆█
[09:15:02][C][wifi:371]:   Channel: 12
[09:15:02][C][wifi:372]:   Subnet: 255.255.255.0
[09:15:02][C][wifi:373]:   Gateway: 192.168.1.1
[09:15:02][C][wifi:374]:   DNS1: 192.168.1.1
[09:15:02][C][wifi:375]:   DNS2: 192.168.1.1
[09:15:02][C][i2c:029]: I2C Bus:
[09:15:02][C][i2c:030]:   SDA Pin: GPIO21
[09:15:02][C][i2c:031]:   SCL Pin: GPIO22
[09:15:02][C][i2c:032]:   Frequency: 50000 Hz
[09:15:02][I][i2c:034]: Scanning i2c bus for active devices...
[09:15:03][I][i2c:041]: Found i2c device at address 0x76
[09:15:03][C][logger:193]: Logger:
[09:15:03][C][logger:194]:   Level: DEBUG
[09:15:03][C][logger:195]:   Log Baud Rate: 0
[09:15:03][C][logger:196]:   Hardware UART: UART0
[09:15:03][C][ota:029]: Over-The-Air Updates:
[09:15:03][C][ota:030]:   Address: puits.local:3232
[09:15:03][C][api:132]: API Server:
[09:15:03][C][api:133]:   Address: puits.local:6053```

martgras, 2021-09-17:
May a crash in a i2c sensor.
Remove one by one to see if it makes a difference.
A serial connection and debug logging would provide more details.

SeByDocKy, 2021-09-19:
> May a crash in a i2c sensor.
> Remove one by one to see if it makes a difference.
> A serial connection and debug logging would provide more details.

I will take a new ESP32 to output serial since the oroginal one is under my house

SeByDocKy, 2021-09-21:
Just tried with the nightly dev version .... and it's even worse :( now I can't reflash it OTA  with the working version.... I will need to go under the house to reflash it manually .... :( :( :(

fabiendostie, 2021-09-21:
i am also stuck i cant flash NONE of the esp32's i have in the house I tried 6 different ones so far!they dont come online anymore in esphome but all the esp8266 do and I can OTA no problem with those, I cant do nothing with all the esp32's... this really sucks! tried from mac, pc, windows, raspberry  all the same, connection fails and cant use esptool, esphomeflasher, in the commandline of inside esphome on my home assistant install... I am freaking out.

SeByDocKy, 2021-09-21:
> i am also stuck i cant flash NONE of the esp32's i have in the house I tried 6 different ones so far!they dont come online anymore in esphome but all the esp8266 do and I can OTA no problem with [...]

You can't reflash it with the stable version ?

fabiendostie, 2021-09-21:
I think i am using the latest stable version. could you confirm the exact version number and where is the file? but the issue is that i cant reach the esp32, connection times out, was fine last time untill i updated... a few weeks back...  After that i cant flash using none of the 5-6 methods mentionned in my previous post!

fabiendostie, 2021-09-21:
even a brand new esp32s doenst work now...

fabiendostie, 2021-09-21:
this is the message I get from esphome v1.30:
Using '/dev/cu.SLAB_USBtoUART' as serial port.
Connecting........_____....._____....._____....._____....._____....._____....._____
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Invalid head of packet (0x00)

i get pretty much the same on terminal using esptool, espflasher, esphomeflasher.... linum, mac, pc, android.... all the same!

SeByDocKy, 2021-09-22:
OK went under house .... here are the serial log when flashed last dev (22/01)

```
Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[08:00:13][fum2-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.
[08:00:13][I][logger:216]: Log initialized
[08:00:13][C][ota:527]: There have been 0 suspected unsuccessful boot attempts.
[08:00:13][D][esp32.preferences:113]: Saving preferences to flash...
[08:00:13][I][app:029]: Running through setup()...
[08:00:13][C][dallas.sensor:032]: Setting up DallasComponent...
[08:00:13]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[08:00:13]Core 1 register dump:
[08:00:13]PC      : 0x4008103c  PS      : 0x00060b33  A0      : 0x800d7eae  A1      : 0x3ffb17e0  
[08:00:13]A2      : 0x00000000  A3      : 0x00000001  A4      : 0xffffff8c  A5      : 0x3ffb985c  
[08:00:13]A6      : 0x00001850  A7      : 0x00007361  A8      : 0x00060b20  A9      : 0x3ffb17d0  
[08:00:13]A10     : 0x3ffb3004  A11     : 0x400877a0  A12     : 0x3ffb180b  A13     : 0x0000ff00  
[08:00:13]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
[08:00:13]EXCVADDR: 0x00000014  LBEG    : 0x40162d15  LEND    : 0x40162d27  LCOUNT  : 0x00000000  
[08:00:13]
[08:00:13]ELF file SHA256: 0000000000000000
[08:00:13]
[08:00:13]Backtrace: 0x4008103c:0x3ffb17e0 0x400d7eab:0x3ffb1800 0x400d82bf:0x3ffb1830 0x400e14c2:0x3ffb18a0 0x40162ba1:0x3ffb18f0 0x400e1215:0x3ffb1910 0x400e4a46:0x3ffb1960 0x400e89ae:0x3ffb1fb0 0x4008a9c6:0x3ffb1fd0
[08:00:13]
[08:00:13]Rebooting...
[08:00:14]ets Jun  8 2016 00:22:57
[08:00:14]
[08:00:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:00:14]configsip: 0, SPIWP:0xee
[08:00:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:00:14]mode:DIO, clock div:2
[08:00:14]load:0x3fff0018,len:4
[08:00:14]load:0x3fff001c,len:1044
[08:00:14]load:0x40078000,len:8896
[08:00:14]load:0x40080400,len:5828
[08:00:14]entry 0x400806ac
[08:00:14][fum2-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.
[08:00:14][I][logger:216]: Log initialized
[08:00:14][C][ota:527]: There have been 1 suspected unsuccessful boot attempts.
[08:00:14][D][esp32.preferences:113]: Saving preferences to flash...
[08:00:14][I][app:029]: Running through setup()...
[08:00:14][C][dallas.sensor:032]: Setting up DallasComponent...
[08:00:14]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[08:00:14]Core 1 register dump:
[08:00:14]PC      : 0x4008103c  PS      : 0x00060933  A0      : 0x800d7eae  A1      : 0x3ffb17e0  
[08:00:14]A2      : 0x00000000  A3      : 0x00000001  A4      : 0xffffff8c  A5      : 0x3ffb2068  
[08:00:14]A6      : 0x00001850  A7      : 0x00007361  A8      : 0x00060920  A9      : 0x3ffb17d0  
[08:00:14]A10     : 0x3ffb2f50  A11     : 0x400877a0  A12     : 0x3ffb180b  A13     : 0x0000ff00  
[08:00:14]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
[08:00:14]EXCVADDR: 0x00000014  LBEG    : 0x40162d15  LEND    : 0x40162d27  LCOUNT  : 0x00000000  
[08:00:14]
[08:00:14]ELF file SHA256: 0000000000000000
[08:00:14]
[08:00:14]Backtrace: 0x4008103c:0x3ffb17e0 0x400d7eab:0x3ffb1800 0x400d82bf:0x3ffb1830 0x400e14c2:0x3ffb18a0 0x40162ba1:0x3ffb18f0 0x400e1215:0x3ffb1910 0x400e4a46:0x3ffb1960 0x400e89ae:0x3ffb1fb0 0x4008a9c6:0x3ffb1fd0
[08:00:14]
[08:00:14]Rebooting...
[08:00:14]ets Jun  8 2016 00:22:57
[08:00:14]
[08:00:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:00:14]configsip: 0, SPIWP:0xee
[08:00:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:00:14]mode:DIO, clock div:2
[08:00:14]load:0x3fff0018,len:4
[08:00:14]load:0x3fff001c,len:1044
[08:00:14]load:0x40078000,len:8896
[08:00:14]load:0x40080400,len:5828
[08:00:14]entry 0x400806ac
[... 273 lines cut ...]
[08:00:21][C][ota:202]: Over-The-Air Updates:
[08:00:21][C][ota:203]:   Address: puits.local:3232
[08:00:21][W][ota:211]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[08:00:21][I][app:060]: setup() finished successfully!
[08:00:21][I][ota:542]: Waiting for OTA attempt.
[08:00:22][I][app:099]: ESPHome version 2021.10.0-dev compiled on Sep 22 2021, 07:57:41
[08:00:22][C][wifi:489]: WiFi:
[08:00:22][C][wifi:352]:   SSID: 'sebydocky'
[08:00:22][C][wifi:353]:   IP Address: 192.168.1.72
[08:00:22][C][wifi:355]:   BSSID: 3C:84:6A:1A:EC:2A
[08:00:22][C][wifi:356]:   Hostname: 'puits'
[08:00:22][C][wifi:358]:   Signal strength: -76 dB ▂▄▆█
[08:00:22][C][wifi:362]:   Channel: 12
[08:00:22][C][wifi:363]:   Subnet: 255.255.255.0
[08:00:22][C][wifi:364]:   Gateway: 192.168.1.1
[08:00:22][C][wifi:365]:   DNS1: 192.168.1.1
[08:00:22][C][wifi:366]:   DNS2: 192.168.1.1
[08:00:22][C][logger:235]: Logger:
[08:00:22][C][logger:236]:   Level: DEBUG
[08:00:22][C][logger:237]:   Log Baud Rate: 115200
[08:00:22][C][logger:238]:   Hardware UART: UART0
[08:00:22][C][ota:202]: Over-The-Air Updates:
[08:00:22][C][ota:203]:   Address: puits.local:3232
[08:00:22][W][ota:211]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
````

closed 2021-09-22 by oxan