esphome/issues #2434: ATH10 Unrealistic humidity (0x0)

wuast94, 2021-09-16:
### The problem

I have installed an AHT10 into my Icefridge and constantly get the message below. HA isnt getting readings becouse of that. 

### Which version of ESPHome has the issue?

2021.10.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

---

### What platform are you using?

ESP8266

### Board

Wemos D1Mini

### Component causing the issue

AHT10

### Example YAML snippet

```yaml
Standard Config as Docs
```

### Anything in the logs that might be useful for us?

```txt
[20:43:00][D][sensor:121]: 'Eisfach Temperature': Sending state -18.88523 °C with 2 decimals of accuracy
[20:43:00][D][sensor:121]: 'Eisfach Humidity': Sending state 67.87586 % with 2 decimals of accuracy
[20:43:10][D][aht10:074]: ATH10 Unrealistic humidity (0x0), retrying...
[20:43:10][D][aht10:067]: AHT10 is busy, waiting...
```

mahlernim, 2021-09-17:
Can you try out the changes described in https://github.com/esphome/issues/issues/1635#issuecomment-907358771

chrta, 2021-09-26:
Excerpt from the datasheet of the AHT10:

> Note: The sensor takes time to collect . After the host sends a
measurement command ( 0xAC ) , the delay is over 75ms and the
converted data is read and the returned status bit is normal. If the
status bit [Bit7 of] is 0 for the data can be read normally, the sensor 1
is the busy state, the host needs to wait for the data processing is
completed.

Tasmota seems to wait 100msec: https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_63_aht1x.ino#L118

So i believe the esphome implementation should just add a longer delay after the measurement command according to the datasheet.