esphome/issues #2435: Sensors publishing works irregularly after 2021.9.0

DrCoolzic, 2021-09-17:
### The problem

I have a ESPHome ESP8266 device using the teleinfo component

Yesterday I have updated ESPHome to 2021.9.0 and since the update some but not all sensors **do not update on a regulars basis anymore**. Normally all the sensors are supposed to update every 60 s as specified in the configuration file but this does not work anymore since 2021.9.0
Here is an example of a total_increasing sensor that reports energy consumption and is still working correctly:
[image]

But if I take another sensor that reports the voltage on phase 1 from the grid (this is a fast changing value) this sensor does not reports value on a regular basis anymore:
[image]
As you can see many values where reported before the update the 16 sept around 3:30PM

Another pictures that shows the problem is a statistical view of the voltage for the three phases
[image]
As you can see there is a lot of fluctuations before the update and only a few after the update.

To better understand the problem I have decided to look at the log file from the node.
If I do that the values are correctly published every 60 seconds, as specified, and all sensors are updated correctly as it used to be before 2021.9.0 ?! :O
Here is an example displaying the same voltage sensor **while looking at the log file at the same time**
[image]
To do this I have stated two HA sessions: on one I look at the log of the component on the other one I look at the values of the sensors.
As you can see when I look at the log output of the teleinfo component on one Chrome window the values are now reported correctly in the other Chrome window

I do not know if this is a problem related to the teleinfo component or if it is a more general problem of ESPHome?

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

teleinfo

### Additional information

For information 
- As I suspected that ESPHome could have been updated incorrectly, I have uninstalled and reinstalled ESPHome but that did not help.
- I have updated HA core from 2021.9.5 to 2021.9.6 at about the same time.
- I did a snapshot just before the ESPHome update in case

defl, 2021-09-18:
I'm seeing the same after a 2021.9.0 upgrade without HA changes (filed as a more generic report under #2441)

DrCoolzic, 2021-09-18:
Thanks
This very irritating because lots of information lost.
How do you roll back to 2021.8 ?

DrCoolzic, 2021-09-20:
I have search the community on how to revert to previous version of an addon and not much information :(
What I did: Before updating ESPHome i made a partial backup for the addon, so I have used that to revert to 2021.8.2
Took only few seconds to revert? and not reflected in supervisor ? So I rebooted the host and now it seems to be using 2021.8.2?
From what I have seen supervisor is fetching the old version?

But unfortunately even after I restored ESPHome 2021.8.2 that did not help
Only half of the sensors are updating regularly the other stay stuck to the same value.
And again if I open another chrome session of HA and if I go to the teleinfo component and if I open the log file then everything is working as normal !!!

As I have mentioned above at about the same time I have updated HA Core (and probably Supervisor). So I do not know for sure what is causing the break?

Is it normal that no one looks at PR ?

OttoWinter, 2021-09-20:
This might be fixed with esphome/esphome#2342 and I've heard reports that it fixes similar issues.

Please try again with 2021.9.1 (which includes that fix), thanks!

DrCoolzic, 2021-09-20:
I have tested with 2021.9.1 and it still does not work for me
Below is a graph that shows 

- Last 24h before 11AM => only very limited number of sensor updates
- After 11 AM I have started to look at the log of the component in a separate HA window => Normal update every 60 sec
- After 2PM I have **updated ESPHome to 2021.9.1** and I have stopped to look at the log file => **no more sensor updates**

[image]

DrCoolzic, 2021-09-20:
for no reason now it seems to work after a period not working???

[image]

I wiil keep you posted

OttoWinter, 2021-09-20:
> Does this mean that when you update the addon you also have to update the firmware downloaded to the devices?

Ah should have mentioned that. Yes that's needed :)

DrCoolzic, 2021-09-20:
Good to know. This probably also explain why when I reverted to 2021.8.2 without reuploading the firmware it did not repaired the problem.
I will wait a couple of days and mark close the problem
Thanks for the great job you are doing.

defl, 2021-09-20:
I’m running esphome 2021.9 in HA and have pushed the 2021.8 firmware via a separate esphome install via cli. Also works flawlessly.

DrCoolzic, 2021-09-20:
Yes indeed this is obvious the version of ESPHome just influence the firmware loaded. After that ESP devices communicates with HA without intervention of ESPHome. During "run time" it only provides an interface to program new or existing components :(

defl, 2021-09-21:
After 8 hours it didn't crash yet. Was running without sensors to try and eliminate as much noise as I could . Scaling up to adding sensors now which lead to a failed update when trying to upload (no screenshot, something about error 104 from memory). Second attempt worked though. Will report back in 24h or so.

DrCoolzic, 2021-09-22:
Since I have updated to firmware to 2021.9.1 everything has been working correctly for the last 24+ hours.
therefore I close this PR

closed 2021-09-22 by DrCoolzic