esphome/issues #2436: GPIO0 and GPIO1 not working as GPIO outputs with Ethernet enabled on Olimex ESP32 POE

RonnyLV, 2021-09-17:
### The problem

GPIO0 and GPIO1 not working as GPIO outputs with Ethernet enabled on Olimex ESP32 POE when i try to switch them on/off.

Works fine if configured on other pin, for example GPIO14.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-poe

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
esphome:
  name: nknet-shed-alarm
  platform: ESP32
  board: esp32-poe

...

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

# Individual outputs
switch:
  - platform: gpio
    pin: GPIO0
    inverted: True
    id: relay
    name: "Relay"
```

OttoWinter, 2021-09-20:
From https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy it looks like this is a limitation with the esp32 arduino framework, it has `CONFIG_ETH_RMII_CLK_OUTPUT` enabled and so GPIO0 is used for the REF_CLK signal.

Not sure why GPIO1 would be impacted too though

RonnyLV, 2021-10-04:
Also, those are not the only GPIO pins that were affected, if needed I can can have a look at all of them.

nagyrobi, 2022-12-08:
Please try 2022.12(Beta), Ethernet driver has been rewritten. The board can be used with `esp-idf` framework.

kpfleming, 2024-06-01:
I have been able to use these boards with both Arduino and ESP-IDF frameworks in recent months, this issue should likely be closed.

closed 2024-06-01 by nagyrobi