esphome/issues #2437: DSMR Hang with new ESPHome version 2021.9.0

Be-Virtual, 2021-09-17:
### The problem

Dsmr is still sending data (when I look at the log of the slimmelezer but on HA sometime 5 min or more between 2 updates on screen.

Second point Belgian correction still not implemented and had to use this code to enable it to wrok:
platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3

Thanks for the great work

Philippe

Log from esphome
INFO slimmelezer.deffet.local: Ping Failed!
INFO Disconnected from ESPHome API for slimmelezer.deffet.local
WARNING Disconnected from API
INFO slimmelezer.deffet.local: Unexpected error while reading incoming messages: 0 bytes read on a total of 116 expected bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 505, in run_forever
    await self._run_once()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 480, in _run_once
    pkt = await self._frame_helper.read_packet()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 219, in read_packet
    return await self._read_packet_plaintext()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 214, in _read_packet_plaintext
    raw_msg = await self._reader.readexactly(length_int)
  File "/usr/lib/python3.9/asyncio/streams.py", line 721, in readexactly
    raise exceptions.IncompleteReadError(incomplete, n)
asyncio.exceptions.IncompleteReadError: 0 bytes read on a total of 116 expected bytes
WARNING Can't connect to ESPHome API for slimmelezer.deffet.local: Timeout while connecting to ('192.168.20.174', 6053)
INFO Trying to reconnect to slimmelezer.deffet.local in the background
INFO Successfully connected to slimmelezer.deffet.local

### Which version of ESPHome has the issue?

v2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

D1Mini (Slimmelezer)

### Component causing the issue

api ?

### Anything in the logs that might be useful for us?

```txt
Log from esphome
INFO slimmelezer.deffet.local: Ping Failed!
INFO Disconnected from ESPHome API for slimmelezer.deffet.local
WARNING Disconnected from API
INFO slimmelezer.deffet.local: Unexpected error while reading incoming messages: 0 bytes read on a total of 116 expected bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 505, in run_forever
    await self._run_once()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 480, in _run_once
    pkt = await self._frame_helper.read_packet()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 219, in read_packet
    return await self._read_packet_plaintext()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 214, in _read_packet_plaintext
    raw_msg = await self._reader.readexactly(length_int)
  File "/usr/lib/python3.9/asyncio/streams.py", line 721, in readexactly
    raise exceptions.IncompleteReadError(incomplete, n)
asyncio.exceptions.IncompleteReadError: 0 bytes read on a total of 116 expected bytes
WARNING Can't connect to ESPHome API for slimmelezer.deffet.local: Timeout while connecting to ('192.168.20.174', 6053)
INFO Trying to reconnect to slimmelezer.deffet.local in the background
INFO Successfully connected to slimmelezer.deffet.local
```

defl, 2021-09-18:
I had other issues when upgrading to 2021.9.0 where the DSMR started to report corrupted data. All of my 2021.9.0 devices started to have issues with their sensors and connectivity so I did a full rollback to 2021.8 and all is fine again. This might not be related to DSMR.

DrCoolzic, 2021-09-18:
I have a similar problem problem with teleinfo component: lots of information missing!
see https://github.com/esphome/issues/issues/2435

OttoWinter, 2021-09-20:
This might be fixed with esphome/esphome#2342 and I've heard reports that it fixes similar issues.

Please try again with 2021.9.1 (which includes that fix), thanks!

zuidwijk, 2021-09-20:
I've noticed with v2021.9.0 that when accessing the webpage, the ping to the device (which I had open) drops... closing the webpage and the ping resumes after a while...

zuidwijk, 2021-09-20:
Look better with 2021.9.1 indeed... Yet I also have the impression that it can be caused by a brownout.
When the esp sends too much data via wifi, it reaches the 250mA limit from the meter.

mmakaay, 2021-09-20:
I have a DSMR reader that runs on the power as delivered by my DSMR v4.2 meter, which it considerably lower than the 250mA for a v5 meter. I have no brownout issues with this one. I do log more dan normally, because I'm working on a UART debugger, which logs all received bytes. Therefore, my first impression is that brownouts shouldn't be the issue here.

I will pump up the logging to see if that results in network issues.

mmakaay, 2021-09-20:
No wait... ignore all of the above. I'm not building using my rebased dev branch, but using a separate branch that I'm building in, whoops *shame on me*. But at least this gives me a really nice base line to check against :-)

mmakaay, 2021-09-20:
My colleague reports that he had a device with some sensors that went off line a lot with 2021.9. This is a device that is powered by a power supply unit, so that can't be a brownout issue. After upgrading to 2021.9.1, the device has become stable (for about 20 minutes, which he hadn't seen before).

mmakaay, 2021-09-20:
I'm now running my smart meter reader with `VERY_VERBOSE` active, and I've seen some interesting behavior:

* One ESP-01 didn't work after a flash operation. When connecting it to my laptop, the serial logger showed a reboot loop. After flashing it, the behavior was fixed. (scary)
* In the smart meter, I did see "Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts" come by, so that might indicate some brownout behavior. (and just now, I see a log message about '4 restarts', so there's a route through which the device will become fully unresponsive in the end)
* If I try to flash the device with a lower log level, sometimes the logging crawls  to a halt and even times out. This happens at different percentages. Here's an example run that failed:
```
INFO Successfully compiled program.
INFO Connecting to 192.168.100.176
INFO Uploading /config/smartmeter/.pioenvs/smartmeter/firmware.bin (392176 bytes)
Uploading: [======================================================      ] 90% 
ERROR Error sending data: timed out
```
After some retries, it worked out, and I've got a working device again.

I'll leave it running now without the excessive amount of logging that is triggered by VERY_VERBOSE.
Note BTW that the UART debugger will help a lot here, because VERY_VERBOSE is only used to see the individual bytes that were read, but the UART debugger will do it in a more friendly way (not 1 log line / 1 byte, unless one configures it willingly as such).

I have the device running now with the stock firmware in `VERBOSE` log mode. While in 2021.8.* I have not seen issues with this, now I pretty soon got the message:
```
ConnectionResetError: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for 192.168.100.176
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for 192.168.100.176: Error connecting to ('192.168.100.176', 6053): [Errno 111] Connect call failed ('192.168.100.176', 6053)
INFO Trying to reconnect to 192.168.100.176 in the background
```

After this message, the logger stays silent. No reconnect is done. Starting a fresh logger doesn't work either.
After this, flashing the device does start, but again:
```
INFO Successfully compiled program.
INFO Connecting to 192.168.100.176
INFO Uploading /config/smartmeter/.pioenvs/smartmeter/firmware.bin (392176 bytes)
Uploading: [===========                                                 ] 18% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```
On a second try, it did work.

I will now switch to `WARN` for the logger. That surpresses most of the standard logging. I'll leave the device running like that to see how it behaves.

mmakaay, 2021-09-20:
With log level WARN, the device has been stable for 5 hours now.

jsuanet, 2021-09-21:
> My colleague reports that he had a device with some sensors that went off line a lot with 2021.9. This is a device that is powered by a power supply unit, so that can't be a brownout issue. After [...]

And this device is still stable after 16 hours 😀 Also my DSMR-device (dsmr-v5), powered by the P1 port,  did not have any disconnect after the recompile with 2021.9.1.

mmakaay, 2021-09-21:
Went back to `VERY_VERBOSE` (because, who wants a stable device?) and things went south again. I could ping and flash the device, but the API connections were refused.

What I tried after this, was to flash it back to `WARN`. Flashing succeeded, multiple times, but the strange thing is that the device stayed unavailable. I would suspect when a flash works, that the device restarts afterwards and will connect perfecly fine with the downtuned logging. After leavingi the device simmer for a while, it did come back online in its stable form. Weird, so I will do some more testing to see what the pattern is.

Now I'm flashing again with `VERY_VERBOSE`, but also with MQTT enabled, to make it possible to view some logs via MQTT. I'm curious to see if MQTT still works when the API is refusing connections.

mmakaay, 2021-09-21:
Okay, getting somewhere.  One problem that we might be looking at for the DSMR component, is that its loop() is very greedy. It will not stop reading until the last available byte on the serial bus is read. Combined with `VERY_VERBOSE` logging, this causes the loop to block for a long time.

This might be even be more prevalent on a DSMR 5 reader, since those tend to provide data over serial a lot more frequently to DSMR 4 (I get updates every 15 seconds, while DSMR 5 is more like 1 to a few seconds I think). If the logging takes more time than the time gap between two data cycles, then this might result in an endlessly blocking loop().

I did a quick hack on my device (against the 2021.9.1 release version of ESPHome) to make that loop less greedy by letting it read a max number of bytes per loop() call, and that seems to help a lot. I now have a device running in `VERY_VERBOSE` mode with the hack, and it has been reporting values for the last 20 minutes without hickups. That's a new record for `VERY_VERBOSE` 🥇 
I'm currently having issues with getting a working device using the dev branch, but as soon as that is figured out, I will provide a PR for fixing this issue.

Another thing that might be useful, is to allow the configuration of a pin that is used for steering the data request line of the P1 port. That request line must be pulled high to request telegrams from the meter. This can effectively be used for preventing overlapping telegrams, by dropping the request line while a telegram is being processed and pulling it up afterwards.
Such request line would also allow for implementing a customized update frequency.
Of course, the hardware must support the use of such request pin. Most circuits simply connect +5V to the data request line to have the meter send datagrams as long as the device is connected to the port.

defl, 2021-09-22:
Just flashed my Slimmelezer to 2021.09.01:

[14:54:20][C][wifi_info:010]: WifiInfo SSID 'Wi-Fi SSID'
[14:54:20][C][wifi_info:011]: WifiInfo BSSID 'Wi-Fi BSSID'
[14:54:21][E][dsmr:138]: !38F4
 ^
Checksum mismatch
[14:54:22][E][dsmr:138]: !8DE8
 ^
Checksum mismatch
[14:54:23][E][dsmr:138]: !48FB
 ^
Checksum mismatch
[14:54:24][E][dsmr:138]: !71F2
 ^
Checksum mismatch
[14:54:25][E][dsmr:138]: !61D8
 ^
Checksum mismatch

Rolling back...

mmakaay, 2021-09-22:
What log level are you at? I could push my device into these checksum mismatches while using `VERY_VERBOSE`. With that setting, I think that some bytes from the telegram body simply went missing for me, because the device was too busy with all the logging involved.
If it's not the log level (you could try and see if `WARN` as level fixes anything for you), then I'd still be interested to see if things will be fixed by the PR that I'm currently preparing.

defl, 2021-09-22:
Just the default of "logger:" without any further config. Will try with WARN

defl, 2021-09-22:
With WARN:

```
INFO Successfully compiled program.
INFO Connecting to 192.168.30.106
INFO Uploading /data/slimmelezer-k1/.pioenvs/slimmelezer-k1/firmware.bin (388576 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.30.106 using esphome API
INFO Successfully connected to 192.168.30.106
[15:26:49][E][dsmr:138]: !C732
 ^
Checksum mismatch
[15:26:50][E][dsmr:138]: !5673
 ^
Checksum mismatch
[15:26:51][E][dsmr:138]: !ADE6
 ^
Checksum mismatch
[15:26:54][E][dsmr:138]: !97C1
 ^
Checksum mismatch
[15:26:55][E][dsmr:138]: !F3AD
```

mmakaay, 2021-09-22:
Still broken, so something else is interfering here.
Are you able to run ESPHome from a dev branch of mine?
It's the latest dev version, plus a few dsmr changes that I did.

And could you share your (password-redacted) YAML config, so I can try reproducing this on my device?

defl, 2021-09-22:
@mmakaay yeah no problem, point me to it.

Config:
```
esphome:
  name: slimmelezer-k1
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"

wifi:
  ssid: !secret wifi_iot_h_ssid
  password: !secret wifi_iot_h_password
  fast_connect: true  # Required for hidden networks
  use_address: 192.168.30.106

logger:
  level: WARN
api:
ota:

uart:
  baud_rate: 115200
  rx_pin: D7

dsmr:
  id: dsmr_instance

sensor:

  - platform: dsmr
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    gas_delivered:
      name: "Gas Consumed"
      state_class: total_increasing

  - platform: uptime
    name: "Uptime"
  - platform: wifi_signal
    name: "slimmelezer_k1 WiFi Signal"
    update_interval: 240s

text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

mmakaay, 2021-09-22:
You might want to try disabling serial logging to enable hardware UART:
```
logger:
  baud_rate: 0
  level: DEBUG
```

Furthermore, you have a lot of sensors defined. Shouldn't be a big issue and certainly not a soluution, but for debugging purposes, could you trim down those to a few and see if that changes the behavior?

I just committed some changes to my fork. It'd be great if you could test drive it:
https://github.com/mmakaay/esphome/tree/dsmr-clogging-fix
I've got my  own reader device running rock solid using this, even with `VERY_VERBOSE` logging active.

defl, 2021-09-22:
This is a trimmed down version of the recommended config for the Slimmelezer, I was assuming the UART there was for the dsmr_instance bit but I didn't check TBH. Running the main 2021.9.1 with baud_rate: 0 solves the problem, no more errors.

mmakaay, 2021-09-22:
Ah nice, then I can try reproducing the issue. Maybe I can add a warning / hint when checksum failures come up with the uart logger active.

mmakaay, 2021-09-22:
Kind of disappointing. Using the same configuration (without baud_rate=0 for the logger) and 2021.9.1, I am unable to trigger the same issue :-/ This is on my own reader device BTW, based on an ESP-01.
Do you have a DSMR v5 meter by any chance, @defl ? Those send more telegrams, so that might trigger the different behavior here.

mmakaay, 2021-09-22:
Ah, the way to reproduce it for me, is to use `VERY_VERBOSE` logging along with UART logging active. Quite harsh, I will apologize to my device 😃

defl, 2021-09-22:
@mmakaay Yes I've got a DSMR v5 and it sends a LOT of messages (to the point where i'd actually appreciate a throttle in the dsmr module so that I don't need to put a filter per sensor in

```
[17:45:43][D][text_sensor:015]: 'DSMR Identification': Sending state 'ISK5\2M550T-1012'
[17:45:43][D][text_sensor:015]: 'DSMR Version': Sending state '50'
```

mmakaay, 2021-09-22:
The best way to throttle would require hardware support. I'm currently extending my design with an option to drive the request pin of the P1 meter, which would make it possible to fully silence the meter when no data has to be read (e.g. during processing of a datagram or when a lower update frequency is required).
Another way would be for the code to simply ignore incoming bytes for a given amount of time. That can be done fully in software. I'll see if I can add a feature for both routes.

 On my device, I can enable the uart logger as long as I keep the log level at `DEBUG` or `VERBOSE`. But my v4 meter sends less data I think, and only once per 15 seconds. So that is likely the difference here. 
 When pushing the logger to its limits, I get the same error messages as you, most likely indicating dropped bytes on the serial line. The best fix for that is what we already found out: make sure that hardware UART is used.

Be-Virtual, 2021-09-22:
Hi, As belgian user we need this piece of code to avoid the checksum error:
Could help with you to.

esphome:
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3

Till the update is taken into the prod version.

mmakaay, 2021-09-23:
The checksum fix in the Dsmr lib probably is not the issue here, since switching to hardware UART fixed the issue. That means that the checksum errors most likely pointed at actual issues in reading the data.

michaelarnauts, 2021-09-24:
I saw a lot of checksum errors in the logs, but those are gone now when setting baud_rate to 0 for logging.

```
logger:
  baud_rate: 0
```

I have a Belgian DSMRv5 meter, and get updates every second. Note that I had to use `glmnet/Dsmr@0.5` like so.
```
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3
```

mmakaay, 2021-09-24:
The 0.5 version of the Dsmr library will be used by default in 2021.10.0.
One thing that I want to look at, is incorporating the parser in the ESPHome code tree directly, to prevent having to do library juggling like described above.

rhvs, 2021-10-19:
> That code is new since yesterday (I changed the accuracy_decimals from 0 to 3). I have this with latest normal ESPHome (2021.9.3) working flawlessly on my test meter.
> 
> Do you get between those checksum mismatch also good readings? Those messages aren't necessarily bad, as a meter can spit out corrupt telegrams and than the checksum doesn't match.

Unfortunately nothing related to energy, just uptime and wifi signal.

mmakaay, 2021-12-10:
Closing this issue. A lot of changes have been done to the DSMR component and they have problem to solve the existing issues. Also, documentation was written for the DSMR component with tips about getting trustworthy readings.

closed 2021-12-10 by mmakaay