esphome/issues #2441: 2021.9.0 upgrade issues across the board

defl, 2021-09-18:
### The problem

After upgrading my ESPHome 2021.8 to 2021.9.0 through HA integration, which succeeded, I started to see problems all over the board. From DSMR reporting corrupted data often, missing DS18B20 sensors, missing log lines (sensor not reported but data points coming in), loss of WIFI connectivity after a few hours. Many different types of platforms and implementations involved, so not device specific.

Rolling back to 2021.8.0 (via command line) has solved all issues and all devices is running nominal again for hours now.

Apologies for the vague report.
(Lesson learned in not using the upgrade all button)

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.6

### What platform are you using?

ESP32

### Board

ESP-WROOM-32, ESP8266 (esp01_1m, H801), ESP8266 (esp8285, sonoff basic)

### Anything in the logs that might be useful for us?

```txt
The bits that are interesting are the missing lines vs 2021.8.0, I was missing a lot of DS18b20's
```

defl, 2021-09-18:
Looks like #2437 and #2435 are more specific reports but sounds like same issue.

DrCoolzic, 2021-09-20:
rolling back to 2021.8.2 did not help for my problem

OttoWinter, 2021-09-20:
This might be fixed with https://github.com/esphome/esphome/pull/2342 and I've heard reports that it fixes similar issues.

Please try again with 2021.9.1 (which includes that fix), thanks!

defl, 2021-09-20:
@OttoWinter updated one sensor and running, let's see what happens (interestingly, that one sensor (esp32) always refused to connect to wifi for reasons unknown but with 2021.9.1 it all of a sudden worked).

defl, 2021-09-22:
2021.9.1 is stable here with sensors for 24h, seems to have solved the main issue here for at least one platform/sensor. Closing.

closed 2021-09-22 by defl