esphome/issues #2442: fatal error when compiling firmware for esp32

xlisticks, 2021-09-18:
### The problem

When compiling firmware for esp 32 esphome stops with fatal error:
In file included from src\esphome\core\application_32.cpp:1:0:
src/esphome/core/application.h:3:18: fatal error: string: No such file or directory
compilation terminated.
*** [.pioenvs\test\src\src\esphome\core\application_esp32.cpp.o] Error1
*** [.pioenvs\test\src\esphome\core\application.cpp.o] Error1

### Which version of ESPHome has the issue?

ESPHome v2021.9.0 running on windows 7 with Python latest ver

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32cam

### Example YAML snippet

```yaml
esphome:
  name: mobile_camera
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "MGI"
  password: "soKARezt"

api:

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20Mhz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  name: Camera1

output:
  platform: gpio
  pin: GPIO4
  id: gpio_4

binary_sensor:
  platform: status
  name: "ESP32Cam status"

light:
  platform: binary
  output: gpio_4
  name: "Cam Light"
```

### Anything in the logs that might be useful for us?

```txt
In file included from src\esphome\core\application_32.cpp:1:0:
src/esphome/core/application.h:3:18: fatal error: string: No such file or directory
compilation terminated.
*** [.pioenvs\test\src\src\esphome\core\application_esp32.cpp.o] Error1
*** [.pioenvs\test\src\esphome\core\application.cpp.o] Error1
```

### Additional information

it compilies firmware for esp8266 well, but not for esp32 and it is always the same error doesnt matter of content of .yaml config.

oxan, 2021-09-19:
I see several weird things in your logs:
- There's no file `application_32.cpp` in ESPHome
- The pioenv is called `test` instead of `mobile_camera` 
- There's two `src` directories in the path of `application_esp32.cpp`

Can you try removing all temporary directories and try it again?

xlisticks, 2021-09-19:
- Hi, i really dont know how to remove temporary  dirs, can you point me?
- How to add application_32.cpp to ESPHome?

OttoWinter, 2021-09-20:
> There's no file application_32.cpp in ESPHome

There is now since https://github.com/esphome/esphome/pull/2096 :wink: 

But yeah this build setup looks quite weird

> Hi, i really dont know how to remove temporary dirs, can you point me?

In the directory where you have your esphome configurations, there should be a directory `mobile_camera` - delete that.

oxan, 2021-09-20:
> > There's no file application_32.cpp in ESPHome
> 
> There is now since [esphome/esphome#2096](https://github.com/esphome/esphome/pull/2096) 😉

No, that one is called `application_esp32.cpp` (note the esp before 32) 😉

xlisticks, 2021-09-20:
[image]

closed 2022-02-02 by github-actions[bot]