esphome/issues #2443: ESPhome does not work on sonoff basic R3

rbr101, 2021-09-19:
### The problem

The ESPhome software does not work on sonoff basic R3 version. The image uploads fine but then WIFI won't connect and connection serial port ([19:58:45]/ISK0!EC6C ERROR Serial port closed!) is not possible. When flashing with tasmo firmware everything works fine.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP8266

### Board

sonoff

### Component causing the issue

bacicR3

### Example YAML snippet

```yaml
esphome:
  name: sonoff-bbq
  platform: ESP8266
  board: esp01_1m

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff BBQ"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: "xxxxx"
  password: "xxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: " "
    password: " "

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/sonoff-bbqtafel.yaml...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
ERROR Serial port closed!
```

OttoWinter, 2021-09-20:
What hardware is the HA addon running on?

I've heard some cases where on RPis the latest linux kernel has a bug where the serial chip driver is broken and closes the port right away.

rbr101, 2021-09-20:
I'm running Home Assistant OS 6.4 on an Intel NUC8i3BEK2

DyArt, 2021-09-25:
I have exactly the same problem with the basic R3. All other devices (Dual R2  and S20) works fine after updating to 2021.9.x. Mine is running ESPHome 2021.9.1 with core-2021.9.7 and supervisor-2021.09.0 on Proxmox Intel

randybb, 2021-09-25:
Guys, all R3 are based on ESP32.

closed 2022-02-12 by github-actions[bot]

platima, 2022-05-04:
Oh, setting 
```
  platformio_options:
    board_build.flash_mode: dio
```
Worked for me