esphome/issues #2444: Background calibration & ABC commands for SenseAir S8 not working

quizzical86, 2021-09-19:
### The problem

@nmaggioni added a great feature on March 19 to be able to have [Background calibration & ABC commands for the SenseAir S8](https://github.com/esphome/esphome/pull/1623).

However it only seems to work when setting the debug level to VERY_VERBOSE.

In any other mode, triggering of the manual background calibration does not seem to work. If I activate it for example for a sensor that reports 500ppm CO2, it will not calibrate that value to 400, but keep on reporting 500.
Also activating or disactivating the ABC on the sensors no longer works. This results in errors such as:
```
[20:27:34][D][senseair:106]: SenseAir Enabling automatic baseline calibration
[20:27:36][W][senseair:031]: Invalid preamble from SenseAir! fe06001f 00b4ac74 fe040800 00
[20:27:40][W][senseair:031]: Invalid preamble from SenseAir! 01021306 77fe0408 00000000 00
[20:27:44][W][senseair:031]: Invalid preamble from SenseAir! 77fe0408 00000000 00010214 47
```

I only managed to get the manual background calibration working by setting the debug level to VERY_VERBOSE (VERBOSE,  DEBUG etc. won't do). It still gives the invalid preamble error though.
I tested this on four devices.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

senseair

nmaggioni, 2021-09-20:
I have a hunch that the `Invalid preamble` warning comes from a dirty buffer somewhere: the first "bad" preamble you quoted, for example, actually contains both the ABC enabling command itself (`fe06001f 00b4ac74`) and the preamble of the expected result (`fe040800`). I'll try to look into this when I'll have the chance.

nmaggioni, 2021-09-20:
Those preamble warnings shouldn't appear anymore, even if they were mostly just a cosmetic nuisance. Calibration and ABC commands are working fine here, so I can't reproduce that issue at the moment.

When/where did you buy your sensors? I didn't see any obvious reason why the debug level should influence this behavior (a race condition somewhere _could_ explain it, but I highly doubt that's the case).

quizzical86, 2021-09-21:
Thanks a lot for looking into it.
These are the purchase dates of my Senseair sensors:

- 24 May 2021
- 5 April 2021
- 15 February 2021
- 11 Nov 2020

I have bought them on Aliexpress (as far as I'm aware there are no fake Senseair S8's on the market). I purchased them [here](https://www.aliexpress.com/item/4000884582128.html?spm=a2g0s.9042311.0.0.27424c4dlJ9iC4) and [here](https://www.aliexpress.com/item/32821588322.html?spm=a2g0s.9042311.0.0.27424c4dlJ9iC4).

I agree it's strange but for all these sensors I clearly see that the PPM does not get calibrated to 400 when I launch the calibration in any other mode than VERY_VERBOSE.

dadav, 2021-11-26:
I have the same issue, calibration is not working. I also bought mine from aliexpress and followed this tutorial: https://www.jeffgeerling.com/blog/2021/airgradient-diy-air-quality-monitor-co2-pm25

Setting the logger to `VERY_VERBOSE` results in my device not be able to connect anymore :(

I've seen there is a fix in the dev-Branch, can't wait that this gets merged!!

dadav, 2021-11-26:
Ok, just updated to the latest esphome-version. It works now :D

nmaggioni, 2021-11-26:
`VERY_VERBOSE` logging breaks protocols that rely on strict timing since dumping everything through serial is slow and messes up the timings. See #2552.

jcastro, 2022-01-28:
Trying on esphome 2022.1.2 I'm getting this error. I have waited way more than one lamp timing on the sensor!

`[D][button:013]: 'Calibration' Pressed.
[D][senseair:082]: SenseAir Starting background calibration
[D][senseair:058]: SenseAir Received CO₂=550ppm Status=0x00
[D][sensor:125]: 'SenseAir CO2 Value': Sending state 550.00000 ppm with 0 decimals of accuracy
[D][senseair:058]: SenseAir Received CO₂=558ppm Status=0x00
[D][sensor:125]: 'SenseAir CO2 Value': Sending state 558.00000 ppm with 0 decimals of accuracy
[D][senseair:090]: SenseAir Requesting background calibration result
[D][senseair:106]: SenseAir Result=NOT_OK (fe0302 0000 ac50)
[D][senseair:058]: SenseAir Received CO₂=539ppm Status=0x00
[D][sensor:125]: 'SenseAir CO2 Value': Sending state 539.00000 ppm with 0 decimals of accuracy
[I][ota:102]: Boot seems successful, resetting boot loop counter.
[D][senseair:058]: SenseAir Received CO₂=552ppm Status=0x00
[D][sensor:125]: 'SenseAir CO2 Value': Sending state 552.00000 ppm with 0 decimals of accuracy
[D][senseair:058]: SenseAir Received CO₂=560ppm Status=0x00
[D][sensor:125]: 'SenseAir CO2 Value': Sending state 560.00000 ppm with 0 decimals of accuracy`

Here's my esphome device code as well
`substitutions:
  devicename: s8test
  friendly_name: s8test
  
esphome:
  name: $devicename
  platform: ESP8266
  board: nodemcuv2

wifi:
  networks:
    - ssid: !secret ssid1
      password: !secret ssid1_password
  domain: .local
  use_address: 10.0.10.193
  fast_connect: True
  power_save_mode: none

api:
ota:

web_server:
logger:
  baud_rate: 0

uart:
  - id: uart_one    
    rx_pin: RX
    tx_pin: TX
    baud_rate: 9600
  - id: uart_two
    rx_pin: D1
    tx_pin: D2
    baud_rate: 9600

sensor:
  - platform: senseair
    id: s8high
    co2:
      name: "SenseAir CO2 Value"
    update_interval: 60s
    uart_id: uart_one

#  - platform: senseair
#    co2:
#      name: "SenseAir CO2 Value2"
#    update_interval: 30s
#    uart_id: uart_two
    
button:
  - platform: template
    name: Calibration
    id: my_button
    on_press:
      then:
        - senseair.background_calibration: s8high
        - delay: 120s
        - senseair.background_calibration_result: s8high`

broekieman, 2022-02-05:
I used your code for mine but don't know how/where to push the calibration button.

nmaggioni, 2022-02-05:
@jcastro The sensor itself refuses to calibrate in unstable CO2 conditions, 550 to 558 ppm may be a significant enough variation to trigger that safeguard. If possible, try to put outside or in front of an open window and let it stabilize its readings in fresh air for a couple of minutes before trying again.

You should also wait way less time before checking the calibration result (the datasheet reports "at least 1 lamp cycle" or something along those lines); in the official docs nothing guarantees that the registries that hold the result will stay valid for much longer.

@broekieman From the HomeAssistant UI: https://esphome.io/components/button/index.html

broekieman, 2022-02-05:
> @jcastro The sensor itself refuses to calibrate in unstable CO2 conditions, 550 to 558 ppm may be a significant enough variation to trigger that safeguard. If possible, try to put outside or in [...]

It's now 6.164 ppm, so much to high. Does it calibrato to 400 ppm? I played a lot with this sensor and it's not stable anymore. When i calibrate it with the original software which you can find here: 

https://senseair.com/media/2132/uip5_0_0_3_42_setup.exe

I can calibrate it but when i disconnect and connect to an esp board the values are many times not hold.
I will put it outside for a while and click the button to see what's happening.

nmaggioni, 2022-02-05:
Such high values indicate that it has been calibrated incorrectly, it indeed resets to the 400ppm baseline when asked to do so. Put in in fresh air, retrigger calibration, and it should fix itself.

quizzical86, 2022-02-05:
@broekieman these sensors have to be handled with care or they might become unreliable. I dropped one and since then it the measurement went up and down by 50ppm randomly. Another one got a bit wet and started smoking, didn't survive ;-)

broekieman, 2022-02-05:
> Trying on esphome 2022.1.2 I'm getting this error. I have waited way more than one lamp timing on the sensor!
> 
> ```
> [D][senseair:082]: SenseAir Starting background calibration�
> [D][senseair:058]: SenseAir Received CO₂=550ppm Status=0x00�
> [D][sensor:125]: 'SenseAir CO2 Value': Sending state 550.00000 ppm with 0 decimals of accuracy�
> [D][senseair:058]: SenseAir Received CO₂=558ppm Status=0x00�
> [D][sensor:125]: 'SenseAir CO2 Value': Sending state 558.00000 ppm with 0 decimals of accuracy�
> [D][senseair:090]: SenseAir Requesting background calibration result�
> [D][senseair:106]: SenseAir Result=NOT_OK (fe0302 0000 ac50)�
> [D][senseair:058]: SenseAir Received CO₂=539ppm Status=0x00�
> ```
> 
> Here's my esphome device code as well
> 
> ```
> substitutions:
>   devicename: s8test
>   friendly_name: s8test
>   
> esphome:
>   name: $devicename
>   platform: ESP8266
>   board: nodemcuv2
> 
> wifi:
>   networks:
>     - ssid: !secret ssid1
>       password: !secret ssid1_password
>   domain: .local
>   use_address: 10.0.10.193
>   fast_connect: True
>   power_save_mode: none
> 
> 
> api:
> ota:
> 
> web_server:
> logger:
>   baud_rate: 0
> 
> uart:
>   - id: uart_one    
>     rx_pin: RX
>     tx_pin: TX
>     baud_rate: 9600
>   - id: uart_two
>     rx_pin: D1
>     tx_pin: D2
>     baud_rate: 9600
> 
> sensor:
>   - platform: senseair
>     id: s8high
>     co2:
>       name: "SenseAir CO2 Value"
>     update_interval: 60s
>     uart_id: uart_one
> 
> #  - platform: senseair
> #    co2:
> #      name: "SenseAir CO2 Value2"
> #    update_interval: 30s
> #    uart_id: uart_two
>     
> button:
>   - platform: template
>     name: Calibration
>     id: my_button
>     on_press:
>       then:
>         - senseair.background_calibration: s8high
>         - delay: 120s
>         - senseair.background_calibration_result: s8high
> ```

Where can i find this information?

jcastro, 2022-02-08:
> button:
>   - platform: template
>     name: Calibration
>     id: my_button
>     on_press:
>       then:
>         - senseair.background_calibration: s8high
>         - delay: 120s
>         - senseair.background_calibration_result: s8high

I have tried a lot of different things, I'm always getting claibration NOT_OK. I ended up triggering the hardware calibration by bridging S1. Thanks!