esphome/issues #2445: iBeacon UUID logging and documentation appear backwards to what actually works.

Scott8586, 2021-09-19:
### The problem

iBeacon UUID shown in verbose logging, and in document examples are backwards to what actually works

Using verbose logging I get the following for information.

```
[12:23:43][VV][esp32_ble_tracker:402]: Parse Result:
[12:23:43][VV][esp32_ble_tracker:419]:   Address: DD:33:0A:11:17:DD (PUBLIC)
[12:23:44][VV][esp32_ble_tracker:421]:   RSSI: -64
[12:23:44][VV][esp32_ble_tracker:422]:   Name: 'Scotts_Charm'
[12:23:44][VV][esp32_ble_tracker:430]:   Ad Flag: 6
[12:23:44][VV][esp32_ble_tracker:436]:   Manufacturer data: 02.15.42.6C.75.65.43.68.61.72.6D.42.65.61.63.6F.6E.83.0E.FE.13.55.C5 (23)
[12:23:44][VV][esp32_ble_tracker:439]:     iBeacon data:
[12:23:44][VV][esp32_ble_tracker:440]:       UUID: 836E6F63-6165-426D-7261-684365756C42
[12:23:44][VV][esp32_ble_tracker:441]:       Major: 3838
[12:23:44][VV][esp32_ble_tracker:442]:       Minor: 4949
[12:23:44][VV][esp32_ble_tracker:443]:       TXPower: -59
```

and this UUID is of a similar format (hex 4-2-2-6) to the documentation, but what really works in the configuration is:

```
binary_sensor:
  # Presence based on  iBeacon UUID
  - platform: ble_presence
#    ibeacon_uuid: '836E6F63-6165-426D-7261-684365756C42'
    ibeacon_uuid: '426C75654368-6172-6D42-6561-636F6E83'
    name: "Scotts Charm iBeacon"
```

I'm just a novice in bluetooth, so I don't know which order is actually correct, but I'm used to seeing the latter/second/what works,  in other BTLE apps, tools, scanners etc.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.5

### What platform are you using?

ESP32

### Board

WROOM-32

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
binary_sensor:
  # Presence based on MAC address
  - platform: ble_presence
#    ibeacon_uuid: '836E6F63-6165-426D-7261-684365756C42' # this doesn't work, backwards
    ibeacon_uuid: '426C75654368-6172-6D42-6561-636F6E83'
    name: "Scotts Charm iBeacon"
```
```

deathbrigner, 2022-01-14:
Same here too. Just burnt 2 hours trying to find what's wrong. There's a big/small endian conversion somewhere and it flips the bytes of the UUID in the VV log. Here is my correct and flipped UUIDs :
'426C7565-4368-6172-6D42-6561636F6E73' <-> '736E6F63-6165-426D-7261-684365756C42'

rpatel3001, 2022-01-22:
The loop at https://github.com/esphome/esphome/blob/a79c6aa9e0d8831349fbb22e7158a29191654631/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp#L463 should be counting up, not down.

wjtje, 2022-08-25:
The situation has changed a bit, it's not completely backward anymore.

The esphome log shows the following information:

```
[21:53:30][VV][esp32_ble_tracker:542]:     iBeacon data:
[21:53:30][VV][esp32_ble_tracker:543]:       UUID: 6576C300-6CCF-B984-EF49-D990C242DC80
[21:53:30][VV][esp32_ble_tracker:544]:       Major: 100
[21:53:30][VV][esp32_ble_tracker:545]:       Minor: 1
[21:53:30][VV][esp32_ble_tracker:546]:       TXPower: -59
```

But the actual UUID is `80dc42c2-90d9-49ef-84b9-cf6c00c37665`

This can be fixed by a small change in the `ESPBTUUID::from_raw()` function,

```diff
ESPBTUUID ESPBTUUID::from_raw(const uint8_t *data) {
  ESPBTUUID ret;
  ret.uuid_.len = ESP_UUID_LEN_128;
  for (size_t i = 0; i < ESP_UUID_LEN_128; i++)
-    ret.uuid_.uuid.uuid128[i] = data[i];
+    ret.uuid_.uuid.uuid128[i] = data[15 - i];
  return ret;
}
```

But the `ble_presence` binary sensor works with or without this change.

rpatel3001, 2025-07-17:
i believe this was fixed by https://github.com/esphome/esphome/pull/7332

closed 2025-07-19 by bdraco