esphome/issues #2446: 2021.9.0 compile not working on Windows

jamesmyatt, 2021-09-19:
### The problem

When I compile any config (e.g. see below) on Windows using the pip install, then I get warnings about "Could not determine Arduino framework version". Furthermore, with the "esp8266_pwm" component, there is an error "#error ESP8266 PWM requires at least arduino_version 2.4.0".

Have confirmed that these configs work on 2021.8.2 pip Windows and 2021.9.0 docker.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

esphome

### Example YAML snippet

```yaml
esphome:
  name: esphome-0
  platform: ESP8266
  board: esp12e

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESPHome-0"
    password: !secret ap_password

output:
  - platform: esp8266_pwm
    pin:
      number: GPIO2
      inverted: true
    frequency: 1000 Hz
    id: led_output
light:
  - platform: monochromatic
    output: led_output
    name: "esphome_led"
```

### Anything in the logs that might be useful for us?
[... 111 lines cut ...]
src/esphome/core/macros.h:43:2: warning: #warning "Could not determine Arduino framework version, update esphome/core/macros.h!" [-Wcpp]
Compiling .pioenvs\esphome-0\src\esphome\components\light\esp_range_view.cpp.o
 #warning "Could not determine Arduino framework version, update esphome/core/macros.h!"
  ^
In file included from src\esphome\components\esp8266_pwm\esp8266_pwm.cpp:2:0:
src/esphome/core/macros.h:43:2: warning: #warning "Could not determine Arduino framework version, update esphome/core/macros.h!" [-Wcpp]
 #warning "Could not determine Arduino framework version, update esphome/core/macros.h!"
  ^
src\esphome\components\esp8266_pwm\esp8266_pwm.cpp:7:2: error: #error ESP8266 PWM requires at least arduino_version 2.4.0
 #error ESP8266 PWM requires at least arduino_version 2.4.0
  ^
In file included from src/esphome/core/log.h:10:0,
                 from src\esphome\components\light\automation.cpp:2:
src/esphome/core/macros.h:43:2: warning: #warning "Could not determine Arduino framework version, update esphome/core/macros.h!" [-Wcpp]
 #warning "Could not determine Arduino framework version, update esphome/core/macros.h!"
  ^
Compiling .pioenvs\esphome-0\src\esphome\components\light\light_call.cpp.o
*** [.pioenvs\esphome-0\src\esphome\components\esp8266_pwm\esp8266_pwm.cpp.o] Error 1
In file included from src/esphome/core/log.h:10:0,
                 from src\esphome\components\light\light_call.cpp:3:
src/esphome/core/macros.h:43:2: warning: #warning "Could not determine Arduino framework version, update esphome/core/macros.h!" [-Wcpp]
 #warning "Could not determine Arduino framework version, update esphome/core/macros.h!"
  ^
============================================== [FAILED] Took 6.45 seconds ==============================================
```

oxan, 2021-09-20:
Can you post the contents of the `.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_version.h` file in your user directory? I don't have a Windows machine to test this...

OttoWinter, 2021-09-20:
Hmm, looks like your platformio setup is in a weird state.

Please try running `pio platform update`

closed 2021-10-09 by jamesmyatt