esphome/issues #2448: No way to set pulldown mode for GPIO

cab404, 2021-09-20:
### The problem

I am using some fussy relays, which are not satisfied with being digitallyWritten to. They actually need me to switch into input with pulldown mode, thus prompting me to write some unsightly code like this:
```yaml
switch:
  - platform: template
    name: "Pick up phone"
    id: t_pickup
    lambda: |-
      return !digitalRead(12);
    turn_on_action:
      - lambda: |-
          pinMode(12, OUTPUT);
          digitalWrite(12, LOW);
    turn_off_action:
      - lambda: |
          pinMode(12, INPUT);
```

Would be nice to have a flag to make "off" state actually use "input_pulldown" mode on gpio.

### Which version of ESPHome has the issue?

1.16

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

cab404, 2021-09-20:
Actually, I use nixos installation of esphome, but there's no way to choose "Other"

OttoWinter, 2021-09-20:
ESP8266 only has a pulldown resistor on GPIO16 (https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html), but pulldown mode is supported for that with `INPUT_PULLDOWN`

However, it looks like what you're trying to do here is an "open drain" output, which is also supported via `OUTPUT_OPEN_DRAIN`

cab404, 2021-09-20:
> `OUTPUT_OPEN_DRAIN`

Oh, thank you!
I guess then it would be nice to add that to wiki, as there's no link to [Pin Schema](https://esphome.io/guides/configuration-types.html?highlight=output_open_drain) in GPIO sections

closed 2022-01-19 by cab404