esphome/issues #2450: ESPHome will not start on docker Raspberry Pi

cleggton, 2021-09-20:
### The problem

I updated my working version of ESPHome from 2021.8.x to 2021.9.x and the docker container fails

### Which version of ESPHome has the issue?

2021.9.x

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

D3 mini

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
INFO Starting dashboard web server on port 6052 and configuration dir /config...,
Traceback (most recent call last):,
  File "/usr/local/bin/esphome", line 33, in <module>,
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')()),
  File "/usr/src/app/esphome/__main__.py", line 770, in main,
    return run_esphome(sys.argv),
  File "/usr/src/app/esphome/__main__.py", line 739, in run_esphome,
    return PRE_CONFIG_ACTIONS[args.command](args),
  File "/usr/src/app/esphome/__main__.py", line 401, in command_dashboard,
    return dashboard.start_web_server(args),
  File "/usr/src/app/esphome/dashboard/dashboard.py", line 854, in start_web_server,
    app.listen(args.port),
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 2109, in listen,
    server.listen(port, address),
  File "/usr/local/lib/python3.9/dist-packages/tornado/tcpserver.py", line 152, in listen,
    self.add_sockets(sockets),
  File "/usr/local/lib/python3.9/dist-packages/tornado/tcpserver.py", line 165, in add_sockets,
    self._handlers[sock.fileno()] = add_accept_handler(,
  File "/usr/local/lib/python3.9/dist-packages/tornado/netutil.py", line 246, in add_accept_handler,
    io_loop = IOLoop.current(),
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 263, in current,
    loop = asyncio.get_event_loop(),
  File "/usr/lib/python3.9/asyncio/events.py", line 639, in get_event_loop,
    self.set_event_loop(self.new_event_loop()),
  File "/usr/lib/python3.9/asyncio/events.py", line 659, in new_event_loop,
    return self._loop_factory(),
  File "/usr/lib/python3.9/asyncio/unix_events.py", line 54, in __init__,
    super().__init__(selector),
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 55, in __init__,
    super().__init__(),
  File "/usr/lib/python3.9/asyncio/base_events.py", line 397, in __init__,
    self._clock_resolution = time.get_clock_info('monotonic').resolution,
PermissionError: [Errno 1] Operation not permitted,
Exception ignored in: <function BaseEventLoop.__del__ at 0x76233538>,
Traceback (most recent call last):,
  File "/usr/lib/python3.9/asyncio/base_events.py", line 681, in __del__,
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self),
  File "/usr/lib/python3.9/asyncio/base_events.py", line 419, in __repr__,
    f'closed={self.is_closed()} debug={self.get_debug()}>',
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1909, in get_debug,
    return self._debug,
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
```

### Additional information

I have rolled back to 2021.8.2 and all starts again

procet87, 2021-09-21:
the same for me!

vijaygill, 2021-09-21:
And same here too. I had to roll back to 2021.8.2 to get things working again.

dougle03, 2021-09-22:
Do the following: It worked for me.
`sudo apt update`
`sudo apt upgrade`
`sudo reboot`
after reboot
`sudo wget http://ftp.us.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.4.4-1~bpo10+1_armhf.deb`
`sudo dpkg -i libseccomp2_2.4.4-1~bpo10+1_armhf.deb`

See here for more info: [https://docs.linuxserver.io/faq](url)

javierguerragiraldez, 2021-10-06:
updating libseccomp didn't work for me:
- raspberry pi 3B+ (buster)
- libseccomp2 2.4.4-1~bpo10+1
- docker.io 18.09.1+dfsg1-7.1+rpi1+rpt1

still getting the same `AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'` in the logs