esphome/issues #2451: ESP8266 generic 1M boot loop on 2021.10.0.dev0

reaper7, 2021-09-20:
### The problem

Wall WiFi Switch do not wake up after flashing last dev (boot loop)

I opened the device and I tried to upload via uart, effect is the same.
I tried the previous version 2021.9.1 and the device booted up properly.

I've never had a situation like this...
I uploaded the first software using tuya-convert long time ago.
I always tested development versions and I have never had to demolish the case to upload the software :)

### Which version of ESPHome has the issue?

2021.10.0.dev0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
# Substitutions
substitutions:
  # Device Names
  device_name: "plug1175"
  friendly_name: "Washer"
  sub_name: ""
  # Icon
  # https://materialdesignicons.com/
  # https://www.esphome.app/docs/icons
  # main_icon: "dishwasher"
  main_icon: "mdi:washing-machine"
  # Default Relay State
  # Aka: `restore_mode` in documentation
  # Options: `RESTORE_DEFAULT_OFF`, `RESTORE_DEFAULT_ON`, `ALWAYS_ON` & `ALWAYS_OFF`
  default_state: "RESTORE_DEFAULT_OFF"
  
# Activity State Threshold
# Threshold (number) that the device will change from `Idle` to `Active` if power is greater than or equal to
globals:
  - id: activity_threshold
    type: int
    restore_value: no
    initial_value: '3'

# CORE CONFIG
# https://esphome.io/components/esphome.html
esphome:
  name: "${device_name}"
  platform: ESP8266
  #board: esp8285
  board: esp01_1m
  esp8266_restore_from_flash: true

# WIFI
# https://esphome.io/components/wifi.html
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  reboot_timeout: !secret reboot_timeout_wifi
  fast_connect: !secret fast_connect_wifi
  manual_ip:
    static_ip: !secret plug1175
    gateway: !secret gateway
    subnet: !secret subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
# https://esphome.io/components/logger.html
logger:
  # level: DEBUG
  level: INFO
  # logs:
  #  i2c.component: DEBUG
  # baud_rate: 115200
  baud_rate: 0  #disable uart log
  # esp8266_store_log_strings_in_flash: false

# Enable Home Assistant API
# https://esphome.io/components/api.html
api:
  reboot_timeout: !secret reboot_timeout_api

# OTA
# https://esphome.io/components/ota.html
ota:

# WEBSERVER
# https://esphome.io/components/web_server.html
web_server:
  #css_url: "http://192.168.0.254:81/esphome/_static/webserver-v1.min.css"
  #js_url: "http://192.168.0.254:81/esphome/_static/webserver-v1.min.js"
  #css_url: ""
  #js_url: ""
  #css_include: "./custom_components/_static/webserver-v1.min.css"
  #js_include: "./custom_components/_static/webserver-v1.min.js"

#time:
#  # https://esphome.io/components/time.html
#  - platform: sntp
#    id: sntp_time
#    timezone: Europe/Warsaw
#    servers: "dhouse.lan, 0.pl.pool.ntp.org, 1.pl.pool.ntp.org"

time:
  # https://esphome.io/components/time.html
  - platform: homeassistant
    id: hass_time
    timezone: Europe/Warsaw

binary_sensor:
  - <<: !include common/binary_sensor/connection_status.config.yaml

  # https://esphome.io/components/binary_sensor/gpio.html
  - platform: gpio
    pin:
      number: GPIO0
#      mode: INPUT_PULLUP
      mode: INPUT
      inverted: True
    name: "${friendly_name} Button"
    on_press:
      - switch.toggle: relay
    internal: True

  # Device Activity State
  # e.g. shows as active if power reading above a certain threshold
  # https://esphome.io/components/text_sensor/template.html
  - platform: template
    name: "${friendly_name} Activity State"
    # icon: "${main_icon}"
    device_class: power
    lambda: |-
      if (id(power).state >= id(activity_threshold)) {
        return true;
      } else {
        return false;
      }
    filters:
      # delayed_on: 20s
      # delayed_on_off: 20s
      delayed_off: 60s

switch:
  - <<: !include common/switch/restart_switch.config.yaml

  # https://esphome.io/components/switch/gpio
  - platform: gpio
    name: "${friendly_name} Relay"
    icon: "mdi:electric-switch"
    pin: GPIO14
    id: relay
    restore_mode: "${default_state}"

# https://esphome.io/components/status_led
status_led:
  pin:
    number: GPIO13
    inverted: yes

sensor:
  - <<: !include common/sensor/uptime.config.yaml
  - <<: !include common/sensor/wifi_signal.config.yaml

  # https://esphome.io/components/sensor/hlw8012.html
  - platform: hlw8012
    sel_pin:
      number: 12
      inverted: true
    cf_pin: 4
    cf1_pin: 5
    current:
      name: "${friendly_name} Current"
      icon: "mdi:gauge"
      unit_of_measurement: A
      accuracy_decimals: 3
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            - 8.51205 -> 8.074389
    voltage:
      name: "${friendly_name} Voltage"
      icon: "mdi:gauge"
      unit_of_measurement: V
      accuracy_decimals: 1
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            - 669.3 -> 231.8
            - 685.8 -> 236.8
    power:
      name: "${friendly_name} Power"
      icon: "mdi:gauge"
      id: power
      unit_of_measurement: W
      accuracy_decimals: 0
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            # - 9.33480 -> 0.0
            # - 303.4 -> 75.0
            # - 365.5 -> 75.0
            - 375.8 -> 75.0
            - 10643.6 -> 1800.0
        - lambda: if (x < (5.71578)) return 0; else return (x - 5.71577);
    energy:
      name: "${friendly_name} Energy"
      icon: "mdi:gauge"
      id: energy
      unit_of_measurement: kWh
      accuracy_decimals: 2
      device_class: energy
      state_class: measurement
      filters:
#        # Multiplication factor from W to kW is 0.001
#        - multiply: 0.001
        # Multiplication factor from W to kW is 0.001 and next multiply by 0.16905 (real energy)
        - multiply: 0.000169205
    change_mode_every: 3
    update_interval: 3s
  # https://esphome.io/components/sensor/total_daily_energy.html
  - platform: total_daily_energy
    name: "${friendly_name} Daily Energy"
    icon: "mdi:counter"
    power_id: power
    device_class: energy
    state_class: measurement
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh
```

### Anything in the logs that might be useful for us?

```txt
[00]
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[00]
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[00]
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[00]
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

mmakaay, 2021-09-21:
A change has been done to fix the reboot loop by @OttoWinter
See https://github.com/esphome/esphome/pull/2362

Note that there are still some issues with OTA updates, so for the time being it might be wise to avoid the bleeding edge dev branch. The changes that are currently being done are for the greater good, but are quite extended. There might be more issues that weren't found yet.

reaper7, 2021-09-22:
@mmakaay - Yes, I tested this PR a few hours ago.
ESP actually starts (flashing via uart),
wifi starts up, web interface is available,
LED and relay do not work (but I'll check it later today).

When trying to upload via OTA, stops at 1%, esp restarts and takes a memory dump.

P.S. the plug is still open(the casing does not look very good), PCB soldered to the converter, so I'm going to test the next fixes :)

mmakaay, 2021-09-22:
All fixes that I provided have been merged into dev.
This includes a fix for the 1% reboot + memory dump issue.
So if you flash one more time using the serial connection, then subsequent flashing operations should work using OTA.

reaper7, 2021-09-22:
so, almost everything works fine,
booting, wifi, web interface, ota, binary sensor(gpio)

but not OUTPUTS.
does not work:
switch gpio (relay and diode)
status led

logs show that the switch is toggling 
but the corresponding esp pins do not change state

mmakaay, 2021-09-22:
Can you create a new issue for that? I'm afraid that it would get lost, buried in this issue.
I'm currently not at my pc, but I'll setup a test when I get back home.

reaper7, 2021-09-22:
current dev resolve booting problem

closed 2021-09-22 by reaper7