esphome/issues #2456: HM3301 (particulate matter sensor) "Checksum Validation Failed" after sensor exits sleep mode.

Teleportist, 2021-09-21:
### The problem

In order to preserve the life of this sensor, which uses a laser having a ~2 year lifetime, I've set up a GPIO switch to pull the sensors' set pin low, sending it in to low level sleep mode. As descrived [here](https://wiki.seeedstudio.com/Grove-Laser_PM2.5_Sensor-HM3301/).

However, when the sensor exits sleep (after returning 3.3v to the set pin) I'm no longer receiving readings. Instead the error message '[W][hm3301.sensor:046]: Checksum validation failed' repeats at the update interval.

I've tried setting up another GPIO to the reset pin of the sensor board. But this has no effect. Only cycling the power to the ESP32 returns normal operation.

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.6 (OS 6.3)

### What platform are you using?

ESP32

### Board

esp32dev (ESP-WROOM-32 38 PIN Development Board)

### Component causing the issue

HM3301 (The Grove - Laser PM2.5 Sensor)

### Example YAML snippet

```yaml
substitutions:
  disp_name: Air Quality
#  update_time: 60s

esphome:
  name: airmon
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 
    gateway: 
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Airmon Fallback Hotspot"
#    password: "***"
# Enable logging
logger:

# Enable Home Assistant API.
api:

ota:
  password: !secret ota_password

i2c:

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
    
  - platform: hm3301
    pm_1_0:
      name: "PM1.0"
    pm_2_5:
      name: "PM2.5"
      #force_update: true
    pm_10_0:
      name: "PM10.0"
    aqi:
      name: "AQI"
      calculation_type: "CAQI"
    address: 0x40
    update_interval: 60s
      
switch:
  - platform: gpio
    pin:
      number: 23
      inverted: true
    id: airmon_sleep
    name: "Sleep"
    
  - platform: gpio
    pin:
      number: 19
      inverted: true
    id: airmon_reset
    name: "Reset"    
  
#interval:
#  - interval: 180s
#    then:
#      - switch.turn_off: airmon_sleep
#      - delay: 90s
#      - switch.turn_on: airmon_sleep
```

### Anything in the logs that might be useful for us?

```txt
[00:07:29][D][switch:021]: 'Sleep' Turning ON.
[00:07:29][D][switch:045]: 'Sleep': Sending state ON
[00:08:15][D][sensor:131]: 'Air Quality WiFi': Sending state -59.00000 dBm with 0 decimals of accuracy
[00:08:19][W][hm3301.sensor:040]: Read result failed
[00:09:15][D][sensor:131]: 'Air Quality WiFi': Sending state -57.00000 dBm with 0 decimals of accuracy
[00:09:19][W][hm3301.sensor:040]: Read result failed
[00:09:21][D][switch:025]: 'Sleep' Turning OFF.
[00:09:21][D][switch:045]: 'Sleep': Sending state OFF
[00:10:15][D][sensor:131]: 'Air Quality WiFi': Sending state -58.00000 dBm with 0 decimals of accuracy
[00:10:19][W][hm3301.sensor:046]: Checksum validation failed
```

Teleportist, 2021-09-24:
Thinking about this another way, is there a way I could force the esp32 to rescan the bus after waking up the sensor? I could the add this command in to my interval.

closed 2022-02-06 by github-actions[bot]

viotemp1, 2022-03-02:
add IDs to I2C and hm3301 
add 
      - lambda: |-
          id(${roomname}_i2c_bus_b).setup();
          id(${roomname}_hm3301).setup();
after - switch.turn_off: airmon_sleep.