esphome/issues #2457: getting cryptography error message when trying to show logs wirelessly

hydazz, 2021-09-22:
### The problem

Every since ESPHome 2021.9.0 when I try to access my devices logs wirelessly I'm getting the following error:

```python
INFO Reading configuration /config/house-heater.yaml...
Traceback (most recent call last):
  File "/usr/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/esphome/__main__.py", line 770, in main
    return run_esphome(sys.argv)
  File "/usr/lib/python3.9/site-packages/esphome/__main__.py", line 757, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/lib/python3.9/site-packages/esphome/__main__.py", line 339, in command_logs
    return show_logs(config, args, port)
  File "/usr/lib/python3.9/site-packages/esphome/__main__.py", line 259, in show_logs
    from esphome.components.api.client import run_logs
  File "/usr/lib/python3.9/site-packages/esphome/components/api/client.py", line 6, in <module>
    from aioesphomeapi import APIClient, ReconnectLogic, APIConnectionError, LogLevel
  File "/usr/lib/python3.9/site-packages/aioesphomeapi/__init__.py", line 2, in <module>
    from .client import APIClient
  File "/usr/lib/python3.9/site-packages/aioesphomeapi/client.py", line 68, in <module>
    from .connection import APIConnection, ConnectionParams
  File "/usr/lib/python3.9/site-packages/aioesphomeapi/connection.py", line 11, in <module>
    from noise.connection import NoiseConnection  # type: ignore
  File "/usr/lib/python3.9/site-packages/noise/connection.py", line 6, in <module>
    from noise.backends.default import noise_backend
  File "/usr/lib/python3.9/site-packages/noise/backends/default/__init__.py", line 1, in <module>
    from .backend import DefaultNoiseBackend
  File "/usr/lib/python3.9/site-packages/noise/backends/default/backend.py", line 1, in <module>
    from noise.backends.default.ciphers import ChaCha20Cipher, AESGCMCipher
  File "/usr/lib/python3.9/site-packages/noise/backends/default/ciphers.py", line 3, in <module>
    from cryptography.hazmat.primitives.ciphers.aead import AESGCM, ChaCha20Poly1305
  File "/config/.local/lib/python3.9/site-packages/cryptography/hazmat/primitives/ciphers/aead.py", line 10, in <module>
    from cryptography.hazmat.backends.openssl import aead
  File "/config/.local/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 6, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/config/.local/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 113, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "/config/.local/lib/python3.9/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 14, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: Error relocating /config/.local/lib/python3.9/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so: FIPS_mode_set: symbol not found
```
This is running in an alpine edge docker container so I can assume that this is an incompatibility between the pip version of cryptography and the alpine version of openssl and if I wait a little cryptography will get updated as well, but I thought I would ask here to see if this is a known issue/if it is an issue with esphome - and if I can fix it myself somehow

### Which version of ESPHome has the issue?

2021.9.x

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

dougle03, 2021-09-22:
Is Alpine `Buster`?
If so, have a look at this - ```https://docs.linuxserver.io/faq```
It solved my dev problem.

Shulyaka, 2021-10-12:
I recently had the same error after I updated my alpine installation. The reason was that openssl-dev package had previously referred to openssl 1.1 version, but after the upgrade it began to refer to openssl 3.0. Fixed by removing `openssl-dev`, installing `openssl1.1-compat-dev` and recompiling `cryptography`.
Something like this:
```
apk del openssl-dev
apk add openssl1.1-compat-dev
pip uninstall cryptography
pip install cryptography==3.4.8 --no-cache-dir
```

hydazz, 2021-10-17:
@Shulyaka I managed to fix this by deleting all the folders in the ESPHome config directory. possible that ESPHome is expecting another version of OpenSSL or something and deleting these folders forces it to relook for the newer version?

Shulyaka, 2021-10-17:
No idea, sorry

closed 2022-11-04 by hydazz