esphome/issues #2458: Problem with binary light automation after upgrading to 2021.9.1

wasyl00, 2021-09-22:
### The problem

My binary light does not switch off according to automation. This broke after latest update to 2021.9.1. This happens on two different boards esp8285 and ESP8266.

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.7

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

Binary Light / Automation

### Example YAML snippet

```yaml
light:
  # 3D Printer light 1
  - platform: monochromatic
    name: "3D printer light"
    output: output_light_1
    id: spot_light
    on_turn_on:
      then:
       - light.turn_on: spot_status_led
       - logger.log: "Status led on!"
    on_turn_off:
      then:
       - light.turn_off: spot_status_led
       - logger.log: "Status led off!"
  - platform: binary
    name: "Spotlight Led"
    output: output_component1
    internal: true
    id: spot_status_led
  - platform: binary
    name: "Fan Led"
    output: output_component2
    internal: true
    id: fan_status_led

output:
  - platform: esp8266_pwm
    id: output_light_1
    pin: D6
  - platform: gpio
    pin: D4
    id: output_component1
  - platform: gpio
    pin: D3
    id: output_component2
```

### Anything in the logs that might be useful for us?

```txt
When I switch on the button after boot the status light switches on but it does not switch off after another press. This worked before but stopped after last update. 

The below happen on toggling light on then off, there is no switch off action for my status led in logger:

[09:55:53][D][binary_sensor:036]: 'Spot Light Switch': Sending state ON
[09:55:54][D][binary_sensor:036]: 'Spot Light Switch': Sending state OFF
[09:55:54][D][main:540]: Single Short Clicked
[09:55:54][D][light:035]: '3D printer light' Setting:
[09:55:54][D][light:046]:   State: ON
[09:55:54][D][light:084]:   Transition length: 1.0s
[09:55:54][D][light:035]: 'Spotlight Led' Setting:
[09:55:54][D][main:566]: Status led on!
[09:55:55][D][binary_sensor:036]: 'Spot Light Switch': Sending state ON
[09:55:55][D][binary_sensor:036]: 'Spot Light Switch': Sending state OFF
[09:55:55][D][main:540]: Single Short Clicked
[09:55:55][D][light:035]: '3D printer light' Setting:
[09:55:55][D][light:046]:   State: OFF
[09:55:55][D][light:084]:   Transition length: 1.0s
```

Protoncek, 2021-09-22:
Same here. Using Sonoff D1 dimmer, version 2021.9.x doesn't turn off lamp, but it stays at appr. 1%. Going back to 2021.8.x solves the problem.

Spirituss, 2021-09-23:
PWM stops working correctly on 2021.9.
Here is my config:
```
output:
  - platform: esp8266_pwm
    frequency: 1000 Hz
    pin: D2
    id: id_relay    
#===============================================================================
light:
  - platform: monochromatic 
    name: ${device_name}
    id: id_light
    output: id_relay
```

**There is no Brightness parameter** in the light component in Home Assistant. I updated to esphome 17.2 and it came back again.

oxan, 2021-09-24:
@Spirituss please open your own issue for your problems, and don't hijack issues that are unrelated to yours.

Spirituss, 2021-09-24:
> @Spirituss please open your own issue for your problems, and don't hijack issues that are unrelated to yours (and please don't @-me, I already have way too much GitHub notifications and it doesn't increase the chance I'll look at it).

Ок, but I supposed these two symptoms concerned with the same problem.

oxan, 2021-09-24:
> Ок, but I supposed these two symptoms concerned with the same problem.

No, they aren't.

Protoncek, 2021-09-28:
I just flashed my Sonoff dimmer with latest 2021.9.2, but problem is still there - no difference. Going back to 2021.8 again...

wasyl00, 2021-09-28:
yes tried too no change with 2021.9.2

wasyl00, 2021-10-09:
Installed and flashed 2021.9.3 - no change.