esphome/issues #2459: Waveshare no updates after PR2366 :dev

dougle03, 2021-09-22:
### The problem

2.9" ePaper (Waveshare) v2.1 no longer updates at all.
Several reboots of the ESP.
No initial full refresh is being performed.
This is a new issue, but could be related to the refresh issue #2334

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

Waveshare 2.9" ePaper v2.1 display

### Example YAML snippet

```yaml
spi:
  clk_pin: D0
  mosi_pin: D1 #DIN
  #miso_pin: D7 #not connected
display:
  - platform: waveshare_epaper
    cs_pin: D2
    dc_pin: D6
    busy_pin: D7
    reset_pin: D5
    model: 2.90inv2
    id: mydisplay
    update_interval: 15s
    full_update_every: 1
    rotation: 90
    #brightness: 100%
    lambda: |-
      it.printf(127, 0, id(font3), TextAlign::TOP_RIGHT, "ePaper1");
```
```

### Anything in the logs that might be useful for us?

```txt
Logs look normal.
```

### Additional information

does work with 2021.9.0 albeit with refreshing issues. see issue #2334 
The work around for that was to set the full_refresh: to 1 - The downside is a full refresh every update.
This is a regression.

dougle03, 2022-01-21:
It's been a while; I need to find time this weekend to re-check this.

emdekext, 2022-05-21:
please fix this asap, thanks. If I can do anything to help (testing) let me know.
There's also another open issue, it seems exactly the same problem: [https://github.com/esphome/issues/issues/2334](url)