esphome/issues #2460: Wemos D1 mini rebooting after update

drakecoldwinter, 2021-09-22:
### The problem

My ESPHome was an old version (1.16). I have several Wemos D1 mini programmed and working. I updated ESPHome to version v2021.10.0-dev I recompiled one of the nodes and uploaded it. The wemos went into a reboot loop (exception 28). To test I decided to create a new configuration (see below)

I tested this new configuration on 2 different wemos. Both present exactly the same behaviour and produce the same error (exception 28)

Currently I refrain uploading to the other wemos as I'm pretty sure all will stop working. I will test a stable release instead of the dev release as well

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N\A

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

N\A

### Example YAML snippet

```yaml
esphome:
  name: testa
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "07fb4eb108974f494e71f8f8eb164878"

wifi:
  ssid: "xxxxxxxxxx"
  password: "xxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testa Fallback Hotspot"
    password: "oYBEPlL6XZGh"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
[09:55:52]
[09:55:52] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[09:55:52]
[09:55:52]load 0x4010f000, len 3584, room 16 
[09:55:52]tail 0
[09:55:52]chksum 0xb0
[09:55:52]csum 0xb0
[09:55:52]v2843a5ac
[09:55:52]~ld
[09:55:52][I][logger:216]: Log initialized
[09:55:52]
[09:55:52]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[09:55:52]
[09:55:52]Exception (28):
[09:55:52]epc1=0x40206660 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[09:55:52]
[09:55:52]>>>stack>>>
[09:55:52]
[09:55:52]ctx: cont
[09:55:52]sp: 3ffffd50 end: 3fffffc0 offset: 0190
[09:55:52]3ffffee0:  3ffef024 3fff057c 3fff057c 40206653  
[09:55:52]3ffffef0:  3ffef024 3fffff80 3ffe8c0f 4020ae60
```

reaper7, 2021-09-22:
try to reinstall dev package, a lot of patches were released this afternoon
https://github.com/esphome/issues/issues/2451

Spirituss, 2021-09-23:
Similar behaviour is here. I updated esp plug firmware to 2021.9 and plug started to reboot every 2-5 minutes, I checked it on some other plugs - the same situation.

oxan, 2021-09-23:
@Spirituss if you're using 2021.9, please open a separate issue; and it would be helpful if you could try to capture the serial logs when it reboots.

drakecoldwinter, 2021-09-23:
I downgraded to v2021.9.1 stable release and the problem is gone. I just recompiled and deployed the nodes and they went back to working properly.

Spirituss, 2021-09-24:
> @Spirituss if you're using 2021.9, please open a separate issue; and it would be helpful if you could try to capture the serial logs when it reboots.

The same behaviour was on previous versions. On 2021.9 it just becomes more intensive. Please, see also https://github.com/esphome/issues/issues/2464

closed 2022-02-06 by github-actions[bot]