esphome/issues #2461: ESP8266 generic 1M OUTPUTS pins are not working on 2021.10.0.dev0

reaper7, 2021-09-22:
### The problem

Physical ESP pins (generally OUTPUTS) don't change states.
in a sample yaml (below) does not work:
switch gpio (relay and led)
status led

logs show that the switch is toggling
but the corresponding esp pins do not change state.

dev.zip 2021.09.22 14:07

### Which version of ESPHome has the issue?

2021.10.0.dev0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

esp01_1m, esp8285

### Example YAML snippet

```yaml
# Substitutions
substitutions:
  # Device Names
  device_name: "plug1175"
  friendly_name: "Washer"
  sub_name: ""
  # Icon
  # https://materialdesignicons.com/
  # https://www.esphome.app/docs/icons
  # main_icon: "dishwasher"
  main_icon: "mdi:washing-machine"
  # Default Relay State
  # Aka: `restore_mode` in documentation
  # Options: `RESTORE_DEFAULT_OFF`, `RESTORE_DEFAULT_ON`, `ALWAYS_ON` & `ALWAYS_OFF`
  default_state: "RESTORE_DEFAULT_OFF"
  
# Activity State Threshold
# Threshold (number) that the device will change from `Idle` to `Active` if power is greater than or equal to
globals:
  - id: activity_threshold
    type: int
    restore_value: no
    initial_value: '3'

# CORE CONFIG
# https://esphome.io/components/esphome.html
esphome:
  name: "${device_name}"
  platform: ESP8266
  #board: esp8285
  board: esp01_1m
  esp8266_restore_from_flash: true

# WIFI
# https://esphome.io/components/wifi.html
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  reboot_timeout: !secret reboot_timeout_wifi
  fast_connect: !secret fast_connect_wifi
  manual_ip:
    static_ip: !secret plug1175
    gateway: !secret gateway
    subnet: !secret subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
# https://esphome.io/components/logger.html
logger:
  # level: DEBUG
  level: INFO
  # logs:
  # baud_rate: 115200
  baud_rate: 0  #disable uart log
  # esp8266_store_log_strings_in_flash: false

# Enable Home Assistant API
# https://esphome.io/components/api.html
api:
  reboot_timeout: !secret reboot_timeout_api

# OTA
# https://esphome.io/components/ota.html
ota:

# WEBSERVER
# https://esphome.io/components/web_server.html
web_server:

time:
  # https://esphome.io/components/time.html
  - platform: homeassistant
    id: hass_time
    timezone: Europe/Warsaw

binary_sensor:
  - <<: !include common/binary_sensor/connection_status.config.yaml

  # https://esphome.io/components/binary_sensor/gpio.html
  - platform: gpio
    pin:
      number: GPIO0
#      mode: INPUT_PULLUP
      mode: INPUT
      inverted: True
    name: "${friendly_name} Button"
    on_press:
      - switch.toggle: relay
    internal: True

  # Device Activity State
  # e.g. shows as active if power reading above a certain threshold
  # https://esphome.io/components/text_sensor/template.html
  - platform: template
    name: "${friendly_name} Activity State"
    # icon: "${main_icon}"
    device_class: power
    lambda: |-
      if (id(power).state >= id(activity_threshold)) {
        return true;
      } else {
        return false;
      }
    filters:
      # delayed_on: 20s
      # delayed_on_off: 20s
      delayed_off: 60s

switch:
  - <<: !include common/switch/restart_switch.config.yaml

  # https://esphome.io/components/switch/gpio
  - platform: gpio
    name: "${friendly_name} Relay"
    icon: "mdi:electric-switch"
    pin: GPIO14
    id: relay
    restore_mode: "${default_state}"

# https://esphome.io/components/status_led
status_led:
  pin:
    number: GPIO13
    inverted: yes

sensor:
  - <<: !include common/sensor/uptime.config.yaml
  - <<: !include common/sensor/wifi_signal.config.yaml

  # https://esphome.io/components/sensor/hlw8012.html
  - platform: hlw8012
    sel_pin:
      number: 12
      inverted: true
    cf_pin: 4
    cf1_pin: 5
    current:
      name: "${friendly_name} Current"
      icon: "mdi:gauge"
      unit_of_measurement: A
      accuracy_decimals: 3
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            - 8.51205 -> 8.074389
    voltage:
      name: "${friendly_name} Voltage"
      icon: "mdi:gauge"
      unit_of_measurement: V
      accuracy_decimals: 1
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            - 669.3 -> 231.8
            - 685.8 -> 236.8
    power:
      name: "${friendly_name} Power"
      icon: "mdi:gauge"
      id: power
      unit_of_measurement: W
      accuracy_decimals: 0
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 0.0 -> 0.0 #Don't care if 0 reading aligns with 0 real Volts since we won't ever measure that
            # - 9.33480 -> 0.0
            # - 303.4 -> 75.0
            # - 365.5 -> 75.0
            - 375.8 -> 75.0
            - 10643.6 -> 1800.0
        - lambda: if (x < (5.71578)) return 0; else return (x - 5.71577);
    energy:
      name: "${friendly_name} Energy"
      icon: "mdi:gauge"
      id: energy
      unit_of_measurement: kWh
      accuracy_decimals: 2
      device_class: energy
      state_class: measurement
      filters:
#        # Multiplication factor from W to kW is 0.001
#        - multiply: 0.001
        # Multiplication factor from W to kW is 0.001 and next multiply by 0.16905 (real energy)
        - multiply: 0.000169205
    change_mode_every: 3
    update_interval: 3s
  # https://esphome.io/components/sensor/total_daily_energy.html
  - platform: total_daily_energy
    name: "${friendly_name} Daily Energy"
    icon: "mdi:counter"
    power_id: power
    device_class: energy
    state_class: measurement
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh
```

paulmonigatti, 2021-09-23:
Potentially fixed by esphome/esphome#2377

reaper7, 2021-09-23:
@paulmonigatti - yes, fixed! works now, tnx!

closed 2021-09-23 by reaper7