esphome/issues #2462: SPI displays not working on ESP32 (ST7920)

jsb5151, 2021-09-22:
### The problem

I tried setting up a ST7920 SPI display on an ESP32 (NodeMCU-32S board) with no success.  Tried changing the CLK/MOSI/CS pins multiple times to no avail; tried both HSPI and VSPI pins as well as other pins such as the ones below.  Uploading a basic Arduino sketch with u8g2 to control the ST7920 via SPI works fine.

### Which version of ESPHome has the issue?

v2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP32

### Board

NodeMCU-32S

### Component causing the issue

spi

### Example YAML snippet

```yaml
esphome:
  name: esp32b
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "foo"

wifi:
  ssid: "foo"
  password: "bar"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32B Fallback Hotspot"
    password: "zoo"

captive_portal:

light:
  - platform: status_led
    name: "ESP32B Status LED"
    pin: 
      number: GPIO2
      inverted: true

spi:
  clk_pin: GPIO19
  mosi_pin: GPIO5

font:
  - file: "dotmatrix.ttf"
    id: dotmatrix
    size: 8
   
sensor:
  - platform: wifi_signal
    id: signal
    name: "ESP32B WiFi Signal"
    update_interval: 5s
    on_value:
      then:
        component.update: lcdd

time:
  - platform: homeassistant
    id: ha_time
    on_time:
      - seconds: 0
        minutes: "*"
        then:
          component.update: lcdd
        
display:
  - platform: st7920
    id: lcdd
    cs_pin: GPIO17
    width: 128
    height: 64
    update_interval: 60s
    lambda: |-
      it.line(0, 10, 127, 10);
      if (id(ha_time).now().is_valid()) { 
        it.strftime(63,  0, id(dotmatrix), TextAlign::TOP_CENTER, "%H:%M", id(ha_time).now());
      };
      it.printf(0, 10, id(dotmatrix), "1: %s", id(lcd1).state.c_str());
      it.printf(0, 20, id(dotmatrix), "2: %s", id(lcd2).state.c_str());
      it.printf(0, 30, id(dotmatrix), "3: %s", id(lcd3).state.c_str());
      it.printf(0, 40, id(dotmatrix), "4: %s", id(lcd4).state.c_str());
      if(id(signal).state >= 0) {
          it.print(0, 50, id(dotmatrix), "No Signal/Error");
          it.filled_rectangle(117, 8, 2, 1);
          it.filled_rectangle(120, 8, 2, 1);
          it.filled_rectangle(123, 8, 2, 1);
          it.filled_rectangle(126, 8, 2, 1);
          it.line(120, 1, 124, 5);
          it.line(124, 1, 120, 5);
        } else if(isnan(id(signal).state)) {
          it.print(0, 50, id(dotmatrix), "Initializing");
          it.filled_rectangle(117, 8, 2, 1);
          it.filled_rectangle(120, 8, 2, 1);
          it.filled_rectangle(123, 8, 2, 1);
          it.filled_rectangle(126, 8, 2, 1);
          it.line(120, 1, 124, 5);
          it.line(124, 1, 120, 5);
          } else if(id(signal).state >= -50) {
          it.print(0, 50, id(dotmatrix), "Very Good Signal");
          it.filled_rectangle(117, 7, 2, 2);
          it.filled_rectangle(120, 5, 2, 4);
          it.filled_rectangle(123, 3, 2, 6);
          it.filled_rectangle(126, 1, 2, 8);
        } else if (id(signal).state >= -65) {
          it.print(0, 50, id(dotmatrix), "Good Signal");
          it.filled_rectangle(117, 7, 2, 2);
          it.filled_rectangle(120, 5, 2, 4);
          it.filled_rectangle(123, 3, 2, 6);
          it.filled_rectangle(126, 8, 2, 1);
        } else if (id(signal).state >= -85) {
          it.print(0, 50, id(dotmatrix), "Fair Signal");
          it.filled_rectangle(117, 7, 2, 2);
          it.filled_rectangle(120, 5, 2, 4);
          it.filled_rectangle(123, 8, 2, 1);
          it.filled_rectangle(126, 8, 2, 1);
        } else if (id(signal).state >= -105) {
          it.print(0, 50, id(dotmatrix), "Low Signal");
          it.filled_rectangle(117, 7, 2, 2);
          it.filled_rectangle(120, 8, 2, 1);
          it.filled_rectangle(123, 8, 2, 1);
          it.filled_rectangle(126, 8, 2, 1);
        } else {
          it.print(0, 50, id(dotmatrix), "Very Low Signal");
          it.filled_rectangle(117, 8, 2, 1);
          it.filled_rectangle(120, 8, 2, 1);
          it.filled_rectangle(123, 8, 2, 1);
          it.filled_rectangle(126, 8, 2, 1);
        };
        
        

mqtt:
  broker: 1.2.3.4
  username: foo
  password: bar
  client_id: esp32b
  birth_message: 
    topic: esp32b/status
    payload: online
  will_message: 
    topic: esp32b/status
    payload: offline
 
text_sensor:
  - platform: mqtt_subscribe
    name: "LCD 1"
    id: lcd1
    topic: esp32b/lcd1
    on_value:
      then:
        component.update: lcdd
  - platform: mqtt_subscribe
    name: "LCD 2"
    id: lcd2
    topic: esp32b/lcd2
    on_value:
      then:
        component.update: lcdd
  - platform: mqtt_subscribe
    name: "LCD 3"
    id: lcd3
    topic: esp32b/lcd3  
    on_value:
      then:
        component.update: lcdd
  - platform: mqtt_subscribe
    name: "LCD 4"
    id: lcd4
    topic: esp32b/lcd4
    on_value:
      then:
        component.update: lcdd
```

### Anything in the logs that might be useful for us?

```txt
Only issue I noticed whilst compiling my program was this:

Compiling /data/esp32b/.pioenvs/esp32b/FrameworkArduino/esp32-hal-time.c.o
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

marsjan155, 2021-09-23:
@jsb5151 
can you try with this config?:
```
....
display:
  - platform: st7920
    id: lcdd
    cs_pin:
      number: GPIO17
      inverted: true
...
```

jsb5151, 2021-09-23:
Well... that was it.  I forgot to invert the CS pin.  And it's documented too... *facepalm*

I spent the whole evening trying to troubleshoot this; I had the inverted: true setting at some point in there as I copied my configuration from a working ESP8266.   Must've accidentally removed it when I set it up for a SH1106.

Thanks!

closed 2021-09-23 by jsb5151

marsjan155, 2021-09-23:
I'm glad I could help, it's good to know that someone else is using this display too, and it works! :)