esphome/issues #2463: Can't resolve mdns broker on ESP32

fgervais, 2021-09-22:
### The problem

Using

```yaml
mqtt:
  broker: home.local
```

results in

> [19:31:47][D][mqtt:113]: Resolving MQTT broker IP address...
[19:31:47][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!

while the same works on my esp8266 S31.

### Which version of ESPHome has the issue?

esphome/esphome:2021.9.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

ESP32-WROVER-E

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
esphome:
  name: julie-light
  platform: ESP32
  board: esp-wrover-kit

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

mqtt:
  broker: home.local
```

### Anything in the logs that might be useful for us?

```txt
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (TTL232R-3V3 - TTL232R-3V3)
  [2] Over The Air (julie-light.local)
(number): 2
INFO Resolving IP address of julie-light.local
INFO  -> 192.168.2.144
INFO Uploading julie-light/.pioenvs/julie-light/firmware.bin (831616 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (TTL232R-3V3 - TTL232R-3V3)
  [2] MQTT (home.local)
(number): 1
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:31:47][I][wifi:515]: WiFi Connected!
[19:31:47][C][wifi:361]:   SSID: 
[19:31:47][C][wifi:362]:   IP Address: 192.168.2.144
[19:31:47][C][wifi:364]:   BSSID: 
[19:31:47][C][wifi:365]:   Hostname: 'julie-light'
[19:31:47][C][wifi:369]:   Signal strength: -66 dB ▂▄▆█
[19:31:47][C][wifi:373]:   Channel: 11
[19:31:47][C][wifi:374]:   Subnet: 255.255.255.0
[19:31:47][C][wifi:375]:   Gateway: 192.168.2.1
[19:31:47][C][wifi:376]:   DNS1: 192.168.2.1
[19:31:47][C][wifi:377]:   DNS2: 0.0.0.0
[19:31:47][C][ota:029]: Over-The-Air Updates:
[19:31:47][C][ota:030]:   Address: julie-light.local:3232
[19:31:47][C][mqtt:027]: Setting up MQTT...
[19:31:47][D][mqtt:113]: Resolving MQTT broker IP address...
[19:31:47][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:31:52][D][mqtt:113]: Resolving MQTT broker IP address...
[19:31:52][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:31:57][D][mqtt:113]: Resolving MQTT broker IP address...
[19:31:57][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:02][D][mqtt:113]: Resolving MQTT broker IP address...
[19:32:02][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:07][D][mqtt:113]: Resolving MQTT broker IP address...
[19:32:07][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:12][D][mqtt:113]: Resolving MQTT broker IP address...
[19:32:12][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:17][D][mqtt:113]: Resolving MQTT broker IP address...
[19:32:17][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:22][D][mqtt:113]: Resolving MQTT broker IP address...
[19:32:22][W][mqtt:137]: Couldn't resolve IP address for 'home.local'!
[19:32:27][D][mqtt:113]: Resolving MQTT broker IP address...
```

fgervais, 2021-09-23:
This post seems to be related, I'm double linking to keep track.

https://community.home-assistant.io/t/how-to-use-domain-names-local-with-esphome-and-esp32/307369/7

bcotton, 2022-01-30:
Seeing the same problem here. mDNS works fine on my 8266 boards (wemos, sonoff) but the ESP32 boards will neither respond to mDNS queries nor resolve other addresses, e.g. HA and MQTT.

arthurkomatsu, 2022-09-30:
I have run into this issue today as well.

DavidWicks, 2022-12-28:
Same problem here

`INFO Successfully connected to 10.1.20.100
[17:56:39][W][mqtt:141]: Couldn't resolve IP address for '10.1.1.18...'!
[17:56:39][D][mqtt:117]: Resolving MQTT broker IP address...
[17:56:52][W][mqtt:141]: Couldn't resolve IP address for '10.1.1.18...'!
[17:56:53][D][mqtt:117]: Resolving MQTT broker IP address...
[17:57:07][W][mqtt:141]: Couldn't resolve IP address for '10.1.1.18...'!
[17:57:07][D][mqtt:117]: Resolving MQTT broker IP address...
[17:57:21][W][mqtt:141]: Couldn't resolve IP address for '10.1.1.18...'!`

Bobsonkz, 2023-09-16:
The same problem: 
[W][mqtt:210]: Couldn't resolve IP address for 'mydomain.ddns.net'!
[W][mqtt:195]: Error resolving MQTT broker IP address: -6

ssieb, 2023-09-16:
@Bobsonkz that's a different issue.  That's not an mdns address.

Bobsonkz, 2023-09-16:
> @Bobsonkz that's a different issue. That's not an mdns address.

So any hint to my problem?
For me it is similar this person want to use domain name insted of IP address.

I have a working system, esp32 boards on remote location with use NOIP for connection. All work on tasamta devices but not in esphome.

ssieb, 2023-09-16:
You problem is off-topic for this issue.  You aren't using mdns.  Come ask on the esphome discord server.