esphome/issues #2464: Periodical esp reboot when using 'platform: integration/total_daily_energy'

Spirituss, 2021-09-23:
### The problem

In case I add the ```platform: integration``` and ```platform: total_daily_energy ``` to esp8266, it starts to reboot every 5-15 hours (statistically). 
If I remove these platforms but keep other parts of the configuration, esp works stable for some months. It seems that integration leads to an unhandled exception which cause esp reboot.

 

### Which version of ESPHome has the issue?

All

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

All

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
Here is my esphome config:

esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp8285
  esp8266_restore_from_flash: true  

....

sensor:
  - platform: integration 
    name: energy_monthly_${device_name}
    id: id_energy_monthly
    sensor: id_power
    time_unit: h
    filters:
      - multiply: 0.001
    accuracy_decimals: 3
    restore: true
    state_class: measurement
    device_class: energy
    last_reset_type: auto
#-------------------------------------------------------------------------------    
  - platform: total_daily_energy 
    name: energy_daily_${device_name}
    id: id_energy_daily
    power_id: id_power
    filters:
      - multiply: 0.001
    accuracy_decimals: 3   
    device_class: energy
#-------------------------------------------------------------------------------
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO05
    cf1_pin: GPIO4
    power:
      name: power_${device_name}
      id: id_power            
    change_mode_every: 4
    update_interval: 10s
```
```

oxan, 2021-09-23:
See also #2366.

oxan, 2021-09-23:
Are you also using MQTT?

Spirituss, 2021-09-24:
> Are you also using MQTT?

Do you mean epshome? No. But I use it in HA.

See also https://github.com/esphome/issues/issues/2460

oxan, 2021-09-26:
Hmm, that's strange, as in #2366 it was resolved by disabling MQTT.

tschundler, 2021-10-04:
I updated my ticket a few days ago, but forgot to update here too. It wasn't MQTT. It lasted over a day, but still rebooted and then again much less than a day. So it was mostly by luck I think.

Spirituss, do you also find it seems to be related to how long current is being drawn?

Spirituss, 2021-10-04:
> Spirituss, do you also find it seems to be related to how long current is being drawn?

Nothing has changed - I still see reboot for some esp every 12-24 hours. I tried different firmware versions, the same result. In case I remove 'integration' component, reboots are gone. So, I'm pretty sure that especially this component is the reason of this headache.

Spirituss, 2022-01-10:
Any news here? The bug is annoying

oxan, 2022-01-10:
I don't think any progress on this bug will be made until someone manages to capture a stack trace during the reboot.