esphome/issues #2467: can't restore esphome add-on via Supervisor Backups

paulius2k, 2021-09-24:
### The problem

When updating the esphome add-on to a new version 2021.9.1, I made a backup of it as suggested by the Supervisor.

I am attempting to restore that backup (2021.8.2), but process fails, see logs.

PS: I can successfully restore other addon (e.g. Terminal & SSH "core-ssh").

PPS: I was redirected here from the Supervisor github (https://github.com/home-assistant/supervisor/issues/3143)

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
21-09-24 14:35:20 INFO (MainThread) [supervisor.addons.addon] Restore config for addon a0d7b954_esphome
21-09-24 14:35:20 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon a0d7b954_esphome
21-09-24 14:35:20 ERROR (MainThread) [supervisor.jobs] Unhandled exception: [Errno 2] No such file or directory: '/data/tmp/tmpxzszliap/data'
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 105, in wrapper
    return await self._method(*args, **kwargs)
  File "/usr/src/supervisor/supervisor/addons/__init__.py", line 358, in restore
    await addon.restore(tar_file)
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 853, in restore
    await self.sys_run_in_executor(_restore_data)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 847, in _restore_data
    shutil.copytree(Path(temp, "data"), self.path_data, symlinks=True)
  File "/usr/local/lib/python3.9/shutil.py", line 555, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [Errno 2] No such file or directory: '/data/tmp/tmpxzszliap/data'
21-09-24 14:35:20 WARNING (MainThread) [supervisor.backups.backup] Can't restore Add-on a0d7b954_esphome: 
21-09-24 14:35:20 INFO (MainThread) [supervisor.backups.manager] Partial-Restore 59101d40 done
```
```

### Additional information

In the Supervisor Dashboard esphome does show with an older version:
```
ESPHome 2021.9.1 is available
You are currently running version 2021.8.2
```

but when trying to start the service, it throws error: 
```
Failed to start add-on
Unknown error, see supervisor
```

aal7, 2021-09-25:
I am having exactly the same problem (ie ESPHome doesn't work after a backup restore). Here are my logs:

> 21-09-25 11:03:52 INFO (MainThread) [supervisor.addons.addon] Restore config for addon a0d7b954_esphome
21-09-25 11:03:52 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon a0d7b954_esphome
21-09-25 11:03:52 ERROR (MainThread) [supervisor.jobs] Unhandled exception: [Errno 2] No such file or directory: '/data/tmp/tmpq21f_vsp/data'
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 105, in wrapper
    return await self._method(*args, **kwargs)
  File "/usr/src/supervisor/supervisor/addons/__init__.py", line 358, in restore
    await addon.restore(tar_file)
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 853, in restore
    await self.sys_run_in_executor(_restore_data)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 847, in _restore_data
    shutil.copytree(Path(temp, "data"), self.path_data, symlinks=True)
  File "/usr/local/lib/python3.9/shutil.py", line 555, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [Errno 2] No such file or directory: '/data/tmp/tmpq21f_vsp/data'
21-09-25 11:03:52 WARNING (MainThread) [supervisor.backups.backup] Can't restore Add-on a0d7b954_esphome:

aal7, 2021-09-29:
Please excuse my naivety on these things. I note that this has been marked as an External Issue. I assume this means the problem lies elsewhere. Do I need to report it to someone else and if so, who? Many thanks.

deftdawg, 2021-10-05:
duplicate of #2430

paulius2k, 2024-01-21:
I tried it now - and it works.  Fixed by time :)

closed 2024-01-21 by paulius2k