esphome/issues #2468: Unexpected error while reading incoming messages: [Errno 104] Connection reset by peer - repeating periodically

paulius2k, 2021-09-24:
### The problem

One of my four esp8266 boards started showing ```[Errno 104] Connection reset by peer``` errors.
I am not sure when exactly this started - _probably_ with ```esphome v2021.8.2```, because my older boards with ```esphome v1.19.4``` don't have that issue.

The connection is reset periodically in exact 10 OR 15 minute intervals - see logs.

I have tried setting fixed IP for that board in the esphome yaml, then reverted as there was no difference.

I changed the power supply as well - no difference.

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Example YAML snippet

```yaml
esphome:
  name: plant-watering
  platform: ESP8266
  board: d1_mini_pro

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:

ota:
  password: "xx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Optional manual IP
  # manual_ip:
  #   static_ip: 192.168.0.141
  #   gateway: 192.168.0.1
  #   subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Plant-Watering Fallback Hotspot"
    password: "LgaWnDRYAl1k"

captive_portal:

switch:
  - platform: gpio
    name: "Plant Watering Relay-Motor"
    pin: D3
[... 201 lines cut ...]
	Line 55011: 2021.09.23 01:54:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55016: 2021.09.23 02:09:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55021: 2021.09.23 02:24:01 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55025: 2021.09.23 02:34:01 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55029: 2021.09.23 02:44:01 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55033: 2021.09.23 02:59:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55038: 2021.09.23 03:09:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55043: 2021.09.23 03:24:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55047: 2021.09.23 03:34:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55054: 2021.09.23 03:59:01 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55057: 2021.09.23 04:09:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55066: 2021.09.23 04:29:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55072: 2021.09.23 04:39:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55083: 2021.09.23 04:59:01 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55087: 2021.09.23 05:14:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55091: 2021.09.23 05:24:02 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55095: 2021.09.23 05:39:07 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55100: 2021.09.23 05:54:02 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55104: 2021.09.23 06:09:02 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55110: 2021.09.23 06:24:07 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55116: 2021.09.23 06:34:06 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55120: 2021.09.23 06:44:02 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55125: 2021.09.23 06:59:07 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
	Line 55129: 2021.09.23 07:14:07 WARNING [homeassistant.components.esphome] Can't connect to ESPHome API for plant-watering (192.168.0.141): Error connecting to ('192.168.0.141', 6053): [Errno 111] Connect call failed ('192.168.0.141', 6053)
```

paulius2k, 2021-09-26:
Update: I have replaced my new router (Netgear AX3000 RAX40v2), on which the problem appeared, with the old router (Netgear R6400v2) and the issue disappeared, just like that... I have no idea what exactly on the router caused this behaviour. If the developers know, this could be put here if somebody bumps into a similar issue.

krzysiek2788, 2021-09-29:
I have the same issue with official build 2021.09.2. I have esp8266+mcp23017+uart and has the same errors I remember some version before all was working fine. Maybe this new security update mess something in wifi configuration?? I have uniFi system at home and my network didn't change.

krzysiek2788, 2021-09-29:
Also my system overview: 
[image]

paulius2k, 2021-09-30:
Even on my other router I still have situations when ESPhome cannot connect to the API and **rebooting** the router resolves this issue.

closed 2022-02-12 by github-actions[bot]