esphome/issues #2469: Sonoff GPIO pin 0 does not support PULLUP_PIN mode

DayRunner131, 2021-09-24:
### The problem

Running ESPHome-dev V2021.10.0-dev on HA Supervisor-2021.09.0, Core-2021.9.7. Went to recompile and update nodes with new ESPHome packages and encountered the following issue during "validate."

Failed config

binary_sensor.gpio: [source /config/esphome/esp-th16-1.yaml:50]
  platform: gpio
  pin:  [source /config/esphome/esp-th16-1.yaml:52]
    number: GPIO0
    
    GPIO Pin 0 does not support pullup pin mode. Please choose another pin.
    mode: INPUT_PULLUP [source /config/esphome/esp-th16-1.yaml:53]
    inverted: True
  name: TH16 #1 Basic Button
  on_press: 
    - switch.toggle: relay

Current documents found here https://esphome.io/devices/sonoff_basic.html still show PULLUP_MODE still used for basic Sonoff power buttons.

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

Sonoff S31 and Sonoff TH16

### Component causing the issue

PULLUP_MODE

### Example YAML snippet

```yaml
substitutions:
# Change the disp_name to something you want  
  disp_name: ESP TH16 1
# Interval of how often the power is updated
  update_time: 10s
  
esphome:
  name: esp-th16-1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "XX"
  password: "XX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Th16-1 Fallback Hotspot"
    password: ""
    
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.50.15
    gateway: 192.168.50.1
    subnet: 255.255.255.0

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

#mqtt:
#  broker: 192.168.1.3
#  username: th161
#  password: th@161
#  discovery: true

ota:

web_server:
  port: 80
  auth:
    username: XX
    password: ""
  
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "TH16 #1 Basic Button"
    on_press:
      - switch.toggle: relay
      
sensor:
  - platform: dht
    pin: GPIO14
    temperature:
      unit_of_measurement: "C"
      name: "TH16 #1 Temperature °C"
      id: "temp"
    humidity:
      unit_of_measurement: "% HUM"
      name: "TH16 #1 Humidity"
    model: SI7021
    update_interval: 30s
    
  - platform: template
    name: "TH16 #1 Temperature"
    unit_of_measurement: "°F"
    lambda: return (id(temp).state * (9.0/5.0) + 32.0);

switch:
  - platform: gpio
    name: "TH16 #1 Basic Relay"
    pin: GPIO12
    id: relay
    
  - platform: restart
    name: ${disp_name} Restart

status_led:
  pin:
    number: GPIO13
    inverted: yes

time:
  - platform: homeassistant
    id: homeassistant_time
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp-th16-1.yaml...
INFO Detected timezone 'EST' with UTC offset -5
Failed config

binary_sensor.gpio: [source /config/esphome/esp-th16-1.yaml:50]
  platform: gpio
  pin:  [source /config/esphome/esp-th16-1.yaml:52]
    number: GPIO0
    
    GPIO Pin 0 does not support pullup pin mode. Please choose another pin.
    mode: INPUT_PULLUP [source /config/esphome/esp-th16-1.yaml:53]
    inverted: True
  name: TH16 #1 Basic Button
  on_press: 
    - switch.toggle: relay
```

zvldz, 2021-09-28:
same with pow r2

benquan, 2021-10-21:
I commented out       

    #  mode: INPUT_PULLUP

Now it compiles and it still works as usual, but not sure if this would be a recommended approach.