esphome/issues #2470: Cannot Turn off the LED

adonno, 2021-09-24:
### The problem

I have code running that should flash the LED fastLED. since ESPHome 2021.09 it has been reported that the LED stays on all the time

https://github.com/adonno/tagreader/blob/master/tagreader.yaml

### Which version of ESPHome has the issue?

2021.09.01

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

FastLED

ssieb, 2021-09-24:
Should be fixed by https://github.com/esphome/esphome/pull/2383

7h30n3, 2021-09-26:
I have a similar issue, maybe with the same cause.

```
    on_multi_click:
    - timing:
        - ON for 40ms to 400ms
        - OFF for at least 50ms
      then:
        - logger.log: "Single Short Clicked"
        - if:
            condition:
              - light.is_off: ${device_id}_light
            then:
              - light.turn_on:
                  id: ${device_id}_light
                  red: 1
                  green: 0
                  blue: 1
                  brightness: 1
            else:
              - light.turn_off:
                  id: ${device_id}_light
    - timing:
        - ON for at least 450ms
      then:
        - logger.log: "Single Long Click"
        - light.turn_on:
            id: "${device_id}_light"
            red: 1
            green: 1
            blue: 1
            brightness: 1
```

I can't get it to work. The problem lies within the "Single Short Click". Either the color changes as intended or the light turns off. It used to work until 2021.09.

oxan, 2021-09-26:
@7h30n3 It's not completely clear to me what your problem is, but since this one is about flashes is it's probably unrelated. Please open a new issue.