esphome/issues #2471: AirThings values are not coming through

Valcob, 2021-09-24:
### The problem

I have an AirThings Plus device which works fine through their mobile app BUT does  not work through BLE integration
though the device is discovered no issues, just values are not coming through the integration

### Which version of ESPHome has the issue?

 ESPHome version 2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.9.7

### What platform are you using?

ESP32

### Board

D1 esp32

### Component causing the issue

airthings_wave_plus

### Example YAML snippet

```yaml
sensor:
  - platform: airthings_wave_plus
    ble_client_id: den_airthings
    update_interval: 5min # default
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"

ble_client:
  - mac_address: "D8:71:4D:12:65:37"
    id: den_airthings

esp32_ble_tracker:
# airthings_ble:
```

### Anything in the logs that might be useful for us?

```txt
[21:16:58][C][ble_client:033]: BLE Client:
[21:16:58][C][ble_client:034]:   Address: d8:71:4d:12:65:37
[21:16:58][C][esp32_ble_tracker:614]: BLE Tracker:
[21:16:58][C][esp32_ble_tracker:615]:   Scan Duration: 300 s
[21:16:58][C][esp32_ble_tracker:616]:   Scan Interval: 320.0 ms
[21:16:58][C][esp32_ble_tracker:617]:   Scan Window: 30.0 ms
[21:16:58][C][esp32_ble_tracker:618]:   Scan Type: ACTIVE
[21:16:58][C][captive_portal:148]: Captive Portal:
[21:16:58][C][web_server:152]: Web Server:
[21:16:58][C][web_server:153]:   Address: den-bluetooth-bridge.local:80
[21:16:58][C][ota:029]: Over-The-Air Updates:
[21:16:58][C][ota:030]:   Address: den-bluetooth-bridge.local:3232
[21:16:58][C][ota:032]:   Using Password.
[21:16:58][C][api:135]: API Server:
[21:16:58][C][api:136]:   Address: den-bluetooth-bridge.local:6053
[21:16:58][C][wifi_signal.sensor:009]: WiFi Signal 'den-bluetooth-bridge - Wifi Signal'
[21:16:58][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:16:58][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:16:58][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:16:58][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:16:58][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[21:17:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -28.00000 dBm with 0 decimals of accuracy
[21:17:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 90.51700 s with 0 decimals of accuracy
[21:18:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -28.00000 dBm with 0 decimals of accuracy
[21:18:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 150.51199 s with 0 decimals of accuracy
[21:18:58][W][airthings_wave_plus:106]: Connection in progress
[21:19:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -29.00000 dBm with 0 decimals of accuracy
[21:19:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 210.51199 s with 0 decimals of accuracy
[21:20:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -29.00000 dBm with 0 decimals of accuracy
[21:20:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 270.51199 s with 0 decimals of accuracy
[21:21:04][I][ota:046]: Boot seems successful, resetting boot loop counter.
[21:21:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -29.00000 dBm with 0 decimals of accuracy
[21:21:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 330.51199 s with 0 decimals of accuracy
[21:22:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -28.00000 dBm with 0 decimals of accuracy
[21:22:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 390.51599 s with 0 decimals of accuracy
[21:23:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -29.00000 dBm with 0 decimals of accuracy
[21:23:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 450.51599 s with 0 decimals of accuracy
[21:23:58][W][airthings_wave_plus:106]: Connection in progress
[21:24:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -30.00000 dBm with 0 decimals of accuracy
[21:24:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 510.51700 s with 0 decimals of accuracy
[21:25:05][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -29.00000 dBm with 0 decimals of accuracy
[21:25:34][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 570.51202 s with 0 decimals of accuracy
[21:26:05][W][esp32_ble_tracker:196]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
[21:26:05][I][app:127]: Forcing a reboot...
INFO Disconnected from ESPHome API for den-bluetooth-bridge.local
WARNING Disconnected from API
INFO den-bluetooth-bridge.local: Unexpected error while reading incoming messages: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 505, in run_forever
    await self._run_once()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 479, in _run_once
    assert self._frame_helper is not None
AssertionError
INFO Successfully connected to den-bluetooth-bridge.local
[21:26:52][D][sensor:121]: 'den-bluetooth-bridge - Uptime': Sending state 46.74900 s with 0 decimals of accuracy
[21:27:02][D][sensor:121]: 'den-bluetooth-bridge - Wifi Signal': Sending state -30.00000 dBm with 0 decimals of accuracy
```

jeromelaban, 2021-09-26:
Thanks for the report. Does the `esp32_ble_tracker` work by itself (by removing the airthings sensor)?

Valcob, 2021-09-26:
yeah it work perfectly fine  as a hub for xiaomi stuff and it does not work in  combined nor just airthings solo scenario.

mandreko, 2021-09-29:
I am also having this issue. 

My initial config for discovery:
```
esphome:
  name: airthings-reader
  platform: ESP32
  board: wemos_d1_mini32

# Enable logging
logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
airthings_ble:
```

Output from discovery:
```
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.4.215 using esphome API
INFO Successfully connected to 192.168.4.215
[16:13:02][I][app:102]: ESPHome version 2021.9.1 compiled on Sep 29 2021, 16:11:13
[16:13:02][C][wifi:501]: WiFi:
[16:13:02][C][wifi:361]:   SSID: [redacted]
[16:13:02][C][wifi:362]:   IP Address: 192.168.4.215
[16:13:02][C][wifi:364]:   BSSID: [redacted]
[16:13:02][C][wifi:365]:   Hostname: 'airthings-reader'
[16:13:02][C][wifi:369]:   Signal strength: -63 dB ▂▄▆█
[16:13:02][C][wifi:373]:   Channel: 6
[16:13:02][C][wifi:374]:   Subnet: 255.255.255.0
[16:13:02][C][wifi:375]:   Gateway: 192.168.4.1
[16:13:02][C][wifi:376]:   DNS1: 192.168.4.1
[16:13:02][C][wifi:377]:   DNS2: 0.0.0.0
[16:13:02][C][logger:193]: Logger:
[16:13:02][C][logger:194]:   Level: DEBUG
[16:13:02][C][logger:195]:   Log Baud Rate: 115200
[16:13:02][C][logger:196]:   Hardware UART: UART0
[16:13:02][C][esp32_ble_tracker:614]: BLE Tracker:
[16:13:02][C][esp32_ble_tracker:615]:   Scan Duration: 300 s
[16:13:02][C][esp32_ble_tracker:616]:   Scan Interval: 320.0 ms
[16:13:02][C][esp32_ble_tracker:617]:   Scan Window: 30.0 ms
[16:13:02][C][esp32_ble_tracker:618]:   Scan Type: ACTIVE
[16:13:02][C][captive_portal:148]: Captive Portal:
[16:13:02][C][ota:029]: Over-The-Air Updates:
[16:13:02][C][ota:030]:   Address: 192.168.4.215:3232
[16:13:02][C][ota:032]:   Using Password.
[16:13:02][C][api:135]: API Server:
[16:13:02][C][api:136]:   Address: 192.168.4.215:6053
[16:13:05][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
[16:13:06][D][esp32_ble_tracker:628]: Found device 5F:61:DE:8C:A5:EA RSSI=-77
[16:13:06][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:13:12][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
[16:13:14][D][esp32_ble_tracker:628]: Found device E3:EF:E8:98:74:7B RSSI=-45
[16:13:14][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:13:15][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
[16:13:19][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
[16:13:22][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
[16:13:26][D][airthings_ble:022]: Found AirThings device Serial:2930083388 (MAC: D8:71:4D:A2:8F:8A)
```

Based on that information, I assume the MAC Address of the Airthings is D8:71:4D:A2:8F:8A, and modify the config accordingly:
```
esphome:
  name: airthings-reader
[... 102 lines cut ...]
[16:19:50][C][ota:030]:   Address: 192.168.4.215:3232
[16:19:50][C][ota:032]:   Using Password.
[16:19:50][C][api:135]: API Server:
[16:19:50][C][api:136]:   Address: 192.168.4.215:6053
[16:19:52][D][ble_client:045]: Found device at MAC address [D8:71:4D:A2:8F:8A]
[16:23:14][W][airthings_wave_plus:106]: Connection in progress
[16:24:42][I][ota:046]: Boot seems successful, resetting boot loop counter.
[16:28:14][W][airthings_wave_plus:106]: Connection in progress
[16:29:43][W][esp32_ble_tracker:196]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
[16:29:43][I][app:127]: Forcing a reboot...
INFO Disconnected from ESPHome API for 192.168.4.215
WARNING Disconnected from API
INFO 192.168.4.215: Unexpected error while reading incoming messages: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 505, in run_forever
    await self._run_once()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 479, in _run_once
    assert self._frame_helper is not None
AssertionError
INFO Successfully connected to 192.168.4.215
[16:29:50][D][api:095]: Accepted 192.168.1.52
[16:29:50][D][api.connection:727]: Home Assistant 2021.9.7 (192.168.1.52): Connected successfully
```

To troubleshoot, I found another ESP32 in my box of MCUs, and tried it to ensure it wasn't something wrong with the first ESP32 that I tried. I get the same results though. :(

jeepin95, 2021-10-01:
I'm seeing this same issue and have tried on two different HelTec ESP32 devices. My ESP32 has direct line of sight to the Wave Plus. I was able to run the version to find the MAC just fine. The serial number detected matched what I see in the Airthings app and and I used the MAC detected. The ESP32 connects to my MQTT broker and the sensors show up in HA, but they never they are always unknown and never receive data.

```[21:23:48][D][airthings_ble:022]: Found AirThings device Serial:29300XXXXX (MAC: 00:81:F9:F8:18:0A)```

Wave Plus Firmware: BLE: 1.5.3, MSP: 1.6.0, SUB:2.0.2

My YAML is below. When things weren't working I switched update_interval from the default 5min to 1min so I could see the connection attempts sooner. It appears to try to connect, but never does. There is an error in the logs for ble that I am only able to see from the serial logs, I'm not sure if that is normal in the startup sequence or not (Lines 18 & 19 of the log below).

```
sensor:
  - platform: airthings_wave_plus
    ble_client_id: airthings01
    update_interval: 1min
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"

ble_client:
  - mac_address: '00:81:F9:F8:18:0A'
    id: airthings01

esp32_ble_tracker:
```

Log
```
[D][ota:247]: OTA in progress: 98.2%
[I][ota:276]: OTA update finished!
[I][app:137]: Rebooting safely...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:174]: Log initialized
[C][ota:379]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[E][ble_client:017]: gattc app register failed. app_id=1 code=259
[E][component:102]: Component ble_client was marked as failed.
[D][esp32_ble_tracker:180]: Starting scan...
[C][wifi:037]: Setting up WiFi...
[D][wifi:382]: Starting scan...
[W][airthings_wave_plus:106]: Connection in progress
[D][esp32_ble_tracker:628]: Found device 37:8A:26:40:D4:52 RSSI=-50
[D][esp32_ble_tracker:649]:   Address Type: RANDOM
[D][esp32_ble_tracker:628]: Found device 58:2F:F7:18:AA:97 RSSI=-96
[D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[D][esp32_ble_tracker:628]: Found device 5A:DA:DA:0A:5A:00 RSSI=-66
[D][esp32_ble_tracker:649]:   Address Type: RANDOM
[D][esp32_ble_tracker:628]: Found device 24:0A:C4:F7:E2:02 RSSI=-69
[D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[D][esp32_ble_tracker:651]:   Name: 'Meshtastic_e200'
[D][esp32_ble_tracker:653]:   TX Power: 2
[... 64 lines cut ...]
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_radon/state'
[C][mqtt.sensor:024]: MQTT Sensor 'WavePlus Radon Long Term':
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_radon_long_term/state'
[C][mqtt.sensor:024]: MQTT Sensor 'WavePlus Temperature':
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_temperature/state'
[C][mqtt.sensor:024]: MQTT Sensor 'WavePlus Pressure':
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_pressure/state'
[C][mqtt.sensor:024]: MQTT Sensor 'WavePlus CO2':
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_co2/state'
[C][mqtt.sensor:024]: MQTT Sensor 'WavePlus VOC':
[C][mqtt.sensor:028]:   State Topic: 'airthings_esp32/sensor/waveplus_voc/state'
[D][esp32_ble_tracker:628]: Found device 7F:FF:FD:F5:4F:BE RSSI=-94
[D][esp32_ble_tracker:649]:   Address Type: RANDOM
[D][esp32_ble_tracker:651]:   Name: '[LG] webOS TV OLED65CXAUA'
[D][esp32_ble_tracker:653]:   TX Power: 2
[D][esp32_ble_tracker:653]:   TX Power: 2
[D][esp32_ble_tracker:628]: Found device 05:E6:11:99:D2:BE RSSI=-54
[D][esp32_ble_tracker:649]:   Address Type: RANDOM
[D][esp32_ble_tracker:628]: Found device B4:52:A9:C4:6B:2F RSSI=-86
[D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[D][ble_client:045]: Found device at MAC address [00:81:F9:F8:18:0A]
[W][airthings_wave_plus:106]: Connection in progress
[W][airthings_wave_plus:106]: Connection in progress

```

jeromelaban, 2021-10-01:
This integration was tested only on an ESP-WROOM-32, which may be why there are some differences.

I wonder though if it's an underlying issue with the the `ble_client` integration.

Could you try with the following: 

```yaml
ble_client:
  - mac_address: xx:xx:xx:xx:xx:xx
    id: mydevice

sensor:
  - platform: ble_client
    ble_client_id: mydevice
    name: "battery level"
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    icon: 'mdi:battery'
    unit_of_measurement: '%'

esp32_ble_tracker:
```

And see if the results are different ?

mandreko, 2021-10-01:
@jeromelaban I just tried that configuration using only raw BLE. It looks like it's not working either. Log excerpt:

```
INFO Successfully uploaded program.
INFO Starting log output from 192.168.4.215 using esphome API
INFO Successfully connected to 192.168.4.215
[09:55:45][I][app:102]: ESPHome version 2021.9.1 compiled on Oct  1 2021, 09:53:51
[09:55:45][C][wifi:501]: WiFi:
[09:55:45][C][wifi:361]:   SSID: [redacted]
[09:55:45][C][wifi:362]:   IP Address: 192.168.4.215
[09:55:45][C][wifi:364]:   BSSID: [redacted]
[09:55:45][C][wifi:365]:   Hostname: 'airthings-reader'
[09:55:45][C][wifi:369]:   Signal strength: -57 dB ▂▄▆█
[09:55:45][C][wifi:373]:   Channel: 6
[09:55:45][C][wifi:374]:   Subnet: 255.255.255.0
[09:55:45][C][wifi:375]:   Gateway: 192.168.4.1
[09:55:45][C][wifi:376]:   DNS1: 192.168.4.1
[09:55:45][C][wifi:377]:   DNS2: 0.0.0.0
[09:55:45][C][logger:193]: Logger:
[09:55:45][C][logger:194]:   Level: DEBUG
[09:55:45][C][logger:195]:   Log Baud Rate: 115200
[09:55:45][C][logger:196]:   Hardware UART: UART0
[09:55:45][C][ble_sensor:019]: BLE Sensor 'Battery Level'
[09:55:45][C][ble_sensor:019]:   State Class: ''
[09:55:45][C][ble_sensor:019]:   Unit of Measurement: '%'
[09:55:45][C][ble_sensor:019]:   Accuracy Decimals: 0
[09:55:45][C][ble_sensor:019]:   Icon: 'mdi:battery'
[09:55:45][C][ble_sensor:020]:   MAC address        : d8:71:4d:a2:8f:8a
[09:55:45][C][ble_sensor:021]:   Service UUID       : 0x180F
[09:55:45][C][ble_sensor:022]:   Characteristic UUID: 0x2A19
[09:55:45][C][ble_sensor:023]:   Descriptor UUID    : 00000000-0000-0000-0000-000000000000
[09:55:45][C][ble_sensor:024]:   Notifications      : NO
[09:55:45][C][ble_sensor:025]:   Update Interval: 60.0s
[09:55:45][C][ble_client:033]: BLE Client:
[09:55:45][C][ble_client:034]:   Address: d8:71:4d:a2:8f:8a
[09:55:45][C][esp32_ble_tracker:614]: BLE Tracker:
[09:55:45][C][esp32_ble_tracker:615]:   Scan Duration: 300 s
[09:55:45][C][esp32_ble_tracker:616]:   Scan Interval: 320.0 ms
[09:55:45][C][esp32_ble_tracker:617]:   Scan Window: 30.0 ms
[09:55:45][C][esp32_ble_tracker:618]:   Scan Type: ACTIVE
[09:55:45][C][captive_portal:148]: Captive Portal:
[09:55:45][C][ota:029]: Over-The-Air Updates:
[09:55:45][C][ota:030]:   Address: 192.168.4.215:3232
[09:55:45][C][ota:032]:   Using Password.
[09:55:45][C][api:135]: API Server:
[09:55:45][C][api:136]:   Address: 192.168.4.215:6053
[09:56:04][D][esp32_ble_tracker:628]: Found device E7:C6:B6:C3:9B:AF RSSI=-54
[09:56:04][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[09:56:19][D][ble_client:045]: Found device at MAC address [D8:71:4D:A2:8F:8A]
[09:56:27][W][ble_sensor:119]: [Battery Level] Cannot poll, not connected
```

Configuration File:
```yaml
esphome:
  name: airthings-reader
  platform: ESP32
  board: wemos_d1_mini32

logger:
  level: DEBUG

api:
ota:
  password: !secret ota_password
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
captive_portal:

sensor:
  - platform: ble_client
    ble_client_id: airthings01
    update_interval: 1min 
    name: "Battery Level"
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    icon: 'mdi:battery'
    unit_of_measurement: '%'

ble_client:
  - mac_address: D8:71:4D:A2:8F:8A
    id: airthings01

esp32_ble_tracker:
```

Valcob, 2021-10-01:
Yeah same not connected and a crash on my side 
[image]

mandreko, 2021-10-01:
From doing some BLE debugging, I'm at least seeing that the service and characteristics in the esphome code  [airthings_wave_plus.cpp](https://github.com/esphome/esphome/blob/dev/esphome/components/airthings_wave_plus/airthings_wave_plus.cpp) appear the same for me.

Using a RaspberryPi, I was able to connect to my AirThings Wave Plus, and interogate the services and characteristics. It seems that data is coming back, although I haven't pulled apart the code to see the formatting of it. But this is indicative of the code references being correct.
[image]

I'm noticing that the errors seem to suggest that the ble_sensor or ble_client are not connecting to the WavePlus. Errors saying "Cannot poll, not connected" seem to point me that way. Using the examples on the [ble_client page](https://esphome.io/components/ble_client.html), I included on_connect events, which I was unable to ever get to work, adding bias to my thought that for some reason the esphome devices are simply not connecting to the WavePlus for some reason. 

Using the AirThings WavePlus code from their [support page](https://www.airthings.com/resources/wave-plus-raspberry-pi), I was able to have it connect and query data:
[image]

Hopefully some of this information is useful.

insurgent, 2021-10-06:
I also had issues and, even though my the devices I tested with were WROOM, I purchased a WROVER board in case that might be the issue (unfortunately no D1 mini32 here to test).

Long story short, hardware wasn't the issue: The `esp32_ble_tracker:` line *MUST* be placed before the `ble_client:` line.  Both types of devices are working great now. Thank you for your work, @jeromelaban !

My working config with 2 devices:
```yaml
esphome:
  name: airthings
  platform: ESP32
  board: nodemcu-32s

logger:
  level: DEBUG

wifi:
  ssid: somessid
  password: somepass

esp32_ble_tracker:

ble_client:
  - mac_address: 80:6F:B0:E4:C3:48
    id: airthingslivingroom
  - mac_address: E0:7D:EA:24:98:BD
    id: airthingsbasement
# Enable Home Assistant API
api:

ota:

sensor:
  - platform: airthings_wave_plus
    ble_client_id: airthingslivingroom
    update_interval: 5min # 5 min is default
    temperature:
      name: "WavePlus Livingroom Temperature"
    radon:
      name: "WavePlus Livingroom Radon"
    radon_long_term:
      name: "WavePlus Livingroom Radon Long Term"
    pressure:
      name: "WavePlus Livingroom Pressure"
    humidity:
      name: "WavePlus Livingroom Humidity"
    co2:
      name: "WavePlus Livingroom CO2"
    tvoc:
      name: "WavePlus Livingroom VOC"
  - platform: airthings_wave_plus
    ble_client_id: airthingsbasement
    update_interval: 5min # 5 min is default
    temperature:
      name: "WavePlus Basement Temperature"
    radon:
      name: "WavePlus Basement Radon"
    radon_long_term:
      name: "WavePlus Basement Radon Long Term"
    pressure:
      name: "WavePlus Basement Pressure"
    humidity:
      name: "WavePlus Basement Humidity"
    co2:
      name: "WavePlus Basement CO2"
    tvoc:
      name: "WavePlus Basement VOC"
```

jeepin95, 2021-10-06:
> 
> Long story short, hardware wasn't the issue: The `esp32_ble_tracker:` line _MUST_ be placed before the `ble_client:` line. Both types of devices are working great now. Thank you for your work, @jeromelaban !
> 

This fix worked for me also on both of my devices, so perhaps just the documentation needs to be updated.