esphome/issues #2472: Error "component.h:99: undefined reference to" For custom component

ArminasTV, 2021-09-24:
### The problem

I integrated the RH_ASK class from the RadioHead library without any problem.

I want to use the RHEncryptedDriver class to add a security layer of the same library.

The compilation is happening but I have this error

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NC

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

CustomComponent

### Anything in the logs that might be useful for us?

```txt
Here is constructor of my class RfTransmitComponent

  RfTransmitComponent(uint8_t txPin, uint8_t rxPin)
  {
    _txPin = txPin;
    _rxPin = rxPin;

    AES256 code;
    code.setKey(key, 4);
    RH_ASK driver(2000, _rxPin, _txPin);
    _rf = new RHEncryptedDriver(driver, code);
    //rf = RH_ASK(2000,rxPin,txPin) ; 
  }

Linking .pioenvs/esp-tr/firmware.elf
.pioenvs/esp-tr/src/main.cpp.o:(.literal._ZN19RfTransmitComponentC5Ehh[RfTransmitComponent::RfTransmitComponent(unsigned char, unsigned char)]+0xc): undefined reference to `RHEncryptedDriver::RHEncryptedDriver(RHGenericDriver&, BlockCipher&)'
.pioenvs/esp-tr/src/main.cpp.o: In function `RfTransmitComponent::RfTransmitComponent(unsigned char, unsigned char)':
/config/esp-tr/src/esphome/core/component.h:99: undefined reference to `RHEncryptedDriver::RHEncryptedDriver(RHGenericDriver&, BlockCipher&)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp-tr/firmware.elf] Error 1
```

OttoWinter, 2021-09-26:
Please post how you included the library.

It looks like you didn't add the cpp file corresponding to the RHEncryptedDriver class to the build system

ArminasTV, 2021-09-26:
esphome:  
  name: esp-tr
  platform: ESP32
  board: esp32dev
  includes: 
    - myesphome/RfTransmitComponent.h
  libraries:
    - 'Crypto'
    - 'RadioHead'
These classes ( RHEncryptedDriver, RH_ASK  )  are part of the same "RadioHead" library

OttoWinter, 2021-09-27:
That appears to be an error in the RadioHead library/pio then.

It looks like the [lib](https://platformio.org/lib/show/124/RadioHead) only has a library.properties in its download (for arduino IDE), but not a library.json which platformio uses. PIO appears to ignore the cpp files in the root of the lib and they don't get compiled in, so you get linker errors as you see above.

Please contact the lib author to see if they can fix the issue (you can also try to replicate the issue with a minimal platformio project with just that library, I think you will get the same issue there)

closed 2022-02-10 by github-actions[bot]