esphome/issues #2474: Unable to compile for ESP32-C3

suyashbansal, 2021-09-25:
### The problem

I'm trying to compile the base ESPHome image for the **ESP32-C3-DevKitM-1** but getting an error every time. I've tried different configurations but it results in a error every time. Also, I'm using the 'Manual Download' option to create the BIN file.

### Which version of ESPHome has the issue?

v2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.3

### What platform are you using?

ESP32

### Board

ESP32-C3-DevKitM-1

### Component causing the issue

firmware.bin

### Example YAML snippet

```yaml
esphome:
  name: esp32-c3
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/esp32-c3/.pioenvs/esp32-c3/lib049/Crypto/AES128.cpp.o
In file included from /root/.platformio/lib/Crypto/AES128.cpp:23:
/root/.platformio/lib/Crypto/AES.h:200:10: fatal error: hwcrypto/aes.h: No such file or directory
 #include "hwcrypto/aes.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/esp32-c3/.pioenvs/esp32-c3/lib049/Crypto/AES128.cpp.o] Error 1
========================= [FAILED] Took 586.29 seconds =========================
```

### Additional information

I've added the complete error log in a pastebin - https://pastebin.com/Fy5Ype71

suyashbansal, 2021-09-25:
I tired adding the config mentioned by @jesserockz over at https://github.com/esphome/issues/issues/2420#issuecomment-918815907 but got similar issue.

```yaml
esphome:
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB
``` 

Here is the complete error log in another pastebin - https://pastebin.com/q86m4NaZ

glmnet, 2021-09-26:
Are you using dsmr?

OttoWinter, 2021-09-26:
> Are you using dsmr?

I've seen when using the docker images the crypto lib is automatically pulled in when web server is enabled even if dsmr is not used.

I think it's due to the PIO lib resolver, which auto detects includes and pulls in matching libs.

Only happens in docker containers because there crypto is globally installed and shipped with the image

suyashbansal, 2021-09-27:
> Are you using dsmr?

I'm not exactly sure what dsmr means... 😅

> Only happens in docker containers because there crypto is globally installed and shipped with the image

Docker is usually more efficient but should I try to install this using the PIP method? Let me know how I can help...

OttoWinter, 2021-09-27:
> I'm not exactly sure what dsmr means... sweat_smile

It's an integration that was recently added (https://esphome.io/components/sensor/dsmr.html), it appears the existance of that integration (even if it's unused) is causing problems with web server

> Docker is usually more efficient but should I try to install this using the PIP method? Let me know how I can help...

I mean this _is_ a bug, so when it's fixed it should also work with docker. Until then you can also try the pip installation, then it should work too.

agners, 2021-10-03:
Hm, this is weird: I can reproduce this with a minimal config in a container installation, but not on my dev environment.

```yaml
esphome:
  name: esp32-c3-test
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    # Currently required, see https://github.com/platformio/platform-espressif32/issues/622
    board_build.flash_mode: dio
    platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0

# Enable logging
logger:

wifi:
  ssid: "Something"
  password: "something"

captive_portal:
```

(Note, no DSMR configuration enabled. Removing `captive_portal` makes the problem go away too).

The fact that the Crypto lib is a globally preinstalled library (in `/root/.platformio/lib/Crypto`) leads it to appear in the dependency chain when the `ESPAsyncWebServer-esphome` is present.

```
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 2.0.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <Crypto> 0.2.0
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
```

However, `Crypto` is not a dependency of `ESPAsyncWebServer-esphome` as far as I can tell. What is going on here?

As for the error itself, when enabling `dsmr` then `Crypto` is legitimately pulled in:

```
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 2.0.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <Crypto> 0.2.0 < ... still?
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Dsmr> 0.5.0
|-- <Crypto> 0.2.0 < OK!
...
```

Which causes the same build error on my dev environment.

It seems that the newer ESP-IDF do not provide `hwcrypto/aes.h`. Using `aes/esp_aes.h` instead seems to work. I guess to properly fix the issue we need to fix Crypto to work with ESP-IDF 4.x...

OttoWinter, 2021-10-03:
Yeah I know _why_, just haven't gotten around to fixing it yet :P

The issue is the platformio [library dependency finder](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode) - _by default_ it looks through the source files of each lib, and auto-adds any dependencies it finds (I'm assuming using some heuristic).

It needs to be disabled for each library separately in [`library.json`](https://docs.platformio.org/en/latest/librarymanager/config.html#libldfmode) afaik.

The reason it's different in docker is because the docker image globally installs all libs we use (including Crypto, https://github.com/esphome/esphome/blob/cee08debffa2f9f33a76e8948d8a6029c1854403/docker/Dockerfile#L63), and apparently that changes the LDF behavior.

agners, 2021-10-03:
> It needs to be disabled for each library separately in library.json afaik.

Hm, that seems a sensible approach, especially for libraries which are under our control like this one.

OttoWinter, 2021-10-03:
Note: I think I looked into disabling LDF globally a couple of years ago, but that failed because back then the LDF was the _only_ way to add "internal" libs (like Update from the arduino framework) that are shipped with the framework package. I think there were some fixes for that in pio in the meantime, so maybe it works now

closed 2021-10-03 by OttoWinter