esphome/issues #2475: Compilation error for custom sensor with delay(int) line

SeByDocKy, 2021-09-25:
### The problem

I had compilation error with the last dev nightly build (09/25/21)  while with the the stable  version it compiles fine. The problem comes from the delay instruction inside the associated .h file

### Which version of ESPHome has the issue?

2021.10.1 (09/25/21) dev build

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.4

### What platform are you using?

ESP32

### Board

ESP32cam

### Component causing the issue

custom sensor

### Example YAML snippet

```yaml
esphome:
  name: potager
  platform: ESP32
  board: esp32dev
  includes:
    - tinovi_esp32_custom_sensor.h
  #libraries:
    #- "https://github.com/tinovi/i2cArduino"
    #- "https://github.com/tinovi/i2cArduino/tree/esp32"

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging

logger:
  #baud_rate: 0 #need this to free up UART pins
  #level: debug

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time  
    
    on_time:
      - seconds: 55
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: template_potager_veille
              state: !lambda return id(daily_potager).state;
              
      - seconds: /10
[... 818 lines cut ...]
	//	ESP_LOGD("custom", "vwc is: %3.2f", vwc);
	//	ESP_LOGD("custom", "vwc_old is: %3.2f", vwc_old);
		vwc_old = vwc;
	}
	/*
	else 
	{
		vwc_sensor->publish_state(vwc_old);
		ESP_LOGD("custom", "vwc_old is: %3.2f", vwc_old);
	}
	*/
	
	/*
	ESP_LOGD("custom", "status TINOVI is: %d", status);
	ESP_LOGD("custom", "e25 is: %3.2f", e25);
    ESP_LOGD("custom", "ec is: %3.2f", ec);
	ESP_LOGD("custom", "temp is: %3.2f", temp);
    ESP_LOGD("custom", "vwc is: %3.2f", vwc);
	*/
	
	//delay(1000);

	}
  }
};```

martgras, 2021-09-25:
Try esphome::delay instead

SeByDocKy, 2021-09-25:
It's compiling now .... but for me it's a breaking change. Second point, I got problem with I2C scanning.... one of two I2C sensors is not seen while with the stable version is it :)

mmakaay, 2021-09-28:
The I2C issue is also reported as an individual issue (better do so for every issue you have, I accidentally found this remark about I2C here, while browsing the latest issues, but it might easily get overlooked this way).
Check out: https://github.com/esphome/issues/issues/2486
In that issue, you can find a PR that I submitted for an I2C scanning fix.

closed 2022-02-10 by github-actions[bot]