esphome/issues #2476: PID climate tuning is broken

nekromant, 2021-09-25:
### The problem

It seems the latest change in climate components ditched 'AUTO' mode that is required for PID autotuning. There's no more 'AUTO' on the climate card in homeassistant. Attempting to do autotuning when climate is set to 'heat' or 'off' doesn't do a thing with logs complaining that the heater should be in 'AUTO' state 

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.7.1

### What platform are you using?

ESP8266

### Board

esp8266 wroom on a custom board

### Component causing the issue

climate/pid

### Example YAML snippet

```yaml
esphome:
  name: $devicename
  platform: ESP8266
  board: esp_wroom_02
  build_path: build/esp8266auto

substitutions:
  devicename: "plateheater"

logger:
  baud_rate: 0
  level: DEBUG

wifi:
  networks:
  - ssid: frostgate
    password: XXXXXXXXX

ota:
  safe_mode: True
  password: whatever
api:

web_server:

status_led:
  pin: GPIO15

switch:
  - platform: template
    name: "$devicename PID Autotune"
    turn_on_action:
      - climate.control:
          id: pid_climate
          mode: AUTO
          target_temperature: 150°C    
      - climate.pid.autotune: 
          id: pid_climate
          positive_output: 5%

output:
  - platform: esp8266_pwm
    id: pwm_gp12
    pin: GPIO12
    frequency: 5 Hz

climate:
  - platform: pid
    id: pid_climate
    name: "$devicename Heat Plate"
    sensor: the_ntc
    visual:
      min_temperature: 30
      max_temperature: 250
      temperature_step: 1
    default_target_temperature: 30°C
    heat_output: pwm_gp12
    control_parameters:
      kp: 0
      ki: 0
      kd: 0

sensor:
  - platform: ntc
    id: the_ntc
    sensor: resistance_sensor
    internal: true
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 100000
    name: "$devicename NTC Temperature"
  # Example source sensors:
  - platform: resistance
    id: resistance_sensor
    sensor: source_sensor
    configuration: UPSTREAM
    resistor: 120
    reference_voltage: 3.3
    name: Resistance Sensor
    internal: true
  - platform: adc
    name: "RAW ADC"
    internal: true
    id: source_sensor
    pin: A0
    update_interval: 0.2s
    filters:
      - median:
          window_size: 10
          send_every: 3
          send_first_at: 3
```

### Anything in the logs that might be useful for us?

```txt
[17:21:24][W][climate:053]:   Mode AUTO is not supported by this device!
[17:21:24][D][climate:038]:   Target Temperature: 150.00
[17:21:24][D][climate:373]: 'plateheater Heat Plate' - Sending state:
[17:21:24][D][climate:376]:   Mode: OFF
[17:21:24][D][climate:378]:   Action: OFF
[17:21:24][D][climate:396]:   Current Temperature: 29.82°C
[17:21:24][D][climate:402]:   Target Temperature: 150.00°C
[17:21:24][I][pid.autotune:134]: PID Autotune:
[17:21:24][I][pid.autotune:179]:   Autotune is still running!
[17:21:24][D][pid.autotune:180]:   Status: Trying to reach nan °C
[17:21:24][D][pid.autotune:181]:   Stats so far:
[17:21:24][D][pid.autotune:182]:     Phases: 0
[17:21:24][D][pid.autotune:183]:     Detected 0 zero-crossings
[17:21:24][D][pid.autotune:185]:     Current Phase Min: nan, Max: nan
[17:21:24][D][adc:104]: 'RAW ADC': Got voltage=0.00V
[17:21:24][D][sensor:121]: 'RAW ADC': Sending state 0.00488 V with 2 decimals of accuracy
[17:21:24][D][resistance:037]: 'Resistance Sensor' - Resistance 80980.8Ω
[17:21:24][D][sensor:121]: 'Resistance Sensor': Sending state 80980.79688 Ω with 1 decimals of accuracy
[17:21:24][D][ntc:026]: 'plateheater NTC Temperature' - Temperature: 29.8°C
[17:21:24][D][sensor:121]: 'plateheater NTC Temperature': Sending state 29.82436 °C with 1 decimals of accuracy
[17:21:24][W][pid.climate:122]: For PID autotuner you need to set AUTO (also called heat/cool) mode!
```

tomaszduda23, 2021-12-17:
It seems to be related to https://github.com/esphome/esphome/pull/1994

nekromant, 2021-12-20:
@tomaszduda23 Since it's now merged, I'll give it a try this weekend. If it fixes the problem that would be awesome!

tomaszduda23, 2022-01-06:
@nekromant the log does not matter. Autotune works even though the log is printed.

`[17:21:24][W][pid.climate:122]: For PID autotuner you need to set AUTO (also called heat/cool) mode!`

This is the problem:
`[17:21:24][D][climate:376]:   Mode: OFF`

You can fix it e.g. by
```
            auto call = id(pid_kuchnia).make_call();
            call.set_target_temperature(150);
            call.set_mode("HEAT");
            call.perform();`
```
Full example https://github.com/esphome/issues/issues/2916

closed 2022-09-30 by nagyrobi

wildekek, 2022-12-08:
> @nekromant the log does not matter. Autotune works even though the log is printed.

This means that there is still an issue imho. The log output is wrong and that just made me lose 1hr while debugging and digging trough the code.

nagyrobi, 2022-12-08:
No need to, there's work in progress for this: https://github.com/esphome/esphome/pull/4136

nekromant, 2022-12-10:
> > @nekromant the log does not matter. Autotune works even though the log is printed.
> 
> This means that there is still an issue imho. The log output is wrong and that just made me lose 1hr while debugging and digging trough the code. @nagyrobi should I open a new issue of do you want to reopen this one?

No need. Seems like logging is a totally different issue. I was not able to reproduce the issue since about the start of 2022. Sorry for not letting you know sooner.