esphome/issues #2478: ESPHome Tires to connect to all available Wifi Networks

adonno, 2021-09-26:
### The problem

After Flashing on esp8266 with a predefined network, and a fallback ap the esp tries to connect to every available network. 

If I disable the fallback ap it just connects to the correct network on the first try. I thought it might be related to my network configuration at first but if I pull out a bin that has been compiled on an older version of ESPHome, it just works as it should. 

### Which version of ESPHome has the issue?

2021.09.01

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

Wifi

### Example YAML snippet

```yaml
# Insert your SSID and Your PWD after inital setup
wifi:
  networks:
    - ssid: tagreader_test
    - password: network_password
  ap:
    ssid: ${devicename}

THE one working is below 

# Insert your SSID and Your PWD after inital setup
wifi:
  networks:
    - ssid: tagreader_test
    - password: network_password
#  ap:
#    ssid: ${devicename}
```

### Anything in the logs that might be useful for us?

```txt
Using 'COM4' as serial port.
Showing logs:
[11:35:38][D][wifi:397]: Found networks:
[11:35:38][I][wifi:443]: - 'AmoresNot' (82:2A:A8:F7:7D:BE) ▂▄▆█
[11:35:38][D][wifi:444]:     Channel: 1
[11:35:38][D][wifi:445]:     RSSI: -45 dB
[11:35:38][I][wifi:443]: - 'tagreader_test' (92:2A:A8:F7:7D:BE) ▂▄▆█
[11:35:38][D][wifi:444]:     Channel: 1
[11:35:38][D][wifi:445]:     RSSI: -47 dB
[11:35:38][I][wifi:443]: - 'Amores' (80:2A:A8:F7:7D:BE) ▂▄▆█
[11:35:38][D][wifi:444]:     Channel: 1
[11:35:38][D][wifi:445]:     RSSI: -47 dB
[11:35:38][I][wifi:443]: - 'Amores-Guest' (A2:2A:A8:F7:7D:BE) ▂▄▆█
[11:35:38][D][wifi:444]:     Channel: 1
[11:35:38][D][wifi:445]:     RSSI: -47 dB
[... 138 lines cut ...]
[11:36:22][C][fastled:020]:   Num LEDs: 1
[11:36:22][C][fastled:021]:   Max refresh rate: 400
[11:36:22][C][logger:193]: Logger:
[11:36:22][C][logger:194]:   Level: DEBUG
[11:36:22][C][logger:195]:   Log Baud Rate: 115200
[11:36:22][C][logger:196]:   Hardware UART: UART0
[11:36:22][C][light:097]: Light 'TagReader LED'
[11:36:22][C][light:099]:   Default Transition Length: 1.0s
[11:36:22][C][light:100]:   Gamma Correct: 2.80
[11:36:22][C][pn532:344]: PN532:
[11:36:22][C][pn532:356]:   Update Interval: 1.0s
[11:36:22][C][pn532_i2c:125]:   Address: 0x24
[11:36:22][C][status:034]: Status Binary Sensor 'TagReader Status'
[11:36:23][C][status:034]:   Device Class: 'connectivity'
[11:36:23][C][rtttl:017]: Rtttl
[11:36:23][C][captive_portal:148]: Captive Portal:
[11:36:23][C][ota:029]: Over-The-Air Updates:
[11:36:23][C][ota:030]:   Address: tagreader.local:8266
[11:36:23][C][api:135]: API Server:
[11:36:23][C][api:136]:   Address: tagreader.local:6053
```

### Additional information

I am running an ubnt unifi network

OttoWinter, 2021-09-26:
```yaml
    - ssid: tagreader_test
    - password: network_password
```
there's the problem. With the two dashes you're creating two list items, the first one will match against any network with ssid "tagreader_test" and open auth. The second one will match against any network and try to connect with the provided password.

it should instead be

```yaml
wifi:
  networks: 
    - ssid: tagreader_test
      password: network_password
```

adonno, 2021-09-26:
thank you that fixed it 
sorry guys

closed 2021-09-26 by adonno