esphome/issues #2481: Addon sometimes crashes Home Assistant OS during compiling

pattyland, 2021-09-27:
### The problem

Hi,

I know these are the worst error messages that "sometimes" happen but I don't know how to describe it better.
Sometimes I compile with the ESPHome addon 20x in a row without problems and now I had 2x in a row following history:

Compiling stops in the middle of the process, the complete log from the web gui is attached to this issue.

My complete Home Assistant instance including all addons is unreachable for about 20 minutes afterwards which causes a big chaos due to the failure of AdGuard Home as DNS server. And then after 20 minutes everything works again without me having to do anything.
The problem occurs sporadically with all ESPHome versions of the last months. It's not specific to a board or a component.

Greets, Sören

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9

### What platform are you using?

ESP32

### Board

ESP32-CAM (does not matter)

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/ender-guard.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/ender-guard
Processing ender-guard (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <Crypto> 0.2.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <noise-c> 0.1.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/ender-guard/.pioenvs/ender-guard/src/esphome/components/api/api_connection.cpp.o
Compiling /data/ender-guard/.pioenvs/ender-guard/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/ender-guard/.pioenvs/ender-guard/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/ender-guard/.pioenvs/ender-guard/src/esphome/components/api/api_pb2_service.cpp.o
```

mmakaay, 2021-09-28:
What are the specs of your Raspberry Pi device?
Issues like this most often seem to turn up when the device runs out of memory during compilation.
If you have a 1 or 2 GB model, then you might be running into this issue.

pattyland, 2021-09-29:
Sorry, I totally forgot to write this (Small feedback: This would be a good field for the form):
I am running HassOS and the ESPHome addon on an Intel NUC BOXNUC6CAYH with 4GB of memory. On average 40% of that is used.
I just updated one of my ESP to ESPHome 2021.9.0 for testing and indeed I see a peak in memory monitoring at 46%. But still far from the maximum...

mmakaay, 2021-09-29:
Okay, then we can strike a memory issue it seems. 
Anything that can be of help in the ha logs? 
It's a bit hard to guess what is going on here. ESPHome runs in its own container and that should never impact the HA container.

pattyland, 2021-10-18:
I have updated some ESP to the latest ESPHome several times in the last few weeks and the problem has not reoccurred. However, the problem has occurred sporadically over the last few years.
I will keep trying and see if anything meaningful ends up in the log....

Yes, the one with the container surprises me the most

pattyland, 2021-10-19:
It happend again, here is the log from ESPHome Addon Web GUI:

```
INFO Reading configuration /config/esphome/esphome01.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esphome01
Processing esphome01 (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Dependency Graph
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <AsyncTCP-esphome> 1.2.1
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <AsyncTCP-esphome> 1.2.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.1
|   |-- <Crypto> 0.2.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <noise-c> 0.1.3
|   |-- <libsodium> 1.10018.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Compiling /data/esphome01/.pioenvs/esphome01/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esphome01/.pioenvs/esphome01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/esphome01/.pioenvs/esphome01/src/esphome/components/api/api_server.cpp.o
Compiling /data/esphome01/.pioenvs/esphome01/src/esphome/components/api/list_entities.cpp.o
```

Again all SSH Sessions disconnect after seconds. I can't access the instance in any way again.

bigmoby, 2022-04-18:
I have the same issue on my RPI3: when I tried to install then Home Assistant becomes unstable and reboots...very annoying  :-(

zymotik, 2022-07-14:
Using an RPI3 it would crash at `Compiling /data/shed/.pioenvs/shed/src/esphome/components/api/api_pb2.cpp.o`. After I disabled every other addon it managed to build!

Henk72, 2022-08-29:
My RPI3B also crashes when compiling with ESPHOME

mmakaay, 2022-10-24:
Unfortunately, that is to be expected. This machine does not have enough physical memory for running the compilation process. As far as I know, there's no feasible work-around for that.

ssieb, 2022-10-27:
A compile option has just been added to dev, so you can limit the number of processes.

closed 2022-10-27 by ssieb