esphome/issues #2486: I2C component: Found no i2c devices!

jsuanet, 2021-09-28:
### The problem

All my ESP8266 boards using i2c and compiled with the latest dev-version stop working because the i2c devices are not found anymore.  Recompile with production version 2021.9.1 solves the problem. It looks like a regression is introduced with te latest changes in the i2c component.

Tested with: ssd1306, bh1750, bm680, lcd_pcf8574

### Which version of ESPHome has the issue?

2021.10.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

i2c

### Example YAML snippet

```yaml
Generic i2c problem with d1_mini boards
```

### Anything in the logs that might be useful for us?

```txt
logs from 2021.10.0-dev

[10:47:25][I][app:099]: ESPHome version 2021.10.0-dev compiled on Sep 28 2021, 10:46:15
[10:47:25][C][i2c.arduino:031]: I2C Bus:
[10:47:25][C][i2c.arduino:032]:   SDA Pin: GPIO5
[10:47:25][C][i2c.arduino:033]:   SCL Pin: GPIO4
[10:47:25][C][i2c.arduino:034]:   Frequency: 50000 Hz
[10:47:25][I][i2c.arduino:036]: Scanning i2c bus for active devices...
[10:47:25][I][i2c.arduino:048]: Found no i2c devices!

when using 2021.9.1 for the same device:

[10:46:44][I][app:102]: ESPHome version 2021.9.1 compiled on Sep 28 2021, 09:07:12
[10:41:28][C][i2c:028]: I2C Bus:
[10:41:28][C][i2c:029]:   SDA Pin: GPIO5
[10:41:28][C][i2c:030]:   SCL Pin: GPIO4
[10:41:28][C][i2c:031]:   Frequency: 50000 Hz
[10:41:28][I][i2c:033]: Scanning i2c bus for active devices...
[10:41:28][I][i2c:040]: Found i2c device at address 0x23
[10:41:28][I][i2c:040]: Found i2c device at address 0x77
```

SeByDocKy, 2021-09-28:
The same for me with the ESP01m  with following yaml code
```esphome:
  name: mosquitto
  platform: ESP8266
  board: esp01_1m
  
  on_boot:
    priority: -100
    then:
      - light.turn_on: mosquito_led_status

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "mosquitto Hotspot"
    password: "mosquitto"

captive_portal:

logger:
  baud_rate: 0 #need this to free up UART pins

# Enable Home Assistant API
api:

# enable over the air updates
ota:

# Enable Web server.
#web_server:
#  port: 80
  
i2c:
  sda: GPIO01  
  scl: GPIO03
  scan: True
  id: bus_i2c  
  
  
#deep_sleep:
#  id: deep_sleep_1

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    timezone:  Europe/Paris #UTC+0  #Africa/Abidjan #Africa/Algiers #Europe/Paris  #Africa/Algiers
    id: homeassistant_time

sun:
  latitude: !secret esphome_lat
  longitude: !secret esphome_long
  
#  on_sunset:
#    - then:
#       - delay: 3h
#       - deep_sleep.enter:
#           id: deep_sleep_1
#           sleep_duration: 3h

output:
  - platform: gpio
    pin:
      number: GPIO2
      inverted: True
    id: gpio_02
    
light:
[... 164 lines cut ...]
                return ( float( id(mosquitto_battery).state )  > float(id(mosquitto_tension_seuil_haut).state) );
         then:
           - switch.turn_on: mosquitto_relay
           
    - if:
         condition:
            and:
             - lambda: |-
                return (  int( id(mosquitto_regulation).state)  == 2);
             - lambda: |-
                return (  int( id(mosquitto_scheduller).state ) == 1 );
         then:
           - switch.turn_on: mosquitto_relay       
     
    - if:
         condition:
            and:
             - lambda: |-
                return ( int( id(mosquitto_regulation).state)  == 2);
             - lambda: |-
                return ( int( id(mosquitto_scheduller).state ) == 0 );
         then:
           - switch.turn_off: mosquitto_relay   
     
     ```

oxan, 2021-09-28:
See also #2453, which should've been solved in current dev.

SeByDocKy, 2021-09-28:
For my ESP01m, most of the time I can't even log wireless 
```INFO Reading configuration /config/esphome/mosquitto.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Starting log output from mosquitto.local using esphome API
WARNING Can't connect to ESPHome API for mosquitto.local: Error resolving IP address: [Errno -2] Name or service not known
INFO Trying to reconnect to mosquitto.local in the background``` and stuck in this situation

closed 2021-10-01 by OttoWinter