esphome/issues #2488: Log not work

javilopezalarcon, 2021-09-29:
### The problem

Good morning, because after waiting for the update, updating esphome and updating the esp32 and esp8266 wemos, the log still does not work. I have no idea why, but neither using the .lan domain nor using the ip of the esp ... if you need more information, let me know, but I have a fairly normal configuration and I no longer know where to look.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

Esp-wrover-kit, d1_mini

### Example YAML snippet

```yaml
esphome:
  name: salon
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
 # necesario para no usar el dominio .local tambien se puede poner una ip
 # use_address: esp-salon.lan
 # enable_mdns: false
  use_address: 192.168.1.25

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  

# Sondas temperatura
dallas:
  - pin: GPIO23
    update_interval: 00:00:05
    

sensor:
  - platform: dallas
    address: 0X3C01205014C73228
    name: "Acuario Comunitario"
    id: acuario_comunitario
  - platform: dallas  
    address: 0XAC0120500C0FF328
    name: "Gambario"
    id: gambario
    
  
# Interruptores    
switch:
  - platform: gpio
    name: Lampara Salon
    pin:
      number: 25
      inverted: true

  - platform: gpio
    name: Esquimer comunitario
    pin:
      number: 13
      inverted: true

  - platform: gpio
    name: Ventiladores Acuario comunitario
    id: ventilador_acuario
    pin:
      number: 16
      inverted: true

  - platform: gpio
    name: Calentador Acuario Comunitario
    id: calentador_acuario
    pin:
      number: 17
      inverted: true
  
# Termostatos Acuarios
climate:
  - platform: thermostat
    name: "Termostato Acuario Comunitario"
    sensor: acuario_comunitario
    default_target_temperature_low: 25 °C
    default_target_temperature_high: 27 °C
    default_mode: heat_cool
    cool_action:
      - switch.turn_on: ventilador_acuario
    heat_action:
      - switch.turn_on: calentador_acuario
    idle_action:
      - switch.turn_off: ventilador_acuario
      - switch.turn_off: calentador_acuario
    min_idle_time: 00:00:20
    min_cooling_run_time: 00:00:30 
    min_cooling_off_time: 00:00:30
    min_heating_run_time: 00:00:30
    min_heating_off_time: 00:00:30
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/salon.yaml...
INFO Starting log output from 192.168.1.25 using esphome API
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/opt/esphome/esphome/__main__.py", line 770, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 757, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 339, in command_logs
    return show_logs(config, args, port)
  File "/opt/esphome/esphome/__main__.py", line 261, in show_logs
    return run_logs(config, port)
  File "/opt/esphome/esphome/components/api/client.py", line 73, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/opt/esphome/esphome/components/api/client.py", line 24, in async_run_logs
    zc = zeroconf.Zeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 410, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 359, in create_sockets
    if add_multicast_member(cast(socket.socket, listen_socket), i):
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 276, in add_multicast_member
    listen_socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, _value)
OSError: [Errno 105] No buffer space available
```

OttoWinter, 2021-10-03:
That looks like an issue in the [zeroconf](https://github.com/jstasiak/python-zeroconf) library (or your OS).

We just call zeroconf to resolve .local hostnames.

Workaround if you're seeing this issue would be to use manual_ip in wifi section until the issue is fixed upstream

sermayoral, 2021-10-03:
@OttoWinter it seems the problem is not resolving .local hostname. In #2438 he has disabled mdns and it looks like he is using a local domain names implementing on his router.

In the log I can see:

```txt
INFO Reading configuration /config/esphome/salon.yaml...
INFO Starting log output from 192.168.1.25 using esphome API
```

So, the ip has been resolved, or it doesn't need to be solved. Maybe the problem is that it is invoking zeroconf when it shouldn't?

javilopezalarcon, 2021-10-03:
@OttoWinter, I have tried using manuap_ip without success.

Here is my configuration:

```esphome:
  name: salon
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
 # necesario para no usar el dominio .local tambien se puede poner una ip
 # use_address: esp-salon.lan
 # enable_mdns: false
  #use_address: 192.168.1.25
  manual_ip:
    static_ip: 192.168.1.25
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  

# Sondas temperatura
dallas:
  - pin: GPIO23
    update_interval: 00:00:05
    

sensor:
  - platform: dallas
    address: 0X3C01205014C73228
    name: "Acuario Comunitario"
    id: acuario_comunitario
  - platform: dallas  
    address: 0XAC0120500C0FF328
    name: "Gambario"
    id: gambario
    
  
# Interruptores    
switch:
  - platform: gpio
    name: Lampara Salon
    pin:
      number: 25
      inverted: true

  - platform: gpio
    name: Esquimer comunitario
    pin:
      number: 13
      inverted: true

  - platform: gpio
    name: Ventiladores Acuario comunitario
    id: ventilador_acuario
    pin:
      number: 16
      inverted: true

  - platform: gpio
    name: Calentador Acuario Comunitario
    id: calentador_acuario
    pin:
      number: 17
      inverted: true
  
# Termostatos Acuarios
climate:
  - platform: thermostat
    name: "Termostato Acuario Comunitario"
    sensor: acuario_comunitario
    default_target_temperature_low: 25 °C
    default_target_temperature_high: 27 °C
    default_mode: heat_cool
    cool_action:
      - switch.turn_on: ventilador_acuario
    heat_action:
      - switch.turn_on: calentador_acuario
    idle_action:
      - switch.turn_off: ventilador_acuario
      - switch.turn_off: calentador_acuario
    min_idle_time: 00:00:20
    min_cooling_run_time: 00:00:30 
    min_cooling_off_time: 00:00:30
    min_heating_run_time: 00:00:30
    min_heating_off_time: 00:00:30
 ```
Here is the log output:

```txt

INFO Reading configuration /config/esphome/salon.yaml...
INFO Starting log output from 192.168.1.25 using esphome API
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/opt/esphome/esphome/__main__.py", line 770, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 757, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 339, in command_logs
    return show_logs(config, args, port)
  File "/opt/esphome/esphome/__main__.py", line 261, in show_logs
    return run_logs(config, port)
  File "/opt/esphome/esphome/components/api/client.py", line 73, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/opt/esphome/esphome/components/api/client.py", line 24, in async_run_logs
    zc = zeroconf.Zeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 410, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 359, in create_sockets
    if add_multicast_member(cast(socket.socket, listen_socket), i):
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 276, in add_multicast_member
    listen_socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, _value)
OSError: [Errno 105] No buffer space available
```

lambchop01, 2021-10-05:
Hey,  I'm seeing the exact same error when trying to view logs through the esphome dashboard.  I'm running both HA and esphome in docker containers. all of the devices are working without issues, and the esphome dashboard says that they are online! I can even update the devices via OTA.  just can't connect to view logs. here is my config;
`esphome:
  name: garage
  platform: ESP8266
  board: nodemcuv2

wifi:
  networks:
  - ssid: "All The Little Sheep"
    password: ""
  - ssid: "TP-LINK_Extender_2.4GHz"
    password: ""
  manual_ip:
    static_ip: 192.168.2.202
    gateway: 192.168.2.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

## pins used
#D1 dht11

sensor:
  - platform: wifi_signal
    name: "Garage WiFi Signal"
    update_interval: 60s

  - platform: dht
    pin: D1
    temperature:
      name: "Garage Temperature"
    humidity:
      name: "Garage Humidity"
    update_interval: 60s
    model: DHT11`

lambchop01, 2021-10-05:
Update:  I read somewhere else that running "docker network prune" to remove unused/orphaned networks solved the problem.  I tried it and everything works as it should now!

javilopezalarcon, 2021-10-05:
> Update: I read somewhere else that running "docker network prune" to remove unused/orphaned networks solved the problem. I tried it and everything works as it should now!

Good morning, I have tried docker network prune just in case after reading that it only deletes the networks that are not used and remains the same, the error that the log does not work remains the same. My Os is Raspbian 32 bits, I have Home assistant supervised and esphome is a Ha addon
[image]

javilopezalarcon, 2021-12-01:
> That looks like an issue in the [zeroconf](https://github.com/jstasiak/python-zeroconf) library (or your OS).
> 
> We just call zeroconf to resolve .local hostnames.
> 
> Workaround if you're seeing this issue would be to use manual_ip in wifi section until the issue is fixed upstream

Good evening, I come to comment that this issue can be closed because I already found the problem and the solution, it had nothing to do with esphome, but it affected me. It looks like a docker problem, if someone needs help with a similar problem they can run sudo sysctl -w net.ipv4.igmp_max_memberships = 50 sudo sysctl -w net.ipv4.igmp_max_msf = 30 This is only a temporary change, if you reboot it would be lost. If it works, I would have to edit the etc / sysctl.conf file and add the lines to it, so when restarting, the changes would not be lost.

Thank you very much for your help 👍🏻👍🏻

closed 2021-12-01 by javilopezalarcon