esphome/issues #2489: Toggling a light via a script does not work

sztanpet, 2021-09-29:
### The problem

I have an SPDT switch that I would like to use to control a monochromatic LED strip. The switch should always toggle the LED strip, but when switching it on, I would like the LED strip to have a brightness of 100% and that rules out using light.toggle in the switch action.
So I wrote a script that manually toggles the LED strip:

```
script:
  - id: toggle_script
    then:
    - if:
        condition:
            light.is_off: light_led
        then:
          - light.turn_on:
              id: light_led
              brightness: 100%
        else:
          - light.turn_off:
              id: light_led
```

The light  only  changes state on the first run of the script and then never again on subsequent tries, but it does work reliably when using light.toggle (it just defaults to the previously set brightness level which makes it non-ideal).

When I switch the light manually from HA, and then flip the switch, the script does toggle the light, but only once (repeated switching of the spdt switch does nothing).

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

lolin_d32

### Component causing the issue

light

### Example YAML snippet

```yaml
output:
  - platform: ledc
    pin: GPIO19
    id: output_led
    frequency: "4882Hz"
  
light:
  - platform: monochromatic
    name: "LED strip"
    id: light_led
    output: output_led

binary_sensor:
  - platform: gpio
    id: led_spdt_switch
    internal: true
    device_class: light
    filters:
      - delayed_on_off: 100ms
    on_press:
      then:
        - script.execute: toggle_script
    on_release:
      then:
        - script.execute: toggle_script
    pin:
      number: GPIO32
      mode: INPUT_PULLUP

script:
  - id: toggle_script
    then:
    - if:
        condition:
            light.is_off: light_led
        then:
          - light.turn_on:
              id: light_led
              brightness: 100%
        else:
          - light.turn_off:
              id: light_led
```

### Anything in the logs that might be useful for us?

```txt
using the `toggle_script` on_press/release, the state of the LED is not set (missing the `State: ` lines)

[10:51:11][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[10:51:11][D][light:035]: 'LED strip' Setting:
[10:51:11][D][light:084]:   Transition length: 1.0s
[10:51:12][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[10:51:12][D][light:035]: 'LED strip' Setting:
[10:51:12][D][light:084]:   Transition length: 1.0s
[10:51:13][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[10:51:13][D][light:035]: 'LED strip' Setting:
[10:51:13][D][light:084]:   Transition length: 1.0s
[10:51:13][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[10:51:13][D][light:035]: 'LED strip' Setting:
[10:51:13][D][light:084]:   Transition length: 1.0s
[10:51:14][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[10:51:14][D][light:035]: 'LED strip' Setting:
[10:51:14][D][light:084]:   Transition length: 1.0s
[10:51:14][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[10:51:14][D][light:035]: 'LED strip' Setting:
[10:51:14][D][light:084]:   Transition length: 1.0s
```

switching from running a script on_press/release, to using `light.toggle: light_led` works:

```
[13:13:04][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[13:13:04][D][light:035]: 'LED strip' Setting:
[13:13:04][D][light:046]:   State: ON
[13:13:04][D][light:084]:   Transition length: 1.0s
[13:13:05][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[13:13:05][D][light:035]: 'LED strip' Setting:
[13:13:05][D][light:046]:   State: OFF
[13:13:05][D][light:084]:   Transition length: 1.0s
[13:13:06][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[13:13:06][D][light:035]: 'LED strip' Setting:
[13:13:06][D][light:046]:   State: ON
[13:13:06][D][light:084]:   Transition length: 1.0s
[13:13:06][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[13:13:06][D][light:035]: 'LED strip' Setting:
[13:13:06][D][light:046]:   State: OFF
[13:13:06][D][light:084]:   Transition length: 1.0s
[13:13:07][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[13:13:07][D][light:035]: 'LED strip' Setting:
[13:13:07][D][light:046]:   State: ON
[13:13:07][D][light:084]:   Transition length: 1.0s
[13:13:07][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[13:13:07][D][light:035]: 'LED strip' Setting:
[13:13:07][D][light:046]:   State: OFF
[13:13:07][D][light:084]:   Transition length: 1.0s
[13:13:08][D][binary_sensor:036]: 'led_spdt_switch': Sending state ON
[13:13:08][D][light:035]: 'LED strip' Setting:
[13:13:08][D][light:046]:   State: ON
[13:13:08][D][light:084]:   Transition length: 1.0s
[13:13:09][D][binary_sensor:036]: 'led_spdt_switch': Sending state OFF
[13:13:09][D][light:035]: 'LED strip' Setting:
[13:13:09][D][light:046]:   State: OFF
[13:13:09][D][light:084]:   Transition length: 1.0s
```
```

### Additional information

Tried using mode: restart in the script, did not help. Tried stopping the script before running it, also did not help.

oxan, 2021-09-29:
Might be related to #2458. Could you try setting `transition_length: 0` on the `light.turn_on` and `light.turn_off` actions?

sztanpet, 2021-09-29:
It works correctly with `transition_length: 0s`

srnoth, 2021-10-05:
Same issue on an ESP8622 (Shelly RGBW2). Setting transition to 0s "fixes" it for now.