esphome/issues #2492: Climate thermostat state is not updated when longer intervals expire

Alex9779, 2021-09-30:
### The problem

Playing with the thermostat component in preparation for making my floor heating smart I found that if I use longer intervals for `min_heating_off_time`, `min_heating_run_time` and `min_idle_time` the status in HA is not updated when the device actually switches to the new state based on the set interval if a change in HA happens within the interval.
The issue occurs if I have an interval for example of 30 seconds for all three options.
Start with a actual temperature of 20° and a set of 18°, the thermostat is off. I switch it to heating and the thermostat changes to 'Idle' as expected. Changing the set-point temperature to 22° after waiting 30 seconds brings it to 'Heating' and the device actually activates the output right away. All good.
But if I change the set-point right after switching heating on before the 30 seconds expire the status stays on 'Idle'. After the time expires the device activates the output but the status in HA does not update, it is still 'Idle' and stays 'Idle' forever.
If I then change the temperature to 18° I get 'Idle', raising the temperature before 30 seconds are over to 23° it still reads 'Idle', the devices activates the output after the 30 seconds expired so.

From what I can tell and see in the DEBUG log from the climate component is that the device only sends an update of its state right after it received a change from HA but not when a timer expires and the action is really done in case of longer intervals that allow the user to send changes while a timeout is still active.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

thermostat

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    name: test1
    sensor: temperature_test1
    default_target_temperature_low: 18 °C
    min_heating_off_time: 30s
    min_heating_run_time: 30s
    min_idle_time: 30s
    heat_action:
      - output.turn_on: hc1
    idle_action:
      - output.turn_off: hc1
```

Alex9779, 2021-09-30:
Though I do not understand the whole thing I just added `publish_state` to all time callback functions staring in line 715 of thermostat_climate.cpp (https://github.com/esphome/esphome/blob/946db3fd506509368a7d18aa070eef133ffaeb3f/esphome/components/thermostat/thermostat_climate.cpp#L715):

´´´
void ThermostatClimate::cooling_max_run_time_timer_callback_() {
  ESP_LOGVV(TAG, "cooling_max_run_time timer expired");
  this->timer_[thermostat::TIMER_COOLING_MAX_RUN_TIME].active = false;
  this->cooling_max_runtime_exceeded_ = true;
  this->trigger_supplemental_action_();
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::cooling_off_timer_callback_() {
  ESP_LOGVV(TAG, "cooling_off timer expired");
  this->timer_[thermostat::TIMER_COOLING_OFF].active = false;
  this->switch_to_action_(this->compute_action_());
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::cooling_on_timer_callback_() {
  ESP_LOGVV(TAG, "cooling_on timer expired");
  this->timer_[thermostat::TIMER_COOLING_ON].active = false;
  this->switch_to_action_(this->compute_action_());
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::fan_mode_timer_callback_() {
  ESP_LOGVV(TAG, "fan_mode timer expired");
  this->timer_[thermostat::TIMER_FAN_MODE].active = false;
  this->switch_to_fan_mode_(this->fan_mode.value_or(climate::CLIMATE_FAN_ON));
  if (this->supports_fan_only_action_uses_fan_mode_timer_)
    this->switch_to_action_(this->compute_action_());
  this->publish_state();
}

void ThermostatClimate::fanning_off_timer_callback_() {
  ESP_LOGVV(TAG, "fanning_off timer expired");
  this->timer_[thermostat::TIMER_FANNING_OFF].active = false;
  this->switch_to_action_(this->compute_action_());
  this->publish_state();
}

void ThermostatClimate::fanning_on_timer_callback_() {
  ESP_LOGVV(TAG, "fanning_on timer expired");
  this->timer_[thermostat::TIMER_FANNING_ON].active = false;
  this->switch_to_action_(this->compute_action_());
  this->publish_state();
}

void ThermostatClimate::heating_max_run_time_timer_callback_() {
  ESP_LOGVV(TAG, "heating_max_run_time timer expired");
  this->timer_[thermostat::TIMER_HEATING_MAX_RUN_TIME].active = false;
  this->heating_max_runtime_exceeded_ = true;
  this->trigger_supplemental_action_();
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::heating_off_timer_callback_() {
  ESP_LOGVV(TAG, "heating_off timer expired");
  this->timer_[thermostat::TIMER_HEATING_OFF].active = false;
  this->switch_to_action_(this->compute_action_());
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::heating_on_timer_callback_() {
  ESP_LOGVV(TAG, "heating_on timer expired");
  this->timer_[thermostat::TIMER_HEATING_ON].active = false;
  this->switch_to_action_(this->compute_action_());
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}

void ThermostatClimate::idle_on_timer_callback_() {
  ESP_LOGVV(TAG, "idle_on timer expired");
  this->timer_[thermostat::TIMER_IDLE_ON].active = false;
  this->switch_to_action_(this->compute_action_());
  this->switch_to_supplemental_action_(this->compute_supplemental_action_());
  this->publish_state();
}
```

This seems to solve my problem. Also after resetting the device the states are sent to HA now properly because the component uses some of those timers at setup to delay the execution of certain actions (see https://github.com/esphome/esphome/blob/946db3fd506509368a7d18aa070eef133ffaeb3f/esphome/components/thermostat/thermostat_climate.cpp#L10)

Could this be a solution or do I miss something?

kbx81, 2021-10-01:
This is interesting -- I use five minutes for all of the delay intervals and I have never noticed this behavior before. That said, what you are describing makes sense. I've got an idea for a cleaner way to implement a fix. I'll set up a branch and push a fix. Would you be able & willing to test?

Alex9779, 2021-10-01:
Sure I am able and willing... Thanks!

Alex9779, 2021-10-01:
I found your branch at https://github.com/kbx81/esphome/tree/thermostat-publish-state-fix and gave it a try and now it works as expected!

kbx81, 2021-10-01:
Great, thanks! I meant to PR it yesterday night but was having some issues with my local builds. Glad to hear it works -- I'll PR it today.

closed 2021-10-03 by OttoWinter