esphome/issues #2496: ESP32 stucks using http_request in the switch.is_turn_off condition

sermayoral, 2021-10-01:
### The problem

I know it may seem strange, but an esp32 gets stuck if I try to compile this code:

```
switch:
  - platform: gpio
    pin:
      number: 33
      inverted: yes
    name: "Mesita Marilena"
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_off:
      then:
        http_request.post: http://192.168.33.42/switch/lamparita_peques/turn_on
```

After flashing it, the api doesn't respond. I can reflash the esp32, and i can ping it, but Home Assistant doesn't recognize it, neither the ESPHome log:

```
INFO Reading configuration /config/esphome/dormitorio.yaml...
INFO Starting log output from esp-dormitorio.lan using esphome API
WARNING Can't connect to ESPHome API for esp-dormitorio.lan: Error connecting to ('192.168.33.38', 6053): [Errno 111] Connect call failed ('192.168.33.38', 6053)
INFO Trying to reconnect to esp-dormitorio.lan in the background
```

However, if I replace the 'on_turn_off' condition for the 'on_turn_on' conditions, after reflash it, it works:

```
switch:
  - platform: gpio
    pin:
      number: 33
      inverted: yes
    name: "Mesita Marilena"
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      then:
        http_request.post: http://192.168.33.42/switch/lamparita_peques/turn_on
```

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

switch

### Anything in the logs that might be useful for us?

```txt
esphome:
  name: dormitorio
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret nombre_wifi
  password: !secret passwd_wifi
    
  # Necesario para no usar el dominio .local. Tambien se puede poner una IP
  use_address: esp-dormitorio.lan
  enable_mdns: false
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dormitorio Fallback Hotspot"
    password: !secret passwd_esphome

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret passwd_esphome

ota:
  password: !secret passwd_esphome
  
# Habilita el acceso web al ESP
# web_server:
#   port: 80

http_request:
  useragent: esphome/dormitorio
  timeout: 10s

# Interruptores
switch:
  - platform: gpio
    pin:
      number: 27
      inverted: yes
    name: "Mesita Sergio"
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: gpio
    pin:
      number: 33
      inverted: yes
    name: "Mesita Marilena"
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_off:
      then:
        http_request.post: http://192.168.33.42/switch/lamparita_peques/turn_on
```

ssieb, 2021-10-01:
I found a reference that indicates this could be caused because wifi isn't setup yet, which is definitely the case here.  Try either checking for wifi being connected before making the request or add `setup_priority: 249` to the switch.

sermayoral, 2021-10-01:
I can confirm it's working if I change the code by the following:

```yaml
switch:
  - platform: gpio
    pin:
      number: 33
      inverted: yes
    name: "Mesita Marilena"
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_off:
      then:
        - if:
            condition:
              wifi.connected:
            then:
              - http_request.post: http://192.168.33.45/light/led_televisin/turn_off
            else:
              - logger.log: 
                  format: "Error sending http_request: Wifi not connected"
                  level: WARN
````

@ssieb thanks! It probably wouldn't have detected it without that reference.

IMHO I think the http_request component should do this check instead of leaving the esp stucked. It is very difficult to realize this.

ssieb, 2021-10-01:
Can you please open a new issue for that?

closed 2021-10-01 by ssieb