esphome/issues #2497: Speed Fan

AjitKIPL, 2021-10-01:
### The problem

Created switch component using Custom UART Text Sensor.
https://esphome.io/cookbook/uart_text_sensor.html
working ok.

Now want to add a Speed Fan Component that will work on UART Frame TX and RX.
UART Write is working ok shown below code, but don't know where to check the UART frame received to turn on Fan & set fan speed using lambdas.
kindly advise me to solve this.

fan:
  - platform: speed
    output: fanoutput
    id: fan_1
    speed_count: 4
    name: "Fan"
    on_turn_on:
      - if:
          condition:
            lambda: return (id(fan_1).speed == 1);
          then:
            - uart.write: "Fa01A11|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 2);
          then:
            - uart.write: "Fa01A12|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 3);
          then:
            - uart.write: "Fa01A13|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 4);
          then:
            - uart.write: "Fa01A14|"

    on_turn_off:
      - if:
          condition:
            lambda: return (id(fan_1).speed == 1);
          then:
            - uart.write: "Fa01A01|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 2);
          then:
            - uart.write: "Fa01A02|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 3);
          then:
            - uart.write: "Fa01A03|"
      - if:
          condition:
            lambda: return (id(fan_1).speed == 4);
          then:
            - uart.write: "Fa01A04|"
 
    on_speed_set:
      - if:
          condition:
            lambda: return ((id(fan_1).state) && (id(fan_1).speed == 1));
          then:
            - uart.write: "Fa01A11|"
      - if:
          condition:
            lambda: return ((id(fan_1).state) && (id(fan_1).speed == 2));
          then:
            - uart.write: "Fa01A12|"
      - if:
          condition:
            lambda: return ((id(fan_1).state) && (id(fan_1).speed == 3));
          then:
            - uart.write: "Fa01A13|"
      - if:
          condition:
            lambda: return ((id(fan_1).state) && (id(fan_1).speed == 4));
          then:
            - uart.write: "Fa01A14|"
            
output:
  - platform: template
    id: fanoutput
    type: float
    write_action:
      - uart.write: ""

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

WeekendWarrior1, 2021-10-05:
This page should help get you started: https://esphome.io/cookbook/uart_text_sensor.html

AjitKIPL, 2021-10-05:
Hi,
Thanks for your reply and suggestion.
see below the steps that I followed,

Step 1) Copy and paste suggested code it shows error Mapping value are not allowed here at below line :
                    return (id(fan_1).state) ? myUartStringsOn[id(fan_1).speed - 1] : "";
Step 2) modify !lambda to !lambda |- in below code mapping error remove
on_speed_set:
        - uart.write: !lambda |-
            static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
            return (id(fan_1).state) ? myUartStringsOn[id(fan_1).speed - 1] : "";

Step 3)
fan:
  - platform: speed
    output: fanoutput
    id: fan_1
    speed_count: 4
    name: "Fan"
    on_turn_on:
        - uart.write: !lambda 
            static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
            return myUartStringsOn[(id(fan_1).speed) - 1];
    on_turn_off:
        - uart.write: !lambda 
            static std::string myUartStringsOff[4] = {"Fa01A01|", "Fa01A02|", "Fa01A03|", "Fa01A04|"};
            return myUartStringsOff[(id(fan_1).speed) - 1];
    on_speed_set:
        - uart.write: !lambda |-
            static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
            return (id(fan_1).state) ? myUartStringsOn[id(fan_1).speed - 1] : "";

After execution shows below error log:-

> INFO Reading configuration /config/esphome/sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/sensor
Processing sensor (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Warning! You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <noise-c> 0.1.1
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/sensor/.pioenvs/sensor/src/main.cpp.o
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:52:138: error: could not convert 'myUartStringsOn[(fan_1->esphome::fan::FanState::speed + -1)]' from 'std::string {aka std::basic_string<char>}' to 'std::vector<unsigned char>'
         - uart.write: !lambda 
                                                                                                                                          ^
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:56:140: error: could not convert 'myUartStringsOff[(fan_1->esphome::fan::FanState::speed + -1)]' from 'std::string {aka std::basic_string<char>}' to 'std::vector<unsigned char>'
         - uart.write: !lambda 
                                                                                                                                            ^
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:62:67: error: could not convert '(fan_1->esphome::fan::FanState::state ?  std::basic_string<char>((*(const std::basic_string<char>*)(& myUartStringsOn[(fan_1->esphome::fan::FanState::speed + -1)]))) : std::basic_string<char>(((const char*)""), (*(const std::allocator<char>*)(& std::allocator<char>()))))' from 'std::string {aka std::basic_string<char>}' to 'std::vector<unsigned char>'
             return (id(fan_1).state) ? myUartStringsOn[id(fan_1).speed - 1] : "";
                                                                   ^
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:53:3: warning: control reaches end of non-void function [-Wreturn-type]
             static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
   ^
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:57:3: warning: control reaches end of non-void function [-Wreturn-type]
             static std::string myUartStringsOff[4] = {"Fa01A01|", "Fa01A02|", "Fa01A03|", "Fa01A04|"};
   ^
/config/esphome/sensor.yaml: In lambda function:
/config/esphome/sensor.yaml:63:3: warning: control reaches end of non-void function [-Wreturn-type]
             
   ^
*** [/data/sensor/.pioenvs/sensor/src/main.cpp.o] Error 1
========================= [FAILED] Took 24.35 seconds =========================

AjitKIPL, 2021-10-05:
Hi,
UART Read by ESP from external MCU 
FAN ON UART frames with speed 1-4 received on UART : {"Fa01K11|", "Fa01K12|", "Fa01K13|", "Fa01K14|"};
FAN OFF UART frames with speed 1-4 received on UART : {"Fa01K01|", "Fa01K02|", "Fa01K03|", "Fa01K04|"};
How & where to write the code for the above logic? need help on that part.

UART Write by ESP from Home Assistant GUI to external MCU working ok with the code that I have shared previously.

WeekendWarrior1, 2021-10-05:
Sorry, actually tested that this compiled for you:
```yaml
fan:
  - platform: speed
    output: fanoutput
    id: fan_1
    speed_count: 4
    name: "Fan"
    on_turn_on:
      - uart.write: !lambda |-
          static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
          std::string chosenString = myUartStringsOn[id(fan_1).speed - 1];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_turn_off:
      - uart.write: !lambda |-
          static std::string myUartStringsOff[4] = {"Fa01A01|", "Fa01A02|", "Fa01A03|", "Fa01A04|"};
          std::string chosenString = myUartStringsOff[id(fan_1).speed - 1];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_speed_set:
    - if:
        condition:
          lambda: 'return (id(fan_1).state);'
        then:
          - uart.write: !lambda |-
              static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
              std::string chosenString = myUartStringsOn[id(fan_1).speed - 1];
              std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
              return vec;
```

WeekendWarrior1, 2021-10-05:
> Hi, UART Read by ESP from external MCU FAN ON UART frames with speed 1-4 received on UART : {"Fa01K11|", "Fa01K12|", "Fa01K13|", "Fa01K14|"}; FAN OFF UART frames with speed 1-4 received on UART : [...]

Follow the guide I linked above:

> https://esphome.io/cookbook/uart_text_sensor.html

AjitKIPL, 2021-10-05:
Thanks for your quick response.
Actually, I have followed the same link to implement to develop 1 Switch and 1 Speed Fan, see below complete code.
try to add lambda's in speed fan component like the switch to check received frames for a fan then it shows the invalid option.

esphome:
  name: sensor
  platform: ESP8266
  board: nodemcuv2
  includes:
    - uart_read_line_sensor.h

# Enable logging
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart

# Enable Home Assistant API
api:

ota:
  password: "XXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: "XXXXX"
  password: "XXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "XXXX Hotspot"
    password: "1234567890"

captive_portal:

uart:
  id: uart_bus
  tx_pin: GPIO1 #D0
  rx_pin: GPIO3 #D1
  baud_rate: 9600

text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"

fan:
  - platform: speed
    output: fanoutput
    id: fan_1
    speed_count: 4
    name: "Fan"
    on_turn_on:
      - uart.write: !lambda |-
          static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
          std::string chosenString = myUartStringsOn[id(fan_1).speed - 1];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_turn_off:
      - uart.write: !lambda |-
          static std::string myUartStringsOff[4] = {"Fa01A01|", "Fa01A02|", "Fa01A03|", "Fa01A04|"};
          std::string chosenString = myUartStringsOff[id(fan_1).speed - 1];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_speed_set:
    - if:
        condition:
          lambda: 'return (id(fan_1).state);'
        then:
          - uart.write: !lambda |-
              static std::string myUartStringsOn[4] = {"Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
              std::string chosenString = myUartStringsOn[id(fan_1).speed - 1];
              std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
              return vec;
      
output:
  - platform: template
    id: fanoutput
    type: float
    write_action:
    - uart.write: ""
   
switch:
  - platform: template
    name: "Switch 1"
    lambda: |-
      if (id(uart_readline).state == "Sa01K10|") {
        return true;
      } else if(id(uart_readline).state == "Sa01K00|") {
        return false;
      } else {
        return{};
      }
    turn_on_action:
      - uart.write: "Sa01A10|"
    turn_off_action:
      - uart.write: "Sa01A00|"

WeekendWarrior1, 2021-10-05:
Please format your code so I can read it:
```` 
```yaml
Your code here
```
````

AjitKIPL, 2021-10-05:
```yaml
esphome:
  name: sensor
  platform: ESP8266
  board: nodemcuv2
  includes:
    - uart_read_line_sensor.h

# Enable logging
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart

# Enable Home Assistant API
api:

ota:
  password: "abeb15ad698606a220176e4313b5a0ec"

wifi:
  ssid: "XXXXX"
  password: "XXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "XXXXX Hotspot"
    password: "1234567890"

captive_portal:

uart:
  id: uart_bus
  tx_pin: GPIO1 #D0
  rx_pin: GPIO3 #D1
  baud_rate: 9600

text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"

fan:
  - platform: speed
    output: fanoutput
    id: fan_1
    speed_count: 4
    name: "Fan"
    on_turn_on:
      - uart.write: !lambda |-
          static std::string myUartStringsOn[5] = {"Fa01A10|", "Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
          if(id(fan_1).speed == 0)
            id(fan_1).speed = 1;
          std::string chosenString = myUartStringsOn[id(fan_1).speed];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_turn_off:
      - uart.write: !lambda |-
          static std::string myUartStringsOff[5] = {"Fa01A00|","Fa01A01|", "Fa01A02|", "Fa01A03|", "Fa01A04|"};
          std::string chosenString = myUartStringsOff[id(fan_1).speed];
          std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
          return vec;
    on_speed_set:
    - if:
        condition:
          lambda: return (id(fan_1).state);
        then:
          - uart.write: !lambda |-
              static std::string myUartStringsOn[5] = {"Fa01A10|", "Fa01A11|", "Fa01A12|", "Fa01A13|", "Fa01A14|"};
              std::string chosenString = myUartStringsOn[id(fan_1).speed];
              std::vector<uint8_t> vec(chosenString.begin(), chosenString.end());
              return vec;
      
output:
  - platform: template
    id: fanoutput
    type: float
    write_action:
    - uart.write: ""
    
switch:
  - platform: template
    name: "Switch 1"
    lambda: |
      if (id(uart_readline).state == "Sa01K10|") {
        return true;
      } else if(id(uart_readline).state == "Sa01K00|") {
        return false;
      } else {
        return{};
      }
    turn_on_action:
      - uart.write: "Sa01A10|"
    turn_off_action:
      - uart.write: "Sa01A00|"
    
```

AjitKIPL, 2021-10-06:
HI,
Have you checked my code?
In Fan component does not have lambdas like the Switch to check received frames.

WeekendWarrior1, 2021-10-06:
So you want to use: https://esphome.io/components/text_sensor/index.html#on-value
as your action to run your automations off, since it will trigger when you receive new packets over serial.

AjitKIPL, 2021-10-06:
Hi,
As per the suggestion modify test_sensor, but not work.
kindly check the code below -

```ymal
text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"
    on_value:
      then:
        - lambda: |-
            if (id(uart_readline).state == "Fa01K11|")
            {
              id(fan_1).speed = 1;
              id(fan_1).state = 1;
            }
            else if (id(uart_readline).state == "Fa01K01|")
            {
              id(fan_1).speed = 1;
              id(fan_1).state = 0;
            }
```

WeekendWarrior1, 2021-10-06:
Can you please elaborate on how it did not work?
Did it not compile? Or did the `on_value` automation not fire?

AjitKIPL, 2021-10-06:
In ESPHome device log shows below - when sending Fa01K01| for Fan OFF & Fa01K11| for Fan ON speed 1, from MCU to ESP chip. But the state of Fan and Speed not update in Home Assistant GUI. Is id(fan_1).state works outside the fan component?

[11:13:45][D][text_sensor:015]: 'uart_readline': Sending state 'Fa01K01|'
[11:13:52][D][text_sensor:015]: 'uart_readline': Sending state 'Fa01K11|'

AjitKIPL, 2021-10-06:
Code compiles and on_value automation fires.

WeekendWarrior1, 2021-10-06:
```yaml
text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"
    on_value:
      then:
        - lambda: |-
            if (id(uart_readline).state == "Fa01K11|")
            {
              auto call = id(fan_1).state->turn_on();
              call.set_speed(1);
              call.perform();
            }
            else if (id(uart_readline).state == "Fa01K01|")
            {
              auto call = id(fan_1).state->turn_off();
              call.perform();
            }
```
Using these calls should publish the state (though I haven't tested this compiles!)

WeekendWarrior1, 2021-10-06:
No problem!
feel free to close this issue then :)

closed 2021-10-06 by AjitKIPL